esp8266/Arduino #3668: Reading json response using ESP 8266 over an HTTPs connection

aritra0723, 2017-10-02:
### Basic Infos

0down votefavorite | I am trying to read I am trying to read the json response over the HTTPS connection from googles direction API and display the response on my serial monitor of an ESP8266 12E. I am able to connect to wifi and also google maps api. But while reading the values all I am getting is Garbage value.

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char* ssid = "NETGEAR90";
const char* password = "thirstychair027";

const char* host = "https://maps.googleapis.com";
const int httpsPort = 443;

void setup() {
Serial.begin(115200);
Serial.println();
Serial.print("connecting to ");
Serial.println(ssid);
WiFi.begin(ssid, password);
while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
}
Serial.println("");
Serial.println("WiFi connected");
Serial.println("IP address: ");
Serial.println(WiFi.localIP());

 // Use WiFiClientSecure class to create TLS connection
WiFiClientSecure client;
Serial.print("connecting to ");
Serial.println(host);
if (client.connect(host, httpsPort))
Serial.println("connected to google maps");

if (!client.connect(host, httpsPort)) {
  Serial.println("connection failed");
  return;
}

   String url = "https://maps.googleapis.com/maps/api/directions/json?
      origin=22.631123,%2088.378373&destination=Kamarhati&departure_time=now
       &&traffic_model=best_guess&key=XXX";
   Serial.print("requesting URL: ");
   Serial.println(url);

   client.print(String("GET ") + url + " HTTP/1.1\r\n" +
           "Host: " + host + "\r\n" 
           );
   Serial.println("request sent");

   while (client.connected()) {
     char c = client.read();
     Serial.print(c);
     }
    Serial.flush();
   delay(5000);

  String line = client.readStringUntil('}');
   /*if (line.startsWith("{\"geocoded_waypoints\":\"
   [{\"geocoder_status\":\"OK\"")) {
       Serial.println("esp8266/Arduino CI successfull!");
    } else {
    Serial.println("esp8266/Arduino CI has failed");
    }
    Serial.println("reply was:");
      Serial.println("==========");*/
      Serial.println(line);
      Serial.println("==========");
      Serial.println("closing connection");
    }

     void loop() {
    }

### Debug Messages

Output is Garbage value: ????????????????????????????????????????????????????????????????????????????????

quarxdmz, 2017-10-06:
maybe you have set your Serial Baudrate to a different value other than what you have defined "`Serial.begin(115200);`"

vshymanskyy, 2017-11-28:
Could be possibly related to https://github.com/esp8266/Arduino/issues/3002

itsashishupadhyay, 2018-04-20:
I think it is because of "https" and esp isn't able to connect as it fails SSL authentication, Try "http://maps.googleapis.com/maps/api/directions/json"

earlephilhower, 2019-01-31:
Original author hasn't been seen in over a year.  We've moved to BearSSL a long time ago, so closing for now.  If the latest 2.5.0 doesn't work, please open a new issue.

closed 2019-01-31 by earlephilhower