esp8266/Arduino #3670: Pulse sensor amped was inaccurate when work with ESP 8266

LanceLDY, 2017-10-02:
If i did't connect to esp8266, my pulse sensor can take the accurate reading(60-90). Once it connect to the esp 8266, the reading is around 90-150.
I also don't know what is the problem... Hope you guys can help me solve the issue. Thanks!
 Below is the code and the circuit diagram (take from online article that I refer to)...
[image]

#include <SoftwareSerial.h>
#define DEBUG true
#include <LiquidCrystal.h>
#include <stdlib.h>
LiquidCrystal lcd(12,11,5,4,3,2);
SoftwareSerial esp8266(9,10); 
/*-----------------ESP8266 Serial WiFi Module---------------*/
String SSID ="vivo V3Max";
String PASS="12345678";
#define IP "184.106.153.149"// thingspeak.com ip
String msg = "GET /update?key=SZDRPGCYMCSNT43W&field1="; //change it with your key...
/*-----------------------------------------------------------*/
//Variables
boolean halo = false;
float temp;
int hum;
String tempC;
int error;
int pulsePin = 0;                 // Pulse Sensor purple wire connected to analog pin 0
int blinkPin = 13;                // pin to blink led at each beat
int fadePin = 5;
int fadeRate = 0;
// Volatile Variables, used in the interrupt service routine!
volatile int BPM;                   // int that holds raw Analog in 0. updated every 2mS
volatile int Signal;                // holds the incoming raw data
volatile int IBI = 600;             // int that holds the time interval between beats! Must be seeded! 
volatile boolean Pulse = false;     // "True" when heartbeat is detected. "False" when not a "live beat". 
volatile boolean QS = false;        // becomes true when Arduino finds a beat.
// Regards Serial OutPut  -- Set This Up to your needs
static boolean serialVisual = true;   // Set to 'false' by Default.  Re-set to 'true' to see Arduino Serial Monitor ASCII Visual Pulse 
volatile int rate[10];                    // array to hold last ten IBI values
volatile unsigned long sampleCounter = 0;          // used to determine pulse timing
volatile unsigned long lastBeatTime = 0;           // used to find IBI
volatile int P =512;                      // used to find peak in pulse wave, seeded
volatile int T = 512;                     // used to find trough in pulse wave, seeded
volatile int thresh = 525;                // used to find instant moment of heart beat, seeded
volatile int amp = 100;                   // used to hold amplitude of pulse waveform, seeded
volatile boolean firstBeat = true;        // used to seed rate array so we startup with reasonable BPM
volatile boolean secondBeat = false;      // used to seed rate array so we startup with reasonable BPM
void setup()
{
  pinMode(6,OUTPUT);
  digitalWrite(6,HIGH);
  pinMode(7,INPUT_PULLUP);
  digitalWrite(7,HIGH);

  
  lcd.begin(16, 2);
  lcd.print("wait a moment...");
  delay(100);
  lcd.setCursor(0,1);
  lcd.print("Connecting...");
  delay(1000);
Serial.begin(9600); //or use default 115200.
  esp8266.begin(9600);
  Serial.println("AT");
  esp8266.println("AT");
  delay(3000);
  if(esp8266.find((char*)"OK")){
    connectWiFi();

  }
  interruptSetup(); 
}
void loop(){
[... 133 lines cut ...]
      runningTotal /= 10;                     // average the last 10 IBI values 
      BPM = 60000/runningTotal;               // how many beats can fit into a minute? that's BPM!
      QS = true;                              // set Quantified Self flag 
      // QS FLAG IS NOT CLEARED INSIDE THIS ISR
    }                       
  }
  if (Signal < thresh && Pulse == true){   // when the values are going down, the beat is over
    digitalWrite(blinkPin,LOW);            // turn off pin 13 LED
    Pulse = false;                         // reset the Pulse flag so we can do it again
    amp = P - T;                           // get amplitude of the pulse wave
    thresh = amp/2 + T;                    // set thresh at 50% of the amplitude
    P = thresh;                            // reset these for next time
    T = thresh;
  }
  if (N > 2500){                           // if 2.5 seconds go by without a beat
    thresh = 512;                          // set thresh default
    P = 512;                               // set P default
    T = 512;                               // set T default      
    firstBeat = true;                      // set these to avoid noise
    secondBeat = false;                    // when we get the heartbeat back
      lastBeatTime = sampleCounter;          // bring the lastBeatTime up to date  
  }
  sei();     
  // enable interrupts when youre done!
}// end isr

LanceLDY, 2017-10-02:
I also had the button problem. I need to press for a while, so that my button will on the pulse sensor

devyte, 2017-10-02:
@LanceLDY your sketch shows AT cmds for the ESP. I think you're in the wrong repo, this is a direct development firmware, not the AT firmware.
I suggest asking at a forum instead of an issue tracker for a repo.
Closing.

closed 2017-10-02 by devyte