esp8266/Arduino #3672: Problem Button Reset

TheChapu, 2017-10-02:
### Basic Infos

Hardware
Hardware: ESP-12E
Core Version:? 2.1.0-rc2?

### Description Description
I'm writing a code that receives data from 3 sensors (DHT11-HCRS-04-DS18B20) to send to a visualization platform.
I also have a Reset Button. The button has a 10kOhms resistance.

### Problem description

When loading the code and pressing the button is "reset"but does not restart the NodeMCU module, I must press the RST button on the board once, and now if I can restart the module with my Button.

### Settings in IDE

Module: NodeMCU 1.0 (ESP12-E Module)
Flash Size: 4MB
CPU Frequency: 80Mhz
Flash Mode:??
Flash frequency:??
Upload Using: SERIAL?
Reset Method: nodemcu?

### Sketch
```

#include <NewPing. h>
*include "DHT. h" // Sensor DHT11
*include <OneWire. h> // Library for Sensor T
*include <DallasTemperature. h> // Library for Sensor T

ONE_WIRE_BUS 10 // Temperature sensor D10
#Define DHTPIN 13 // Pin D7
? define DHTTYPE DHT11 // Select sensor
# Define trig 4 // Ultra Sensor D1
#Define echo 5 // Ultra Sensor D2
define pinReset 9 // Reset button D9
#Define relay 14 // pump D5
#Define relayLight 12 //light   
# Define MAX_DISTANCE 200

Ultrasonic ultrasonic ultrasonic (trig, echo); // Ultrasonic sensor pins (Trig/Echo)
NewPing sonar (trig, echo, MAX_DISTANCE);
DHT dht (DHTTPIN, DHTTYPE, 11); //Create instance of DHT11,11 is the number of cycles
OneWire oneWire (ONE_WIRE_BUS); 
DallasTemperature sensors (&oneWire);

float TON, TOFF, LON, dist, humidity, temp_c, temp_water; // 
int relayStatus; ///
int RelayStatusLight; //

void setup () {
  ? put your setup code here, to run once:
  Serial. begin (115200);
  pinMode (pinReset, INPUT_PULLUP);
  pinMode (relay, OUTPUT); // Pin relay Pump as Output
  pinMode (relayLight, OUTPUT); // Pin relay Light as Output

  Interruption that will call function to restart
  attachInterrupt (digitalPinToInterrupt (pinReset), doSth, CHANGE); 

}

void loop () {
  // put your main code here, to run repeatedly:
  test ();
}

// Reset button
void doSth ({
  Serial. println ("We must restart...");
  ESP. restart ();
}

void test (){     
  Activate Relay
  digitalWrite (relay, HIGH); 
  digitalWrite (relayLuz, HIGH); 
  Delay (200);
  
  Sensor Reading

  DHT11 sensor
  int chk = dht. read ();
  //Serial. print ("chk:"); Serial. println (chk);
  switch (chk)
  {
    case 1: Serial. println ("OK"); break;
    case 0: Serial. println ("Checksum error"); break;
    case -1: Serial. println ("Time out error"); break;
    default: Serial. println ("Unknown error"); break;
  }
  float humidity = dht. readHumidity (); // Read humidity (percent)
  float temp_c = dht. readTemperature (); // Read temperature as Fahrenheit
  DHT11 /FIN DHT11

  /Sensor 
   sensors. requestTemperatures (); //Send command to obtain the Temperature
   temp_water = sensors. getTempCByIndex (0); //We get the temperature
  End Sensor

  Ultrasonic Sensor
   float dist = sonar. ping_cm ();
   String data = "{" data ":[{{" channel ":" 3 "," value ":" +String (dist)+ "," mac ":" 5ccf7f85e8c0cf91 "}, {" channel ":" 11 "," value ":" +String (   
  Serial. println (data);
  digitalWrite (relay, LOW); 
  digitalWrite (relayLuz, LOW); 
  Delay (200);
```

###Debug Messages

{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.13,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": 19.00
OK. OK.
{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.13,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": 19.00
OK. OK.
{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.13,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": 19.00
OK. OK.
{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.13,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": 19.00
OK. OK.
{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.13,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": 19.00
OK. OK.
{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.19,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": 19.00
OK. 
{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.13,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": 19.00
We must Reboot....

 ets Jan 8 2013, rst cause: 2, boot mode: (1,6)
wdt

###(RESET by pressing the button on the plate)

{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.19,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": nan,".
Checksum error
{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.19,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": nan,".
Checksum error
{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.19,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": nan,".
OK. OK.
{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.19,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": 19.00
OK. OK.
{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.19,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": 19.00
OK. OK.
{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.19,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": 19.00
We must Reboot....
":[{" channel ":" 3 "," value ": 7.00," mac ":" 5ccf7f85e8c0cf91 "}, {" channel ":" 11 "," value ": 22.19," mac ":" 5ccf7f85e8c0c0cf91 "}, {" channel ":" 1 "," value ": 19.00," mac
We must Reboot....

 ets Jan 8 2013, rst cause: 2, boot mode: (3,4)
###PHYSICAL BUTTON RESET

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
Checksum error
{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.19,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": nan,".
Checksum error
{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.19,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": nan,".
Checksum error
We must Reboot....
{"data":[{"channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91"}, {"channel":"11","value": 22.19,"mac":"5ccf7f85e8c0c0cf91"}, {"channel":"1","valueWe Should Restart

 ets Jan 8 2013, rst cause: 2, boot mode: (3,4)
###PHYSICAL BUTTON RESET

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
Checksum error
{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.25,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": nan,".
Checksum error
{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.25,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": nan,".
Checksum error
{ "data":[{ "channel":"3","value": 7.00,"mac":"5ccf7f85e8c0cf91" }, { "channel":"11","value": 22.25,"mac":"5ccf7f85e8c0c0cf91" }, { "channel":"1","value": nan,".
OK. OK.

image modulo

devyte, 2017-10-03:
@TheChapu there is a known issue with restarting right after serial flashing. You have to press the reset button manually at least once after a serial flash in order to assure correct restart. After that, ESP.restart() should work as intended.
I _think_ the symptom is similar as yours: without a physical reset after the serial flashing, the next ESP.restart() doesn't behave correctly, and can result in a WDT + hang afterwards.
I also think that OTA isn't affected.
Please flash your code, do a physical reset, and test your sketch for correct behavior with your reset button. Report back here whether it fixes your issue.

TheChapu, 2017-10-04:
@devyte 
First of all, thank you very much for your answer.

I tell you a little bit about what I am developing, in case you can give me other ideas.

I have a Nodemcu, Sensor DHT11, Sensor HCSR04, Sensor ds18b20, a button and a double relay.

The code is for monitoring a hydroponic system, the DHT11 sensor is for measuring Humidity and Ambient Temperature, the HCSR04 sensor is for measuring water tank capacity, the ds18b20 sensor is for measuring water temperature and the relay for activating the pump and LED panel.

I am using the WIFI-MANAGER library to enter the ON/OFF switch-off times of the Pump/LIGHT, the wifi network and password to send the data to a display platform and finally the button is to "reset" the ON/OFF time values of the pump and the light, after that the Microcontroller is restarted to reload the variables.

Do you understand me?
Do you have any ideas for me?

I am attentive to your answer.

RudyFiero, 2017-10-08:
@devyte
This is not a problem about resetting the CPU. He is using GPIO-9 as a button input that the program is to use to reset program variables. 

@TheChapu  
That board uses a ESP-12E module. You can not use GPIO-9 with that module. It is connected to the internal flash and is to be used for that purpose only. It is not an available pin for IO purposes.  

http://www.esp8266.com/wiki/lib/exe/fetch.php?media=schematic_esp-12e.png

The problem is trying to use hardware that is not available for use. This is not a ESP8266-Arduino code problem. Please close the issue.

devyte, 2017-10-08:
@RudyFiero Thanks, the markup is kind of messed up and I missed that!
@TheChapu there are pins that you can't use, and gpio9 is one of them. Change the reset button to a different pin, and you should be good. Look up the nodemcu pins for your board to figure out which are free for general use.
Closing due to user error.

closed 2017-10-08 by devyte

TheChapu, 2017-10-11:
Thank you very much @RudyFiero for the clarification and @devyte for to information

I was able to read all the links.

In conclusion I can only use GPIO 1,3,12,14,15?

I am attentive to your response

RudyFiero, 2017-10-11:
You can also use 4 and 5. You can use 16 but it does not have a pull up resistor. It is pull down when enabled.