esp8266/Arduino #3673: Can't access the Sigma Delta functions

Jsolo1, 2017-10-03:
#define fanpwm 15
#include "sigma_delta.h"

void setup() {
  pinMode(fanpwm, OUTPUT);
  // put your setup code here, to run once:
sigma_delta_setup(PERIPHS_IO_MUX_MTDI_U,fanpwm,FUNC_GPIO15);
set_sigma_duty_312KHz(2);

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages

error: 'sigma_delta_setup' was not declared in this scope

```
I can't use the Sigma delta functions, tried to add the file manually but still fails. How can I use them? Might it be problem with the "unused" extension of the file?

Thanks
```

Jsolo1, 2017-10-03:
Things I've done:
- Removed the extension ".unused" and the file seems working now
- Redeclared the functions, specially the  "sigma_delta_setup" which is completely undeclared in the header file
- Include the header file in the sketch: "#include "sigma_delta.h" "
Now I have the following errors:

```
In file included from C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/include/ets_sys.h:12:0,

                 from C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\core_esp8266_sigma_delta.c:31:

C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\core_esp8266_sigma_delta.c: In function 'sigma_delta_setup':

C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\core_esp8266_sigma_delta.c:78:98: error: 'GPIO_SIGMA_DELTA_NUM' undeclared (first use in this function)

                                                         (GPIO_REG_READ(GPIO_PIN_ADDR(GPIO_ID_PIN(GPIO_SIGMA_DELTA_NUM))) &(~GPIO_SIGMA_DELTA_SETTING_MASK))| 

                                                                                                  ^

C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/include/eagle_soc.h:50:98: note: in definition of macro 'WRITE_PERI_REG'

 #define WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val)

                                                                                                  ^

C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\core_esp8266_sigma_delta.c:77:5: note: in expansion of macro 'GPIO_REG_WRITE'

     GPIO_REG_WRITE(GPIO_SIGMA_DELTA, 

     ^

C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/include/eagle_soc.h:49:53: note: in expansion of macro 'ETS_UNCACHED_ADDR'

 #define READ_PERI_REG(addr) (*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr)))

                                                     ^

C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/include/eagle_soc.h:79:52: note: in expansion of macro 'READ_PERI_REG'

 #define GPIO_REG_READ(reg)                         READ_PERI_REG(PERIPHS_GPIO_BASEADDR + reg)

                                                    ^

C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\core_esp8266_sigma_delta.c:78:58: note: in expansion of macro 'GPIO_REG_READ'

                                                         (GPIO_REG_READ(GPIO_PIN_ADDR(GPIO_ID_PIN(GPIO_SIGMA_DELTA_NUM))) &(~GPIO_SIGMA_DELTA_SETTING_MASK))| 

                                                          ^

C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\core_esp8266_sigma_delta.c:78:72: note: in expansion of macro 'GPIO_PIN_ADDR'

                                                         (GPIO_REG_READ(GPIO_PIN_ADDR(GPIO_ID_PIN(GPIO_SIGMA_DELTA_NUM))) &(~GPIO_SIGMA_DELTA_SETTING_MASK))| 

                                                                        ^

C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\core_esp8266_sigma_delta.c:78:86: note: in expansion of macro 'GPIO_ID_PIN'

                                                         (GPIO_REG_READ(GPIO_PIN_ADDR(GPIO_ID_PIN(GPIO_SIGMA_DELTA_NUM))) &(~GPIO_SIGMA_DELTA_SETTING_MASK))| 

                                                                                      ^

C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\core_esp8266_sigma_delta.c:78:98: note: each undeclared identifier is reported only once for each function it appears in

                                                         (GPIO_REG_READ(GPIO_PIN_ADDR(GPIO_ID_PIN(GPIO_SIGMA_DELTA_NUM))) &(~GPIO_SIGMA_DELTA_SETTING_MASK))| 

                                                                                                  ^

C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/include/eagle_soc.h:50:98: note: in definition of macro 'WRITE_PERI_REG'

 #define WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val)
[... 163 lines cut ...]
                                                     ^

C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/include/eagle_soc.h:79:52: note: in expansion of macro 'READ_PERI_REG'

 #define GPIO_REG_READ(reg)                         READ_PERI_REG(PERIPHS_GPIO_BASEADDR + reg)

                                                    ^

C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\core_esp8266_sigma_delta.c:166:58: note: in expansion of macro 'GPIO_REG_READ'

                                                         (GPIO_REG_READ(GPIO_PIN_ADDR(GPIO_ID_PIN(GPIO_SIGMA_DELTA_NUM))) &(~GPIO_SIGMA_DELTA_PRESCALE_MASK))| 

                                                          ^

C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\core_esp8266_sigma_delta.c:166:72: note: in expansion of macro 'GPIO_PIN_ADDR'

                                                         (GPIO_REG_READ(GPIO_PIN_ADDR(GPIO_ID_PIN(GPIO_SIGMA_DELTA_NUM))) &(~GPIO_SIGMA_DELTA_PRESCALE_MASK))| 

                                                                        ^

C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\core_esp8266_sigma_delta.c:166:86: note: in expansion of macro 'GPIO_ID_PIN'

                                                         (GPIO_REG_READ(GPIO_PIN_ADDR(GPIO_ID_PIN(GPIO_SIGMA_DELTA_NUM))) &(~GPIO_SIGMA_DELTA_PRESCALE_MASK))| 

```

igrr, 2017-10-06:
Sorry, i don't think that sigma_delta was ever tested/supported, which is why it has .unused extension. I'm actually surprised it was checked into git, I must have done that by accident.

If you can get it to work, then a PR with the necessary changes would certainly be appreciated.

Jsolo1, 2017-10-09:
Ok.

I found the following link: http://bbs.espressif.com/viewtopic.php?t=49
And I modified the sigma_delta.c file as follows:
```

/*
/******************************************************************************
 * MODULEName : set_sigma MODULE

EACH PIN CAN CONNET TO A SIGMA-DELTA , ALL PINS SHEARS THE SAME SIGMA-DELTA SOURCE.

THE TARGET DUTY AND FREQUENCY CAN BE MODIFIED VIA THE REG ADDR GPIO_SIGMA_DELTA

THE TARGET FREQUENCY IS DEFINED AS:

FREQ = 80,000,000/prescale * target /256  HZ,     0<target<128
FREQ = 80,000,000/prescale * (256-target) /256  HZ,     128<target<256
target: duty ,0-255
prescale: clk_div,0-255
so the target and prescale will both affect the freq.

YOU CAN DO THE TEST LIKE THIS:
1. INIT :  sigma_delta_setup(PERIPHS_IO_MUX_MTDI_U,12,FUNC_GPIO12);
2. USE 312K:   set_sigma_duty_312KHz(2);
OR 2. SET VAL:  set_sigma_target(uint8 target) AND  set_sigma_prescale

3. DEINIT AND DISABLE: sigma_delta_close(uint32 GPIO_NUM), eg.sigma_delta_close(2)

*******************************************************************************/

#include "osapi.h"
#include "c_types.h"
#include "ets_sys.h"
#include "eagle_soc.h"
#include "gpio.h"
//#include "esp8266.h"

//uint32 GPIO_SIGMA_DELTA_NUM=0x68;

#define GPIO_SIGMA_DELTA_NUM  0x68
 // modify for pin

#define GPIO_SIGMA_DELTA      0x68  //defined in gpio register.xls

#define   PERIPHS_GPIO_BASEADDR   0x60000300
#define GPIO_PIN_ADDR(i)    (GPIO_PIN0_ADDRESS + i*4)
#define GPIO_SIGMA_DELTA_ADDRESS   (PERIPHS_GPIO_BASEADDR + 0x68)  //defined in gpio register.xls

#define GPIO_SIGMA_DELTA_SETTING_MASK  (0x00000001ff)

#define GPIO_SIGMA_DELTA_ENABLE  1
#define GPIO_SIGMA_DELTA_DISABLE   (~GPIO_SIGMA_DELTA_ENABLE)
#define GPIO_SIGMA_DELTA_MSB                     16
#define GPIO_SIGMA_DELTA_LSB                      16
#define GPIO_SIGMA_DELTA_MASK                  (0x00000001<<GPIO_SIGMA_DELTA_LSB)
#define GPIO_SIGMA_DELTA_GET(x)                 (((x) & GPIO_SIGMA_DELTA_MASK) >> GPIO_SIGMA_DELTA_LSB)
#define GPIO_SIGMA_DELTA_SET(x)                  (((x) << GPIO_SIGMA_DELTA_LSB) & GPIO_SIGMA_DELTA_MASK)

#define GPIO_SIGMA_DELTA_TARGET_MSB  7
#define GPIO_SIGMA_DELTA_TARGET_LSB  0
#define GPIO_SIGMA_DELTA_TARGET_MASK                  (0x000000FF<<GPIO_SIGMA_DELTA_TARGET_LSB)
#define GPIO_SIGMA_DELTA_TARGET_GET(x)                 (((x) & GPIO_SIGMA_DELTA_TARGET_MASK) >> GPIO_SIGMA_DELTA_TARGET_LSB)
#define GPIO_SIGMA_DELTA_TARGET_SET(x)                  (((x) << GPIO_SIGMA_DELTA_TARGET_LSB) & GPIO_SIGMA_DELTA_TARGET_MASK)

#define GPIO_SIGMA_DELTA_PRESCALE_MSB  15
#define GPIO_SIGMA_DELTA_PRESCALE_LSB  8
#define GPIO_SIGMA_DELTA_PRESCALE_MASK  (0x000000FF<<GPIO_SIGMA_DELTA_PRESCALE_LSB)
#define GPIO_SIGMA_DELTA_PRESCALE_GET(x)                 (((x) & GPIO_SIGMA_DELTA_PRESCALE_MASK) >> GPIO_SIGMA_DELTA_PRESCALE_LSB)
#define GPIO_SIGMA_DELTA_PRESCALE_SET(x)                  (((x) << GPIO_SIGMA_DELTA_PRESCALE_LSB) & GPIO_SIGMA_DELTA_PRESCALE_MASK)
[... 182 lines cut ...]

#include <stdint.h>

void sigma_delta_setup(uint32_t GPIO_MUX,uint32_t GPIO_NUM,uint32_t GPIO_FUNC);
void sigma_delta_close(uint32_t gpio);
//void sigma_delta_close(uint32 GPIO_NUM);
void set_sigma_target(uint8_t target);
//void set_sigma_target(uint8 target);
void set_sigma_prescale(uint8_t prescale);
//void set_sigma_prescale(uint8 prescale);
void set_sigma_duty_312KHz(uint8_t duty);
//void set_sigma_duty_312KHz(uint8 duty);

#endif//SIGMA_DELTA_H
```

And it works! There is output in the pin, but the output is very strange it fades away, like a capacitor being discharged:
[image]

And when the pulse is shorter:
[image]

So this might be something related with the hardware itself? Probably the sigma_delta is somwhow designed to work at high frequencies only?

Regards

Jsolo1, 2017-10-11:
Ok, i found a solution, in order to make it push-pull output, I added this line in the arduino:

`GPC(15) = (GPC(15) & (0 << GPCD)); `

Would be good to figure it out which registers is it modifying and do it automatically in the library.
I also added this other functions to the library to  create pwm signals at a different frequency:

```
int get_duty(uint8 prescale, uint8 duty, float freq){
	int duty_new;
	int target;
	
	target = (int)((prescale + 1)*freq);
	//target = (duty > 128)?(256-duty):duty;
	duty_new = (duty>128)?(256-target):target;
	return duty_new;
	
}

int get_prescale(uint8 duty, float freq){
	
	uint8 target = 0;
	int prescale=0;
    target = (duty>128)?(256-duty):duty;
	if (target == 0){
		prescale = 0;
	}else{
		prescale = (int)(((float)target) / freq) - 1;
		
	}
	//prescale = (int)(((float)target) / freq) - 1;
	return prescale;
	
}

/******************************************************************************
 * FunctionName : set_sigma_duty_freq.
 * Description  : Creates a pwm signal at a defined frequency and duty cycle.
 * Parameters   : uint8 duty, TARGET DUTY,  0-255
 *				  float freq, TARGET FREQUENCY,  0.05-120, 1 means 312Khz. Frequency range 1.2Khz-37.5MHz
 *				  bool const_freq, if true, it will limit the duty cycle to try to keep the frequency constant.
 *							if false, the frequency will shift at high duty cycles and "freq" less than 1,
 *							but the duty cycle will remain as constant as possible
 * Returns      : none
*******************************************************************************/
void ICACHE_FLASH_ATTR set_sigma_duty_freq(uint8 duty, float freq, bool const_freq)
{
	uint8 target = 0;
	int prescale = 0;
	int duty_new = duty;
    //target = (duty > 128)?(256-duty):duty;
	
	prescale = get_prescale(duty,freq);	
	
	if (prescale <= 0){
		prescale = 0;
		duty_new = get_duty(prescale,duty,freq);
	}
	
	
	if (const_freq == true){
		//target = 1;
		if (prescale > 255){
			prescale = 255;
			duty_new = get_duty(prescale,duty,freq);
		}
	}else{
		//target = 10;
		if (prescale > 255){
			prescale = 255;
		}
	}
	//prescale = (target==0)?0:(target-1);
    //freq = 80000 (khz) /256 /duty_target * (prescale+1)
    set_sigma_target((uint8)duty_new);//SET DUTY TARGET
    set_sigma_prescale((uint8)prescale);//SET CLK DIV

}
```

stefaandesmet2003, 2018-01-08:
A working demo, and suggestion for interfacing the sigma delta generator.
still figuring out how to do a pull request and upload the code properly ..
[SigmaDeltaDemo.zip](https://github.com/esp8266/Arduino/files/1613345/SigmaDeltaDemo.zip)

devyte, 2018-01-09:
@stefaandesmet2003 fork the repo, map your fork to your ccomputer, create a branch, submit your changes, push upstream (to your remote, I.e. your fork).
Then, visit this repo again, and you should see a notification about your submit. You can review, and create the PR right there on the webpage with just a click on the button.

stefaandesmet2003, 2018-01-09:
Ok, thanks!

pashinen, 2018-06-05:
Im trying to run above sigma delta program in non-os SDK.. will it work?
If yes how can i put my dynamic data on i2s sigma delta.

pashinen, 2018-06-05:
I'm generating a sine lookup table of size 256 and want tp play random frequency via play function.
Let me know how can i put samples on i2s sigma delta pin/line.

stefaandesmet2003, 2018-06-05:
Yes, the example code will work in arduino (non-os sdk) environment.The sigma-delta generator is different from the i2s peripheral. For creating i2s samples to send to an external DAC, you will need the i2s peripheral.The SD generator has only limited control over the output frequency, I haven't tried if it's sufficiently accurate in the audio frequency range. Best have a look at the esp8266 documentation.

pashinen, 2018-06-05:
Also how can i link i2s peripheral to SD because i need to integrate a mic through codec which requires i2s so eventually i need i2s.
And how can i read i2s_in data from i2s_in line, ESP8266(as a slave).

earlephilhower, 2018-06-05:
@pashinen Please use www.esp8266.com or other forum for these kinds of questions.  This issue handler is for bugs in the Arduino core.

Closing since SigmaDelta was added a while back, thanks to @stefaandesmet2003's work.

closed 2018-06-05 by earlephilhower

pashinen, 2018-06-20:
Hi stefaandesmet2003,

SigmaDeltaDemo is not working, i cant find any led blinking via sigma delta only toggle is working in that code,
Please tell me what is the issue.

pashinen, 2018-06-20:
Hi stefaandesmet2003,

Its working but need to connect led outside.

stefaandesmet2003, 2018-06-20:
ok, great!
Looks like LED_BUILTIN definition doesn't match with your board

pashinen, 2018-06-20:
Hi Jsolo1,

Hot to generate the desired frequency?

pashinen, 2018-06-20:
Hi Jsolo1,

How you relate the freq parameter with desired frequency?..

positron96, 2018-09-01:
Hello. What's the status of sigma-delta library? Cannot find it in current release (2.4.something), as well as SigmaDeltaDemo example.

earlephilhower, 2018-09-01:
@positron96 2.4.2 definitely has it included.  Double check you've installed the latest version in the IDE, or use the GIT head.

positron96, 2018-09-03:
@earlephilhower I had 2.4.1 and it only had ``core_esp8266_sigma_delta.c.unused `` source file. Version 2.4.2 has ``core_esp8266_sigma_delta.c`` source file, but I can't seem to find an example of usage. Is it present?