esp8266/Arduino #3674: EEPROM cleared Suddenly.

arihantdaga, 2017-10-03:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	1.8.3
Using Arduino Version.

### Description

I am working on ESP8266 for quiet some time.Its been good so far. I am using EEPROM for storing some custom settings. For the purpose i am using Embedis Library(https://github.com/thingSoC/embedis). And i begin EEPROM using
EEPROM.begin(4096);
Most of the time its working fine, but one night suddenly EEPROM got flushed and it was left with 0xff for the complete 4096 bytes. It keep happening sometimes out of blue. For most of the time its working fine, but suddenly sometimes it is erased. I could not figure out any solid reason for this. 

### Settings in IDE

Module:  			?NodeMCU 1.0 ?
Flash Size: 		4MB/(3MB SPIFSS)
CPU Frequency:		80Mhz
Flash Frequency:	80Mhz
Upload Using:		SERIAL

devyte, 2017-10-03:
@arihantdaga How long have you been using the eeprom? How often do you write to it?
The ESP doesn't have eeprom. The eeprom api is meant for compatibility, the data is really stored on the flash in a specific area, which means if you write too often, the flash sector(s) in question will wear out quickly and start failing.
The number of writes to a specific sector before damage depends on the flash vendor, but as a rule of thumb you consider 10000 full write cycles.

arihantdaga, 2017-10-03:
Thank you @devyte . I am using eeprom for storing settings as well as storing switch status(I have 5 switches). 
I have been using it for quiet some time though. but what do you mean by " 10000 full write cycles." Does one eeprom.commit means one full cycle ? . 
And what will happen after 10000 cycles, will it still work sometime and not work sometimes? or it stops working completely ? . Because in my case i am able to write the config and retrieve it but just sometimes it looses the data suddenly. 

And 10,000 can be very less in my case, is there an alternative for persistent storage in esp apart from EEPROM then ? for storing settings. ?

devyte, 2017-10-03:
@arihantdaga after a sector is damages, anything can happen. It can fail to work at all, it can work somtimes, or what's worst of all, it can seem to work, but then the data is lost after some short time.
About the write cycle, you should read up on flash tech. I'm not sure exactly how the EEPROM layer works, nor how the flash api works either, but in general, when you write a piece of data, you have to read a page, erase the page, merge your data into what was read, and then write the page back. Depending on implementation of the translation layer, and of the writing code, this doesn't necessarily happen on every write operation. However, if you are writing small pieces of data, it will happen for the same sector each time.
I personally avoid the EEPROM like the plague, and also avoid using e.g.: Wifi with persistent true. And I use the SPIFFS for storing data, in a file. SPIFFS does wear leveling under the hood, so write operations get distributed all over the SPIFFS flash area, which is a lot.
This issue is either hardware issue, or usage error. In either case, closing, see #3655 .

closed 2017-10-03 by devyte