esp8266/Arduino #3675: WiFiClientSecure gets stuck with bad connection

aignatiev, 2017-10-03:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.3.0

### Description

The problem is that WiFiClientSecure gets stuck when the WiFi connection is weak. It just gets stuck for a random amount of time, usually it's just less than a minute or two, but at least once it was stuck for 20 minutes. It does not restart, it does not print any debug messages, it's just stuck. The test with 20 minutes was done by using a phone as a hotspot and slowly walking away from the ESP, the ESP continued to be stuck even when the phone was brought closer, the ESP continued to function normally only after the hotspot was switched off. I also tried to apply the fix listed in #3537, but it didn't seem to help.

Additional info: I'm making a sensor that logs data to flash memory whenever there is no internet, and uploads this data to Amazon AWS Lambda function whenever there is internet. The samples should be taken arond every 16 seconds and these random delays are no good. Would also be nice to control the timeouts, because the data can always be resent from memory. Please help me to fix this problem or give any advice how to overcome this.

### Sketch

```cpp
WiFiClientSecure client;
...
char * buffer[1000] = "Long JSON here";
client.write(buffer, strlen(buffer));
```

### Debug Messages

```
none
```

igrr, 2017-10-06:
Could you please enable logging and post the log for the situation when this happens? In Arduino IDE: Tools > Debug Level = Core + SSL, and add `Serial.setDebugOutput(true);` to `setup` function.

The reason why i'm asking is that i'm having trouble reproducing this issue. Logs will likely contain at least some information which can provide a clue. Thanks.

aignatiev, 2017-10-07:
I'm using PlatformIO instead of Arduino IDE, so I don't know how to enable other debug prints than `Serial.setDebugOutput(true);`. And here's what it gives after being stuck for 35 seconds (trying to upload around 200 bytes of data):
```
bcn_timout,ap_probe_send_start
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
f r0, 
```
Additional info: the connection is kept open, because the upload happens every minute or so and reopening a secure connection every time uses lots of data. The status of the connection is checked just before an upload with `if(!client.connected()) return false;`.

igrr, 2017-10-07:
Can you try [adding extra flags](http://docs.platformio.org/en/latest/projectconf.html) to CPPFLAGS? (or CFLAGS and CXXFLAGS)

`-DDEBUG_ESP_CORE -DDEBUG_ESP_SSL`

Once you do that successfully, you should see lots of debug messages from WiFiClient (starting with a colon) as well as messages from axTLS.

aignatiev, 2017-10-09:
I suppose `build_flags = -DDEBUG_ESP_CORE -DDEBUG_ESP_SSL` did the trick and I see more debug prints. This thing is difficult to replicate but this time it was nicely stuck for around two minutes and the pattern is very easy to see..
```
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:sent 261
:ww
[1575443] HTTP header sent
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:er -9 0 1
:ww
Alert: close notify
```
The code is supposed to first upload the HTTP header and then the JSON data, it was stuck for 51 seconds uploading the header and 65 more seconds for the JSON, which eventually failed.

Rob58329, 2017-10-12:
Hardware: Wemos D1 mini
Core Version: 2.4.0-rc1 (via ArduinoIDE Boards Manager on a Windows-PC)

I am also having what is probably the same issue with WiFiClientSecure::print("....."), where the ESP8266 just hangs if the internet-connection disappears (and sometimes after a few minutes generates an exception and reboots).  It is most easily caused by simply disconnecting the internet connection (but NOT the wifi): My ESP8266 is connected to the hotspot of my mobile-phone, and then this is connected to the internet over 4G.  If you just turn-off the 4G whilst the ESP8266 is in the middle of doing a sequence of WiFiClientSecure::print("....."), the ESP8266 hangs...  (NB. the ESP8266 is still conected via wifi to the hotspot, but just no longer to the internet.)

A solution would be appreciated!!

machadoroger, 2017-10-25:
I'm facing the same issue. I've tried almost (I hope) everything to fix it with no success.

ESP8266 just hangs for no good reason while trying to connect the server (client.connect(HOST, PORT)). It can take place after 20 min, 1h, 4h, and so on... The same happens with the time the ESP8266 remains hanging. I mean... sometimes it can take 10 min and sometimes it can last 2 hours.

For some reason, as @Rob58329 has mentioned, in the meantime, if I turn off/on the internet and re-connect, the ESP8266 starts responding.

Please @igrr , let me know if you need more information. Thank you in advance.

igrr, 2017-10-25:
@machadoroger from your description it doesn't look like the same issue. The issue described in the original post is related to writes not being handled correctly. In you case it is the connection phase. Are you using WiFiClientSecure or WiFiClient?

machadoroger, 2017-10-25:
@igrr thanks for your quick reply. I'm using WifiClientSecure. Just like @Rob58329 , when I'm using WifiClient everything works fine.

aignatiev, 2017-11-17:
In the meantime I tried to use this code with 2.4.0-rc2 and it seems that I cannot reproduce this issue with it. The current code has been updated and has a payload of around 1k, but I still managed to get it stuck with 2.3.0 and with 2.4.0-rc2 it just seems to work.

P.S.
I've found an easy way to use Arduino 2.4.0-rc2 with PlatformIO and for those who are interested, use this in platformio.ini 
```
platform = https://github.com/platformio/platform-espressif8266.git#feature/2.4.0-rc2
```
It should work as long as the branch exists.

zipiju, 2017-11-24:
@igrr  I seem to have a similiar problem which can be reproduced (end fixed?) easily.

I use WifiClientSecure to connect to a web server and to download a file.
If during the file download I break the connection (in a way that IP is still assigned and ESP is still associated to a WLAN - fe. by unplugging LAN cable from AP) the ESP gets stuck in a manner that WDT won't fire.

It seems that any calls to client (at least client->connected() and client->available()) are blocking indefinitely.

I was able to fix this blocking behavior by removing SSL_READ_BLOCKING from ssl_ctx_new() fc. in WiFiClientSecure.cpp SSLContext() class constructor.
Not sure what other implications this have, but if none maybe it should be changed.
Or can this be changed in a way that user will be able to choose blockin/non-blocking socket?

And also noticed that WiFiClientSecure.h is missing timeout parm. in connect() fc., so I guess there is no way to specify connect failed timeout.

Thank you.

pjunni, 2018-02-13:
I get timeouts much longer than defined (WIFI_CLIENT_TIMEOUT in code). Tested with values 1, 2000 and 5500 and each time the timeout is around 6600 - 7000 milliseconds. My code is:

```
client = new WiFiClientSecure; 
client->setTimeout(WIFI_CLIENT_TIMEOUT);
int startTime = millis();
client->connect(URL, PORT);
Serial.println(String("total time: ") + String(millis() - startTime));```

devyte, 2018-05-29:
BearSSL is merged in #4273 , with alternate BearSSL::WiFi* classes. Although axtls-based classes are still available and even the default, they are planned for deprecation and then retirement, hence won't be fixed. Any issues with BearSSL-based classes should be reported in new issues.
Closing.

closed 2018-05-29 by devyte

macedolfm, 2019-01-28:
Did anyone manage to fix the WiFiClientSecure library as this a VERY important feature for any reasonable secure IoT application? For what I have tested the Client is still very unstable, gets disconnected during publish for no reason, may get stuck connecting for minutes, crashes the ESP32 sending long (over 1024 bytes), get stuck if MQTT server stops, get stuck if WiFi stops and all sort of unexpeceted probles in a very random fashion.

I´ve been trying everything for the last couple days and letting the MCU run overnight to collec error data and just can´t figure out what is wrong (I am NOT a inner guts expert), tried with "arduino-esp32-master" version 1.0.0 & 1.0.1, even tried [this guy tips](http://www.iotsharing.com/2017/08/how-to-use-esp32-mqtts-with-mqtts-mosquitto-broker-tls-ssl.html?m=1) and a ["this partial release"](https://github.com/espressif/arduino-esp32/tree/a38ffe58fe1d73a56295edb0c6051a46932e0175) of WiFiClientSecure but still, lots of trouble.

I managed to addapt the own library example to show case the many problems:

`#include <WiFi.h> 
#include <WiFiClientSecure.h>

#define  ARDUINOJSON_USE_LONG_LONG  1
#define  ARDUINOJSON_USE_DOUBLE     1
#include <ArduinoJson.h>

#define MQTT_MAX_PACKET_SIZE 4096
#include <PubSubClient.h>

#include <SerialDebug.h> // https://github.com/JoaoLopesF/SerialDebug

// Lets Encrypt root CA-CRT

const char* test_root_ca = "-----BEGIN CERTIFICATE-----\n" \
  "MIIEkjCCA3qgAwIBAgIQCgFBQgAAAVOFc2oLheynCDANBgkqhkiG9w0BAQsFADA/\n" \
  "MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT\n" \
  "DkRTVCBSb290IENBIFgzMB4XDTE2MDMxNzE2NDA0NloXDTIxMDMxNzE2NDA0Nlow\n" \
  "SjELMAkGA1UEBhMCVVMxFjAUBgNVBAoTDUxldCdzIEVuY3J5cHQxIzAhBgNVBAMT\n" \
  "GkxldCdzIEVuY3J5cHQgQXV0aG9yaXR5IFgzMIIBIjANBgkqhkiG9w0BAQEFAAOC\n" \
  "AQ8AMIIBCgKCAQEAnNMM8FrlLke3cl03g7NoYzDq1zUmGSXhvb418XCSL7e4S0EF\n" \
  "q6meNQhY7LEqxGiHC6PjdeTm86dicbp5gWAf15Gan/PQeGdxyGkOlZHP/uaZ6WA8\n" \
  "SMx+yk13EiSdRxta67nsHjcAHJyse6cF6s5K671B5TaYucv9bTyWaN8jKkKQDIZ0\n" \
  "Z8h/pZq4UmEUEz9l6YKHy9v6Dlb2honzhT+Xhq+w3Brvaw2VFn3EK6BlspkENnWA\n" \
  "a6xK8xuQSXgvopZPKiAlKQTGdMDQMc2PMTiVFrqoM7hD8bEfwzB/onkxEz0tNvjj\n" \
  "/PIzark5McWvxI0NHWQWM6r6hCm21AvA2H3DkwIDAQABo4IBfTCCAXkwEgYDVR0T\n" \
  "AQH/BAgwBgEB/wIBADAOBgNVHQ8BAf8EBAMCAYYwfwYIKwYBBQUHAQEEczBxMDIG\n" \
  "CCsGAQUFBzABhiZodHRwOi8vaXNyZy50cnVzdGlkLm9jc3AuaWRlbnRydXN0LmNv\n" \
  "bTA7BggrBgEFBQcwAoYvaHR0cDovL2FwcHMuaWRlbnRydXN0LmNvbS9yb290cy9k\n" \
  "c3Ryb290Y2F4My5wN2MwHwYDVR0jBBgwFoAUxKexpHsscfrb4UuQdf/EFWCFiRAw\n" \
  "VAYDVR0gBE0wSzAIBgZngQwBAgEwPwYLKwYBBAGC3xMBAQEwMDAuBggrBgEFBQcC\n" \
  "ARYiaHR0cDovL2Nwcy5yb290LXgxLmxldHNlbmNyeXB0Lm9yZzA8BgNVHR8ENTAz\n" \
  "MDGgL6AthitodHRwOi8vY3JsLmlkZW50cnVzdC5jb20vRFNUUk9PVENBWDNDUkwu\n" \
  "Y3JsMB0GA1UdDgQWBBSoSmpjBH3duubRObemRWXv86jsoTANBgkqhkiG9w0BAQsF\n" \
  "AAOCAQEA3TPXEfNjWDjdGBX7CVW+dla5cEilaUcne8IkCJLxWh9KEik3JHRRHGJo\n" \
  "uM2VcGfl96S8TihRzZvoroed6ti6WqEBmtzw3Wodatg+VyOeph4EYpr/1wXKtx8/\n" \
  "wApIvJSwtmVi4MFU5aMqrSDE6ea73Mj2tcMyo5jMd6jmeWUHK8so/joWUoHOUgwu\n" \
  "X4Po1QYz+3dszkDqMp4fklxBwXRsW10KXzPMTZ+sOPAveyxindmjkW8lGy+QsRlG\n" \
  "PfZ+G6Z6h7mjem0Y+iWlkYcV4PIWL1iwBi8saCbGS5jN2p8M+X+Q7UNKEkROb3N6\n" \
  "KOqkqm57TH2H3eDJAkSnh6/DNFu0Qg==\n" \
  "-----END CERTIFICATE-----";

const char* foo_4000 = "01234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678 [... cut ...]
    Serial.println("");

    if (WiFi.status() == WL_CONNECTED) {
      debugE("WIFI_CONNECT_OK"); 
    }
    else {
      debugE("WIFI_CONNECT_ERR"); 
    }
  }
}

void mqttOnSubEvent(char* topic, byte* payload, unsigned int length) {
  Serial.println("");
  for (int i = 0; i < length; i++)
  {
    Serial.print(payload[i]);
  }
}`

And just running it for a few min utes you can see that the MQTT has to reconnect over & over to get the job done.

Crying for HELP here, as I am in a tight corner here.

Cheers evryone !

macedolfm, 2019-01-28:
OK, to be fare, the WiFiClient also did NOT perform perfectly, more or less it is as realiable as the WiFiClientSecure, using the above example and replacing:

#include <WiFiClientSecure.h> => #include <WiFiClient.h> 
and
WiFiClientSecure wifi_client; => WiFiClient wifi_client;

I still get lots of errors ans some success in publishing to Mosquitto, so I gotta be doing something VERY wrong:

```
(E p:^10001 mqttConnect C1) WIFI_CONNECTING
(E p:^6119 mqttConnect C1) WIFI_CONNECT_OK
(E p:^3882 mqttConnect C1) MQTT_CONNECTING
(E p:^16150 mqttConnect C1) MQTT_CONNECT_OK
(E p:^0004 loop C1) MQTT_PUBLISHING
(E p:^0002 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0001 loop C1) MQTT_PUB_OK
(E p:^10000 loop C1) MQTT_PUBLISHING
(E p:^10001 loop C1) MQTT_PUB_ERR
(E p:^0001 mqttConnect C1) MQTT_CONNECTING
(E p:^6039 mqttConnect C1) MQTT_CONNECT_OK
(E p:^3964 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0001 loop C1) MQTT_PUB_OK
(E p:^11989 loop C1) MQTT_PUBLISHING
(E p:^10000 loop C1) MQTT_PUB_ERR
(E p:^0001 mqttConnect C1) MQTT_CONNECTING
(E p:^4285 mqttConnect C1) MQTT_CONNECT_OK
(E p:^5718 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^12993 loop C1) MQTT_PUBLISHING
(E p:^4154 loop C1) MQTT_PUB_OK
(E p:^5843 mqttConnect C1) MQTT_CONNECTING
(E p:^5828 mqttConnect C1) MQTT_CONNECT_OK
(E p:^4175 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9997 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0001 loop C1) MQTT_PUB_OK
(E p:^9999 loop C1) MQTT_PUBLISHING
(E p:^10001 loop C1) MQTT_PUB_ERR
(E p:^0002 mqttConnect C1) MQTT_CONNECTING
(E p:^3991 mqttConnect C1) MQTT_CONNECT_OK
(E p:^6012 loop C1) MQTT_PUBLISHING
(E p:^0002 loop C1) MQTT_PUB_OK
(E p:^9999 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0002 loop C1) MQTT_PUB_OK
(E p:^9999 loop C1) MQTT_PUBLISHING
(E p:^10001 loop C1) MQTT_PUB_ERR
(E p:^0001 mqttConnect C1) MQTT_CONNECTING
(E p:^18265 mqttConnect C1) MQTT_CONNECT_ERR
(E p:^0000 mqttConnect C1) MQTT_CONNECTING
(E p:^6718 mqttConnect C1) MQTT_CONNECT_OK
(E p:^3284 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9997 loop C1) MQTT_PUBLISHING
(E p:^0002 loop C1) MQTT_PUB_OK
(E p:^9999 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0002 loop C1) MQTT_PUB_OK
(E p:^9999 loop C1) MQTT_PUBLISHING
(E p:^0002 loop C1) MQTT_PUB_OK
(E p:^9999 loop C1) MQTT_PUBLISHING
(E p:^7008 loop C1) MQTT_PUB_OK
(E p:^2991 mqttConnect C1) MQTT_CONNECTING
(E p:^6022 mqttConnect C1) MQTT_CONNECT_OK
(E p:^3981 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
```

d-a-v, 2019-01-28:
You'd better use an esp8266 :-)

macedolfm, 2019-01-28:
> You'd better use an esp8266 :-)

Yes, I know, but I´m just in love with this thing [Heltec_ESP32_LoRa_BLE_OLED](http://www.heltec.cn/project/wifi-lora-32/?lang=en)

earlephilhower, 2019-01-28:
@macedolfm , what @d-a-v was trying to say is you're posting about ESP32 problems on the ESP8266 Arduino site.  Pop to https://github.com/espressif/arduino-esp32 if you want help, because we can't do anything for you here unless you get that ESP8266. :)

macedolfm, 2019-01-28:
> @macedolfm , what @d-a-v was trying to say is you're posting about ESP32 problems on the ESP8266 Arduino site. Pop to https://github.com/espressif/arduino-esp32 if you want help, because we can't do anything for you here unless you get that ESP8266. :)

Oooooopps, I am really, really sorry, did not realize that, after researching for so long this was the only discussion about the them that really matched my concerns, flipped over the fact that we are in the ESP8266 forum, sory, will delete my comments, thx

vishwasvogga, 2020-07-18:
Content-Length header shall be equal to the body length.