esp8266/Arduino #3682: MQTT stack overflow error 3

nealre, 2017-10-05:
Below is the error I get when connecting to MQTT.
WiFi connected
IP address: 
192.168.0.113
Connecting to MQTT... 
Exception (3):
epc1=0x4000bf3c epc2=0x00000000 epc3=0x00000000 excvaddr=0x4022fe94 depc=0x00000000

ctx: cont 
sp: 3ffefd00 end: 3ffeff40 offset: 01a0

>>>stack>>>
3ffefea0:  40104f6e 00000001 3ffe8765 3ffeef18  
3ffefeb0:  3fffdad0 3ffeec94 3ffeec94 4020331e  
3ffefec0:  3ffe8460 0000000d 3ffeeeec 40203779  
3ffefed0:  3ffe8764 3ffeed80 3ffeeeec 3ffeef18  
3ffefee0:  3fffdad0 3ffeec94 3ffeeeec 40201ec4  
3ffefef0:  7100a8c0 00000000 3ffeef10 40201f18  
3ffeff00:  3ffe83e0 3ffeed80 3ffeeeec 40201daf  
3ffeff10:  3ffe8760 7100a8c0 feefeffe 3ffeef18  
3ffeff20:  3fffdad0 00000000 3ffeef10 40203c00  
3ffeff30:  feefeffe feefeffe 3ffeef20 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld

_15-5-17_______________________________________________________
I appreciate any help you could give. I'm some what familiar with programming and stack operations but
this is above my head.

thanks,
Rick

----------------------------- Remove above -----------------------------
*********************************************Copy of the Code to connect the blinds to open and close.
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <WiFiClientSecure.h>
#include <WiFiServer.h>
#include <WiFiUdp.h>

/***************************************************
  Adafruit MQTT Library ESP8266 Example
  Must use ESP8266 Arduino from:
    https://github.com/esp8266/Arduino
  Works great with Adafruit's Huzzah ESP board:
  ----> https://www.adafruit.com/product/2471
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!
  Written by Tony DiCola for Adafruit Industries.
  MIT license, all text above must be included in any redistribution
 ****************************************************/
//#include <ESP8266WiFi.h>
#include <Servo.h>
#include "Adafruit_MQTT.h"
#include "Adafruit_MQTT_Client.h"
/************************* WiFi Access Point (CHANGE THIS) *********************************/
#define WLAN_SSID       "belkin.2ee"
#define WLAN_PASS       "ce963696"
/************************* MQTT Setup (CHANGE THIS) *********************************/
#define AIO_SERVER      "m12.cloudmqtt.com"
#define AIO_SERVERPORT  16726
#define AIO_USERNAME    "ha2"
#define AIO_KEY         "ha"
/************ Global State (Don't Change) ******************/
// Create an ESP8266 WiFiClient class to connect to the MQTT server.
WiFiClient client;
// Store the MQTT server, username, and password in flash memory.
// This is required for using the Adafruit MQTT library.
const char MQTT_SERVER[] PROGMEM    = AIO_SERVER;
const char MQTT_USERNAME[] PROGMEM  = AIO_USERNAME;
const char MQTT_PASSWORD[] PROGMEM  = AIO_KEY;
// Setup the MQTT client class by passing in the WiFi client and MQTT server and login details.
Adafruit_MQTT_Client mqtt(&client, MQTT_SERVER, AIO_SERVERPORT, MQTT_USERNAME, MQTT_PASSWORD);
/****************************** Feeds (CHANGE THESE)***************************************/
const char BLINDSTATUS[] PROGMEM = AIO_USERNAME "/blinds/state";
Adafruit_MQTT_Publish blindstatuspub = Adafruit_MQTT_Publish(&mqtt, BLINDSTATUS);
const char BLINDCOMMAND[] PROGMEM = AIO_USERNAME "/blinds/command";
Adafruit_MQTT_Subscribe blindcommandsub = Adafruit_MQTT_Subscribe(&mqtt, BLINDCOMMAND);

/*************************** Sketch Code ************************************/
Servo myservo;
int state = 0;
int prevstate = 0;
int dirc = 0;
int spintime = 0;
int servoPin = D4; //CHANGE TO WHATEVETR PIN YOUR USING
//void MQTT_connect();

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  delay(10);
  Serial.println("Blind Startup Sequence");
  Serial.println(); Serial.println();
  Serial.print("Connecting to ");
  Serial.println(WLAN_SSID);
  WiFi.begin(WLAN_SSID, WLAN_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  Serial.println("WiFi connected");
  Serial.println("IP address: "); Serial.println(WiFi.localIP());
  mqtt.subscribe(&blindcommandsub);

WiFi.persistent(false);
}
void loop() {
 MQTT_connect();
  Adafruit_MQTT_Subscribe *subscription;
  while ((subscription = mqtt.readSubscription(1000))) {
    if (subscription == &blindcommandsub) {
      Serial.print(F("Receieved command: "));
      String blindcommand = (char *)blindcommandsub.lastread;
      Serial.print(blindcommand);
      if ( blindcommand == "open") {
        Serial.println("  433MHZ OPEN COMMAND RECEIVED");
        dirc = 180; // direction for servo to run
        spintime = 3700; // << CHANGE TO WHATEVER TIME YOU NEED TO OPEN YOUR BLINDS
        state = 1; //sets current state
      }
      else if ( blindcommand == "close" ) {
        Serial.println("  433MHZ CLOSE COMMAND RECEIVED");
        dirc = 0;
        spintime = 3300; // << CHANGE TO WHATEVER TIME YOU NEED TO CLOSE YOUR BLINDS
        state = 2; //sets current state
      }
      if (state != prevstate) {
        Serial.println("State change!");
        servo_move();
      }
      prevstate = state;
    }
    if (! mqtt.ping()) {
      mqtt.disconnect();
    }
    delay(500);
  }
}
void servo_move() {
  Serial.println("State Change. Rotating Servo");
  if ( dirc == 180) {
    blindstatuspub.publish("opened");
  }
  else if (dirc == 0) {
    blindstatuspub.publish("closed");
  }
  myservo.attach(servoPin);
  myservo.write(dirc);
  delay(spintime);
  myservo.detach();
  Serial.println("Returning to main loop");
  return;
}
void MQTT_connect() {
   Serial.println("Void MQTTroutine");
  int8_t ret;
  if (mqtt.connected()) {
    return;
  }
  Serial.print("Connecting to MQTT... ");// connect will return 0 for connected
  
  while ((ret = mqtt.connect()) != 0) { 
    Serial.println(mqtt.connectErrorString(ret));
    Serial.println("Retrying MQTT connection in 5 seconds...");
    mqtt.disconnect();
    delay(5000);  // wait 5 seconds
  }
  Serial.println("MQTT Connected!");
  }

devyte, 2017-10-06:
@nealre no basic info, no IDE settings. The issue template exists for a reason.
You are essentially requesting help with usage of a 3rd party lib: Adafruit's mqtt class. I don't see anything obvious in your code, so some observations and generic advice:
- there is a tool to decode the stack dump. You should use it, and look at the results to figure out exactly where the crash is happening.
- a run through the loop can't take too long to execute. You might want to do some timing (e.g.: with millis() ) to see if something is taking too long.
- the stack is limited on the ESP to about 4KB. If there is some large object declared locally within a function, it could blow up the stack (I don't see this in the code you provided).
- try a pin other than D4.
- make sure your power supply is good.
- your messages say rst cause:2, which means the reset pin.
- you should request help in a community forum, or the owner of the lib code.

Closing due to 3rd party lib usage.

closed 2017-10-06 by devyte

nealre, 2017-10-08:
Thank you for your help. As you can tell I'm a noobee. but , little by little i'm learning thanks to people like yourself. I'll try some of the suggestions you have made and see. The debug tool sounds good I'll search for one.
Rick