esp8266/Arduino #3684: sprintf_P seems to be broken

mamama1, 2017-10-06:
Stack trace (decoded):

```

Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
Decoding 46 results
0x4022c75b: ets_vsnprintf at ?? line ?
0x40243260: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4022c742: ets_vsnprintf at ?? line ?
0x40243260: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4010746c: vsnprintf at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/libc_replacements.c line 342
0x401004d8: malloc at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1664
0x4021358c: operator new[](unsigned int) at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/abi.cpp line 84
0x40213f51: vsnprintf_P(char*, unsigned int, char const*, __va_list_tag) at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/pgmspace.cpp line 146
0x40213f92: sprintf_P(char*, char const*, ...) at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/pgmspace.cpp line 146
0x40243260: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40242f0c: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40243250: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40207585: wsHandleNotFound() at C:\Users\user\AppData\Local\Temp\VMBuilds\myProject\esp8266_nodemcuv2\Debug/myProject.cpp line 3233
0x401040ec: lmacProcessTXStartData at ?? line ?
0x401040e9: lmacProcessTXStartData at ?? line ?
0x401030c6: wDev_ProcessFiq at ?? line ?
0x40102f1c: wDev_ProcessFiq at ?? line ?
0x401004ba: _umm_malloc at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1462
0x40101ea2: pp_post at ?? line ?
0x4010517f: lmacRxDone at ?? line ?
0x40102bb5: trc_NeedRTS at ?? line ?
0x401050ca: lmacTxFrame at ?? line ?
0x4010020c: _umm_free at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x40100668: _umm_realloc at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1629
:  (inlined by) realloc at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1709
0x4010020c: _umm_free at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4010053d: _umm_realloc at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1491
:  (inlined by) realloc at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1709
0x40234d92: pbuf_free at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/pbuf.c line 752
0x40234da9: pbuf_free at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/pbuf.c line 758
0x40215770: String::changeBuffer(unsigned int) at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x402157bf: String::reserve(unsigned int) at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x4010020c: _umm_free at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4010068c: free at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x40215714: String::~String() at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x40212501: ESP8266WebServer::_parseRequest(WiFiClient&) at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/Parsing.cpp line 239
0x40215de2: std::_Function_handler ::_M_invoke(std::_Any_data const&) at c:\users\user\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x40210f2a: std::function ::operator()() const at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x4021100e: ESP8266WebServer::_handleRequest() at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x40210254: FunctionRequestHandler::canHandle(HTTPMethod, String) at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x4020c2b2: processTCPServer() at C:\Users\user\AppData\Local\Temp\VMBuilds\myProject\esp8266_nodemcuv2\Debug/myProject.cpp line 1116
0x40211233: ESP8266WebServer::handleClient() at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x402095b0: processIRRemote() at C:\Users\user\AppData\Local\Temp\VMBuilds\myProject\esp8266_nodemcuv2\Debug/myProject.cpp line 1799
0x4020cc9a: loop at C:\Users\user\AppData\Local\Temp\VMBuilds\myProject\esp8266_nodemcuv2\Debug/myProject.cpp line 1032
0x402136b4: loop_wrapper at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\user\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109

```

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			NodeMCU 1.0
Core Version:      	2.3.0

### Description

The use of 
`
char httpBuffer[1481];
sprintf_P(httpBuffer, PSTR("%s%s%s"), html_top, "<div style=\"padding-left: 10px;\"><h1>404 URL not found</div>", html_bottom);
`
causes the above exception.
html_top and html_bottom are globally defined "static const char html_top[] PROGMEM".

When I put the string together using strcat_P and strcopy_P, there is no problem at all.
I've also found this: https://www.bountysource.com/issues/43327406-sprintf_p-broken
maybe it is related.

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		1MB
CPU Frequency:		80Mhz
Flash Mode:			?
Flash Frequency:	?
Upload Using:		OTA / SERIAL (both same result)
Reset Method:		?

earlephilhower, 2017-10-06:
sprintf_p only handles PROGMEM format strings, not parameters.  This crash is the expected behavior...

mamama1, 2017-10-06:
I'm sorry, I'm not familiar enough with programming to correctly interpret what you mean.
Could you please explain a bit more detailled what the issue is?

What in my code snipped is considered a parameter? PSTR? How would I make a progmem format string? Just like the other strings I've used in this command?

According to [https://github.com/esp8266/Arduino/blob/master/doc/PROGMEM.rst](url) PSTR should work at least with printf_P...

Thanks!

mamama1, 2017-10-06:
oh my god, the solution is, to use %S instead of %s in the format string.

%S is similar to the s format, except the pointer is expected to point to a program-
memory (ROM) string instead of a RAM string.

earlephilhower, 2017-10-06:
D'oh, sorry for the wrong information!  Glad you figured it out.

mamama1, 2017-10-06:
no worries, thanks for trying to help!

mamama1, 2017-10-07:
ok not solved yet. it's not crashing anymore, but everything that's contained in the buffer after my operation, is %S%S%S so it seems that simply it is not done what had to be done.

Code is now:
```

char httpBuffer[1481];
sprintf_P(httpBuffer, PSTR("%S%S%S"), html_top, "<div style=\"padding-left: 10px;\"><h1>404 URL not found</div>", html_bottom);

```

**httpBuffer contains "%S%S%S".**

html_top and html_bottom are still globally defined "static const char variableName[] PROGMEM".

When trying the same with decimals, like PSTR("%d") as format string and 3 or whatever number as a parameter, the buffer contains the correct data.

When trying this code, with is basically the same, with an Arduino UNO, the buffer contains the expected result:
```
static const char header_1[] PROGMEM = "Dissolved O2";
char buffer[100];

void setup() {
   Serial.begin(57600);
   sprintf_P(buffer,PSTR( "The column header is %S") ,header_1);
   Serial.println(buffer);
}
void loop(){}
```

Does the ESP implementation ignore %S?

mamama1, 2017-10-07:
clue anyone?
thanks!

Makuna, 2017-10-07:
As stated, on Esp8266, the arguments can not be PROGMEM strings; only the format string can be PROGMEM.  The capital s is not supported.

You really shouldn't be using printf to concat strings like your example is doing anyway.  use strcat_P().

mamama1, 2017-10-07:
OK thanks. Is this undocumented because it is an AVR speciality anyway?
I've searched the whole internet :D and didn't find anything on this...

I wanted to concat multiple strings, without calling strcat multiple times. Somewhere I did read that printf will have a smaller performance impact for this task than strcat...

Anyways, issue closed, thanks again.

closed 2017-10-07 by mamama1