esp8266/Arduino #3685: WiFiClient.h : missing documentation about a function to refresh website immediately

tofrnr, 2017-10-07:
abstract:
a documentation is missing about a function to refresh website immediately.

status quo:
Currently my website is refreshed automatically each 20 sec and if a button on the website was pressed.

Information is missing how to force a refresh at certain events during runtime  (e.g., a switch button connected to a digital pin was pressed manually or an analog sensor value has reached a critical value), in order to show it up immediately in the www., not only after 20 sec. auto refresh 

my code excerpts:
```
#include <ESP8266WiFi.h>
WiFiServer       wifiserver(80);
#define PIN_OUT1  D6
int8_t  OUT1=0;

void handleWebsite(){      
  
    WiFiClient client = wifiserver.available();
     // Check if a client has connected     
     if (!client) {
        return;
     }     
     // Read the first line of the request
     String request = client.readStringUntil('\r');
     Serial.println(request);
     client.flush();

     // Match the request
     if ((request.indexOf("/OUT1=ON") != -1)  ) { 
       digitalWrite(PIN_OUT1, HIGH);  
       OUT1 = HIGH;
     }
     if ((request.indexOf("/OUT1=OFF") != -1) ) {
       digitalWrite(PIN_OUT1, LOW);
       OUT1 = LOW;
     }

     // **SNIP**
     String script="";
     script+= ("HTTP/1.1 200 OK \n");
     script+= ("Content-Type: text/html \n");
     script+= ("\n"); //  do not forget this one
     script+= ("<!DOCTYPE html> \n");
     script+= ("<html> \n");  
     script+= ("<head> \n");
     // refresh each 20 sec.
     script+= ("<meta http-equiv=\"refresh\" content=\"20; URL=http://192.168.2.117\"> \n");
    // **SNIP**
     script+= (" <h2>" + OUT1name + " is now: "); 
     if(OUT1 == HIGH) 
        { script+= ("&nbsp;<wbr>On &nbsp; <wbr> <wbr> "); } 
       else 
        { script+= ("Off &nbsp; <wbr> <wbr> ");  }  
     script+= ("<a href=\" /OUT1=ON\"\"> <button style=\"height:70px;width:140px\" > Turn On </button></a>  ");
     script+= ("<a href=\" /OUT1=OFF\"\"> <button style=\"height:70px;width:140px\" > Turn Off </button></a> <br /> ");      
     script+= ("<br> \n");
    
    // **SNIP
}

```

devyte, 2017-10-07:
@tofrnr you can't initiate actions on the side of http server. This is due to the nature of http. If you need server-side actions, look into websockets.
Closing due to off topic.

closed 2017-10-07 by devyte

tofrnr, 2017-10-07:
ok, thanks, I missed that point.
So is there a ESP8266WiFi lib extension which provides a ESP8266websocket client or server?

vdeconinck, 2017-10-07:
http://lmgtfy.com/?q=esp8266+websocket

tofrnr, 2017-10-08:
no, I didn't mean vague Google search results of questionable 3rd party libs, I meant a reliable tested lib extension here in this esp8266/Arduino repository, addionally to all the libs already liste here:
https://github.com/esp8266/Arduino/tree/master/libraries

vdeconinck, 2017-10-08:
This ticket has been closed because this is not an issue with the core, If you need further assistance, please direct your request to esp8266.com or stackoverlow.

tofrnr, 2017-10-09:
if this functionality not availble yet, then it's an improvement request.
@ developers:
Shall I open a new issue for this improvement request?

igrr, 2017-10-09:
Please have a look at https://github.com/Links2004/arduinoWebSockets. It is not part of this repository because it also supports [other hardware](https://github.com/Links2004/arduinoWebSockets#supported-hardware). It can be installed using Arduino Library Manager.

tofrnr, 2017-10-09:
thank you very much, that is really appreciated, nonetheless I must say that I need to have a lib which is 100% compliant to the lib and the code I osted above using
#include <ESP8266WiFi.h>
WiFiServer       wifiserver(80); 
just providing an extension to send additionally a "immediate reload website" command in case of important sensor events detected at the server's proprietary IOs :

`#include <ESP8266WiFi.h>
WiFiServer       wifiserver(80);
#define PIN_OUT1  D6
int8_t  OUT1=0;

void handleWebsite(){      
  
    WiFiClient client = wifiserver.available();
     // Check if a client has connected     
     if (!client) {
        return;
     }     
     // Read the first line of the request
     String request = client.readStringUntil('\r');
     Serial.println(request);
     client.flush();

     // Match the request
     if ((request.indexOf("/OUT1=ON") != -1)  ) { 
       digitalWrite(PIN_OUT1, HIGH);  
       OUT1 = HIGH;
     }
     if ((request.indexOf("/OUT1=OFF") != -1) ) {
       digitalWrite(PIN_OUT1, LOW);
       OUT1 = LOW;
     }

     // **SNIP**
     String script="";
     script+= ("HTTP/1.1 200 OK \n");
     script+= ("Content-Type: text/html \n");
     script+= ("\n"); //  do not forget this one
     script+= ("<!DOCTYPE html> \n");
     script+= ("<html> \n");  
     script+= ("<head> \n");
     // refresh each 20 sec.
     script+= ("<meta http-equiv=\"refresh\" content=\"20; URL=http://192.168.2.117\"> \n");
    // **SNIP**
     script+= (" <h2>" + OUT1name + " is now: "); 
     if(OUT1 == HIGH) 
        { script+= ("&nbsp;<wbr>On &nbsp; <wbr> <wbr> "); } 
       else 
        { script+= ("Off &nbsp; <wbr> <wbr> ");  }  
     script+= ("<a href=\" /OUT1=ON\"\"> <button style=\"height:70px;width:140px\" > Turn On </button></a>  ");
     script+= ("<a href=\" /OUT1=OFF\"\"> <button style=\"height:70px;width:140px\" > Turn Off </button></a> <br /> ");      
     script+= ("<br> \n");
    
    // **SNIP
    client.print(script);
}`

igrr, 2017-10-09:
As mentioned above, it is unlikely that something like this will be implemented. The way your sketch is written, at the point when you want to initiate page refresh from the server side, no TCP connection between client and server exists. Using websockets can solve this, but the structure of the sketch will be very different (and client side implementation will be significantly more complex than what you have).

I suggest you to try prototyping this design on the PC first, without ESP8266 involved. Write the server in Python or node.js or other language of your choice, make it trigger refresh when e.g. a key is pressed on the console. This will help you understand client-server interaction in an environment which is easy to understand and debug. Once you come up with the architecture which works, you can port the server code to the esp8266. Good luck!

tofrnr, 2017-10-09:
OMG, I have no idea about how to write web server and/or  client code, I am just a beginner to ESP8266WiFi and all that, and all I can do is copy and paste code example snippets which already work. If I could program all that in real C++, I wouldn't need Arduinish at all, but use just plain GCC.

devyte, 2017-10-09:
@tofrnr this is an issue tracker, meant to track issues in the core code hosted in this repo. This is not a forum to ask for help on topics on which you are a beginner. Please request help at a community forum like esp8266.com or stackoverflow. Also, I have already directed you towards what you need to research to accomplish what you need: websockets.
The issue is already closed per issue policy #3655 . There have been 5 off topic replies since then. Locking the conversation.