esp8266/Arduino #3687: Exception (28): epc1=0x4000dfc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

hemangjoshi37a, 2017-10-07:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	I DONT KNOW

### Description

When I send a data from device to wifi clients it crashes giving the error "
Exception (28):
epc1=0x4000dfc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
" but when esp8266 is receiving data from the clients, It works fine.

### Settings in IDE

Module:  			ESP8266 Module ESP12
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch
```
#include <ESP8266WiFi.h>

//how many clients should be able to telnet to this ESP8266
#define MAX_SRV_CLIENTS 1

const char *ssid = "Tenda_5F19A8";
const char *password = "bhumika_301";

WiFiServer server(23);
WiFiClient serverClients[MAX_SRV_CLIENTS];

byte a, alst;
String astr;
unsigned int count = 1;

void setup() {
  a = 0;
  alst = 1;
  pinMode(A0, INPUT);
  Serial1.begin(115200);
  //Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial1.print("\nConnecting to "); Serial1.println(ssid);
  uint8_t i = 0;
  while (WiFi.status() != WL_CONNECTED && i++ < 20) delay(500);
  if (i == 21) {
    Serial1.print("Could not connect to"); Serial1.println(ssid);
    while (1) delay(500);
  }
  //start UART and the server
  Serial.begin(115200);
  server.begin();
  server.setNoDelay(true);

  Serial1.print("Ready! Use 'telnet ");
  Serial1.print(WiFi.localIP());
  Serial1.println(" 23' to connect");

  a = (byte)analogRead(A0);
  astr = String(a);
}

void loop() {
  uint8_t i;
  //check if there are any new clients
  if (server.hasClient()) {
    for (i = 0; i < MAX_SRV_CLIENTS; i++) {
      //find free/disconnected spot
      if (!serverClients[i] || !serverClients[i].connected()) {
        if (serverClients[i]) serverClients[i].stop();
        serverClients[i] = server.available();
        Serial1.print("New client: "); Serial1.print(i);
        continue;
      }
    }
    //no free/disconnected spot so reject
    WiFiClient serverClient = server.available();
    serverClient.stop();
  }
  /////////////////////////////////////////////////////////////////////////////////////
  //check clients for data
  for (i = 0; i < MAX_SRV_CLIENTS; i++) {
    if (serverClients[i] && serverClients[i].connected()) {
      if (serverClients[i].available()) {
        //get data from the telnet client and push it to the UART
        while (serverClients[i].available()) Serial.write(serverClients[i].read());
      }
    }
  }
  ///////////////////////////////////////////////////////////////////////////////////
  //check UART for data
  if (Serial.available()) {
    size_t len = Serial.available();
    uint8_t sbuf[len];
    Serial.readBytes(sbuf, len);
    //push UART data to all connected telnet clients
    for (i = 0; i < MAX_SRV_CLIENTS; i++) {
      if (serverClients[i] && serverClients[i].connected()) {
        serverClients[i].write(sbuf, len);
        delay(1);
      }
    }
  }

  //////////////////////////////////////////////////////////////////////////////////////

  if (Serial.available()) {
    size_t len = Serial.available();
    uint8_t sbuf[len];
    Serial.readBytes(sbuf, len);

    a = (byte)analogRead(A0);

 //   if (alst != a)
    {

      astr = String(a);
      size_t len = astr.length();
      len = 1;
      uint8_t aread[len];
      //push UART data to all connected telnet clients
      for (i = 0; i < MAX_SRV_CLIENTS; i++) {
        if (serverClients[i] && serverClients[i].connected()) {
          serverClients[i].write((uint8_t *)a, len);
          delay(100);
        }
      }
      alst = a;
      //delay(1);
    }
    count++;
  }
}
```
### Serial Messages

```
Exception (28):
epc1=0x4000dfc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffef2d0 end: 3ffef570 offset: 01a0

>>>stack>>>
3ffef470:  3fff023c 00000000 00000001 00000000  
3ffef480:  402230ca 3fff03d4 00000000 402230a0  
3ffef490:  00000000 00000001 00000000 00000002  
3ffef4a0:  00000001 00000000 00000000 3fff06fc  
3ffef4b0:  00000001 000005b3 00000000 4020335d  
3ffef4c0:  3ffee4fc 000000df 000000df 3ffef510  
3ffef4d0:  3ffef510 00000001 3fff04bc 402028ae  
3ffef4e0:  3ffef510 3ffee3ed 3ffee3f0 402028f0  
3ffef4f0:  00000000 3ffee51c 3ffee3f0 40202027  
3ffef500:  55545559 3ffe0a0d 3ffee3f0 40201f72  
3ffef510:  00000000 00000000 00000000 40203190  
3ffef520:  00000000 00000000 3ffee500 40201e00  
3ffef530:  3ffef510 00000000 00000000 feefeffe  
3ffef540:  00000000 00000000 00000001 3ffee548  
3ffef550:  3fffdad0 00000000 3ffee540 402034b0  
3ffef560:  feefeffe feefeffe 3ffee550 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
```
Please help me. I am stuck with this problem from days. I have read somewhere that using core version 2.3.0 solves this issue but i don't know where to download it and replace with existing version.

igrr, 2017-10-07:
Please use EspExceptionDecoder to decode that stack trace.

hemangjoshi37a, 2017-10-09:
Okay sir,
Thank you for help. It worked. Now I have decoded the stack looks like this 👍 
Decoding 12 results

```
0x40203490: loop_wrapper at C:\Users\aakash\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x402230ca: tcp_write at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 575
0x402230a0: tcp_pbuf_prealloc at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 257
:  (inlined by) tcp_write at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 535
0x4020335d: String::operator=(char const*) at C:\Users\aakash\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 66
0x402028ae: ClientContext::write(char const*, unsigned int) at C:\Users\aakash\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x402028f0: WiFiClient::write(unsigned char const*, unsigned int) at C:\Users\aakash\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x40202027: loop at C:\Users\aakash\Documents\Energia\WiFiTelnetToSerial\WiFiTelnetToSerial/WiFiTelnetToSerial.ino line 105
0x40201f72: loop at C:\Users\aakash\Documents\Energia\WiFiTelnetToSerial\WiFiTelnetToSerial/WiFiTelnetToSerial.ino line 88
0x40203190: String::~String() at C:\Users\aakash\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 66
0x40201e00: setup at C:\Users\aakash\Documents\Energia\WiFiTelnetToSerial\WiFiTelnetToSerial/WiFiTelnetToSerial.ino line 40
0x402034b0: loop_wrapper at C:\Users\aakash\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\aakash\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109
0x40203490: loop_wrapper at C:\Users\aakash\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x402230ca: tcp_write at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 575
0x402230a0: tcp_pbuf_prealloc at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 257
:  (inlined by) tcp_write at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 535
0x4020335d: String::operator=(char const*) at C:\Users\aakash\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 66
0x402028ae: ClientContext::write(char const*, unsigned int) at C:\Users\aakash\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x402028f0: WiFiClient::write(unsigned char const*, unsigned int) at C:\Users\aakash\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x40202027: loop at C:\Users\aakash\Documents\Energia\WiFiTelnetToSerial\WiFiTelnetToSerial/WiFiTelnetToSerial.ino line 105
0x40201f72: loop at C:\Users\aakash\Documents\Energia\WiFiTelnetToSerial\WiFiTelnetToSerial/WiFiTelnetToSerial.ino line 88
0x40203190: String::~String() at C:\Users\aakash\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 66
0x40201e00: setup at C:\Users\aakash\Documents\Energia\WiFiTelnetToSerial\WiFiTelnetToSerial/WiFiTelnetToSerial.ino line 40
0x402034b0: loop_wrapper at C:\Users\aakash\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\aakash\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109
```
Can i solve the error using this decoding??

igrr, 2017-10-09:
`serverClients[i].write((uint8_t *)a, len);`

You are casting a value obtained from `analogRead` to a pointer, are you sure that is what you want?

hemangjoshi37a, 2017-10-09:
ohh okay. Actually I want to send my analog reading to telnet serial port. So have written it. Is it wrong thing to do? 
My analog reading is firlstly stored in variable a which is integer but I have to convert it into some kind of array like uint8_t type datatype. So is it wrong thing to do?

devyte, 2017-10-10:
@hemangjoshi37a I think this is wrong:
`serverClients[i].write((uint8_t *)a, len);
`
most likely what you meant to do is:
`serverClients[i].write(&a, len);
`

Please review your code, and debug carefully.
Closing as user error.

closed 2017-10-10 by devyte

hemangjoshi37a, 2017-10-10:
Please sir help me. I am stuck into this for 10-15 days.

hemangjoshi37a, 2017-10-10:
@devyte Thank you for help. I will try it and let you know if it works.

hemangjoshi37a, 2017-10-10:
@devyte yo man. It is working but it is giving weird character on the receiving side(on client side). Please take a look at the screenshot. Link : http://hemangjoshi.blogspot.in/2017/10/sent-and-received-data.html The square looking weird characters are sent data from esp.

And my code looks something like this : 
```
#include <ESP8266WiFi.h>

//how many clients should be able to telnet to this ESP8266
#define MAX_SRV_CLIENTS 1

const char *ssid = "Tenda_5F19A8";
const char *password = "bhumika_301";

WiFiServer server(23);
WiFiClient serverClients[MAX_SRV_CLIENTS];

byte a, alst;
String astr;
unsigned int count = 1;

void setup() {
  a = 0;
  alst = 1;
  pinMode(A0, INPUT);
  Serial1.begin(115200);
  //Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial1.print("\nConnecting to "); Serial1.println(ssid);
  uint8_t i = 0;
  while (WiFi.status() != WL_CONNECTED && i++ < 20) delay(500);
  if (i == 21) {
    Serial1.print("Could not connect to"); Serial1.println(ssid);
    while (1) delay(500);
  }
  //start UART and the server
  Serial.begin(115200);
  server.begin();
  server.setNoDelay(true);

  Serial1.print("Ready! Use 'telnet ");
  Serial1.print(WiFi.localIP());
  Serial1.println(" 23' to connect");

  a = (byte)analogRead(A0);
  astr = String(a);
}

void loop() {
  Serial.println("ok");
  uint8_t i;
  //check if there are any new clients
  if (server.hasClient()) {
    for (i = 0; i < MAX_SRV_CLIENTS; i++) {
      //find free/disconnected spot
      if (!serverClients[i] || !serverClients[i].connected()) {
        if (serverClients[i]) serverClients[i].stop();
        serverClients[i] = server.available();
        Serial1.print("New client: "); Serial1.print(i);
        continue;
      }
    }
    //no free/disconnected spot so reject
    WiFiClient serverClient = server.available();
    serverClient.stop();
  }
  /////////////////////////////////////////////////////////////////////////////////////
  //check clients for data
  for (i = 0; i < MAX_SRV_CLIENTS; i++) {
    if (serverClients[i] && serverClients[i].connected()) {
      if (serverClients[i].available()) {
        //get data from the telnet client and push it to the UART
        while (serverClients[i].available()) Serial.write(serverClients[i].read());
      }
    }
  }
  ///////////////////////////////////////////////////////////////////////////////////
  //check UART for data
  if (Serial.available()) {
    size_t len = Serial.available();
    uint8_t sbuf[len];
    Serial.readBytes(sbuf, len);
    //push UART data to all connected telnet clients
    for (i = 0; i < MAX_SRV_CLIENTS; i++) {
      if (serverClients[i] && serverClients[i].connected()) {
        serverClients[i].write(sbuf, len);
        delay(1);
      }
    }
  }

  //////////////////////////////////////////////////////////////////////////////////////

  //  size_t len = astr.length();
  //  uint8_t sbuf[len];
  //  Serial.readBytes(sbuf, len);
  //  a = (byte)analogRead(A0);
  //  astr = String(a);
  //  len = astr.length();
  //  // len = 1;
  //  uint8_t aread[len];
  //  //push UART data to all connected telnet clients
  //  for (i = 0; i < MAX_SRV_CLIENTS; i++) {
  //    if (serverClients[i] && serverClients[i].connected()) {
  //      serverClients[i].write(&a, len);
  //      delay(100);
  //    }
  //f
  //  alst = a;
  //  delay(1);
  ///////////////////////////////////////////////////////////////////////

  a = analogRead(A0);
  size_t len = 5;
  //push UART data to all connected telnet clients
  for (i = 0; i < MAX_SRV_CLIENTS; i++) {

    serverClients[i].write(&a, len);
    Serial.println(a);
    delay(100);
    delay(1);

  }
  // Serial.println(a);
  Serial.println("End");
  //  count++;

}
```

devyte, 2017-10-10:
@hemangjoshi37a you are reading the analog value as binary, and sending it directly as binary to the clients. Then, you are looking at it with Putty, who doesn't understand binary.
Most likely what you need is to convert the analog value to String, then send the string over, along the lines of:
```
String astr(a);
serverClients[i].write(astr.c_str(), astr.length());
```
Please discuss further at a forum, this is an issue tracker, not a forum for requesting help for your code.

hemangjoshi37a, 2017-10-10:
ohh my god I never thought that would be the case. Let me try and I will let you know.

hemangjoshi37a, 2017-10-10:
@devyte Awesome .... It works just fine... Please let me know anytime if I can help you in anyway...
Thank you...

alexgrauer, 2018-07-30:
Hi All, sorry to revive this thread.
Im getting the same exception, and the Decoder is returning me the following:

Decoding stack results
0x4020c432: String::String(char const*) at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/WString.cpp line 36
0x4020c4e6: String::operator=(String&&) at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/WString.cpp line 228
0x4020899b: SIM5320_SendToServer(unsigned char, unsigned short) at /Users/myuser/Downloads/NODEMCU_060818_2.0_GSM/NODEMCU_060818_2.0_GSM.ino line 1308
0x402013b6: timer1_attachInterrupt at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_timer.c line 51
0x401070a6: NODEMCU_External_Interrupt_Handle() at /Users/myuser/Downloads/NODEMCU_060818_2.0_GSM/NODEMCU_060818_2.0_GSM.ino line 607
0x40107588: interrupt_handler at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_wiring_digital.c line 135
0x40107572: interrupt_handler at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_wiring_digital.c line 131
0x40107134: onTimerISR() at /Users/myuser/Downloads/NODEMCU_060818_2.0_GSM/NODEMCU_060818_2.0_GSM.ino line 589
0x401074ec: interrupt_handler at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_wiring_digital.c line 113
0x40107360: timer1_isr_handler at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_timer.c line 33
0x401073a8: timer1_isr_handler at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_timer.c line 40
0x4020a01e: SPIClass::transferBytes_(unsigned char*, unsigned char*, unsigned char) at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/SPI/SPI.cpp line 458
0x4020a082: SPIClass::transferBytes_(unsigned char*, unsigned char*, unsigned char) at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/SPI/SPI.cpp line 476
0x40107360: timer1_isr_handler at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_timer.c line 33
0x4010020c: _umm_free at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1287
0x402013b6: timer1_attachInterrupt at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_timer.c line 51
0x401004d8: malloc at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1664
0x401004d8: malloc at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1664
0x40107360: timer1_isr_handler at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_timer.c line 33
0x401073a8: timer1_isr_handler at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_timer.c line 40
0x4020a7ab: spiRec() at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/SD/src/utility/Sd2Card.cpp line 52
0x4020a7ab: spiRec() at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/SD/src/utility/Sd2Card.cpp line 52
0x402013b6: timer1_attachInterrupt at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_timer.c line 51
0x401004d8: malloc at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1664
0x401004d8: malloc at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1664
0x40208f3c: loop() at /Users/myuser/Downloads/NODEMCU_060818_2.0_GSM/NODEMCU_060818_2.0_GSM.ino line 730
0x40208eeb: loop() at /Users/myuser/Downloads/NODEMCU_060818_2.0_GSM/NODEMCU_060818_2.0_GSM.ino line 718
0x4020d0a8: Print::write(unsigned char const*, unsigned int) at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Print.cpp line 38
0x4020bee9: Print::write(char const*) at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Print.h line 60
0x4020bee9: Print::write(char const*) at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Print.h line 60
0x4020bd12: HardwareSerial::available() at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/HardwareSerial.cpp line 118
0x4020468a: NODEMCU_checkSerial() at /Users/myuser/Downloads/NODEMCU_060818_2.0_GSM/NODEMCU_060818_2.0_GSM.ino line 973
0x40204e70: NODEMCU_Software_Serial_Print_New_Data() at /Users/myuser/Downloads/NODEMCU_060818_2.0_GSM/NODEMCU_060818_2.0_GSM.ino line 970
0x4020920c: loop() at /Users/myuser/Downloads/NODEMCU_060818_2.0_GSM/NODEMCU_060818_2.0_GSM.ino line 855
0x4020bfcc: Print::print(char const*) at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Print.cpp line 84
0x4020cc64: loop_wrapper() at /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp line 123

The code is running fine for more than 3 hours before it crashes.
I don't really understand whats going on.

If anyone could help me, or point me to the right direction, that would be great.
Thanks :)

d-a-v, 2018-07-30:
This looks like OOM.
* There are debug options you can enable (specifically OOM)
* Print and examine ESP.getFreeHeap() in your main loop
* This is a closed thread, with unrelated issue.
Please debug your application and *open a fresh new issue* if you happen to find a bug in the core.

alexgrauer, 2018-07-30:
Thanks @d-a-v, and sorry for writing in a closed thread. It was the same exception I was getting so I thought it was ok to write here.
ESP.getFreeHeap helped me find out that there's a function reducing the memory every time it runs (once a minute). I guess after 3 hours im running out of memory, and therefore crashing.
Thanks very much

sandevid, 2019-03-11:
### How to fix this , i use nodeMCU 

_Source Code :_ 

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <TelegramBot.h>

#define LED 14 //D3
 
const char* ssid = "Telegram";
const char* password = "getupdate";
const char BotToken[] = "660345429:ACGlSKhCRMmgKHNJH4pG3ut_1R_66flenqQ";

WiFiClientSecure net_ssl;
TelegramBot bot (BotToken, net_ssl);

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
     delay(500);
     Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println(WiFi.localIP());

  bot.begin(); 
    
  pinMode(LED, OUTPUT); 
}

void loop() {

  message m = bot.getUpdates(); 
  
  if (m.text.equals("on")){  
    digitalWrite(LED, HIGH);   
    bot.sendMessage(m.chat_id, "The Led 1 is now ON");  
    
  }else if (m.text.equals("off")){  
    digitalWrite(LED, LOW);   
    bot.sendMessage(m.chat_id, "The Led 1 is now OFF"); 
  }
}`

1. Board : NodeMCU 1.0 (ESP-12E Module)
2. Upload Speed : 115200
3. Flash Size : 4M (no-SPIFFS)
4. Debug Port : Disabled
5. Debug level : None
6. IwIP Variant : v2 lower memoty
7. VTables : Flash
8. Exceptions : Disable
9. Erase Flash : Only Sketch
10. Port : COM4

> Please help, 
> I have searched everywhere but there is no right solution

ariehoffman, 2020-02-06:
I am suck with the same problem- any idea?

Erik Hoffman

PanwarParamveer, 2020-05-09:
ESP.eraseConfig(); <--- use this before begin
  WiFi.begin("", "");

joctorres13, 2020-10-18:
Try a different flash_mode on your platform.ini file, by default it is "qio". Try dout mode:

`board_build.flash_mode = dout`

Here are some references:
[](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#more-options)
[](https://github.com/espressif/esptool/wiki/SPI-Flash-Modes)

Kos-M, 2020-12-27:
https://links2004.github.io/Arduino/dc/deb/md_esp8266_doc_exception_causes.html