esp8266/Arduino #3688: How to enable debuggin on NodeMCU 1.0 (ESP-12E Module) and see result of ssl handshake

bscheshir, 2017-10-07:
In instruction
>for better debug messages:
>https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

say about enable debuggin from Arduino tools menu.

But if I set module to "NodeMCU 1.0 (ESP-12E Module)" **I can't see debuggin settings**

I can enable debuggin from `#define` some constants or any else way? (yes)

**How about add this methid in instruction?**

#### Hardware
Hardware:			NodeMCU v3 (ESP-12N)
Core Version:      	2.3.0-rc2

### Description

I want to see more info about ssl handshake.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
CPU Frequency:		80Mhz
Flash Size: 		4M(3M SPIFFS)

### Sketch

```cpp

#define DEBUG_ESP_SSL

#include "FS.h"
#include <ESP8266WiFi.h>

WiFiClientSecure espClient;

...
void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.setDebugOutput(true);
...
  //Load cert, key, ca
  if (!SPIFFS.begin()) {
    Serial.println("Failed to mount file system");
    return;
  }
  // Load certificate file
  // $ openssl x509 -in client01.crt -out client01.crt.der -outform DER 
  File cert = SPIFFS.open("/client01.crt.der", "r"); //replace client01.crt.der eith your uploaded file name
  if (!cert) {
    Serial.println("Failed to open cert file");
  }
  else
    Serial.println("Success to open cert file");
  delay(1000);
  if (espClient.loadCertificate(cert))
    Serial.println("cert loaded");
  else
    Serial.println("cert not loaded");

  // Load private key file
  // $ openssl rsa -in client01.key -out client01.key.der -outform DER
  File privateKey = SPIFFS.open("/client01.key.der", "r"); //replace client01.key.der eith your uploaded file name
  if (!privateKey) {
    Serial.println("Failed to open private cert file");
  }
  else
    Serial.println("Success to open private cert file");
  delay(1000);
  if (espClient.loadPrivateKey(privateKey))
    Serial.println("private key loaded");
  else
    Serial.println("private key not loaded");
/*
  // Load CA file
  File ca = SPIFFS.open("/ca.crt.der", "r"); //replace ca.crt.der eith your uploaded file name
  if (!ca) {
      Serial.println("Failed to open ca ");
  }
  else
    Serial.println("Success to open ca");
  delay(1000);
  
  if(espClient.loadCACert(ca))
    Serial.println("ca loaded");
  else
    Serial.println("ca failed");
*/
  //connect to host
  Serial.print("connecting to ");
  Serial.println((String)host);
  if (!espClient.connect(host, httpsPort)) {
    Serial.println("connection failed");
    return;
  }

}

void loop() {

}
```

### Debug Messages

```
Connecting to HOMENET
scandone
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with HOMENET, channel 10
dhcp client start...
..ip:192.168.1.41,mask:255.255.255.0,gw:192.168.1.1
.
WiFi connected
Server started
192.168.1.41
Success to open cert file
cert loaded
Success to open private cert file
private key loaded
connecting to 192.168.1.39
please start sntp first !
connection failed

```

bscheshir, 2017-10-07:
So... How to see handshake log? sntp don't matter:
```
...
Setting time using SNTP
Sat Oct 07 21:43:10 2017

Success to open cert file
cert loaded
Success to open private cert file
private key loaded
connecting to 192.168.1.39
connection failed

```

igrr, 2017-10-07:
Try selecting "Generic ESP8266 Module" and adjust the settings for your
nodeMCU (flash size, 'nodemcu' reset method). Then you can enable debug
options.

bscheshir, 2017-10-08:
Thank you!
After change settings to
```
Плата: "Generic ESP8266 Module"
Flash Mode: "DIO"
Flash Frequency: "80MHz"
CPU Frequency: "160 MHz"
Flash Size: "4M (3M SPIFFS)"
Debug port: "Serial"
Debug Level: "Core + SSL + TLS Mem"
Reset Method: "nodemcu"
Upload Speed: "115200"
Порт: "COM5"
```

**AND upload sketch** I can see details output.

closed 2017-10-08 by bscheshir

Gopal433, 2019-02-19:
Hi , i want configure lolin node mcu wifi module in SDIO slave mode. I want access this SDIO slave interface from external sdio host. Please help, how to configure it in SDIO slave mode.