esp8266/Arduino #3689: .md -> .rst in "new issue" pattern

bscheshir, 2017-10-07:
> if you have a stack dump decode it:
>https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

>for better debug messages:
>https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

But actual instructions in `rst` 
[https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst](https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst)
[https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst](https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst)

devyte, 2017-10-10:
@bscheshir do you mean the links are wrong? or rather, the docs need to be converted to .md?

bscheshir, 2017-10-10:
@devyte see PR

PR #3704 (2017-10-15): md to rst
actual instructions in rst

fixed #3689
closed 2017-10-15 by igrr