esp8266/Arduino #3691: [Enhancement] Make Wire and SPI lib compatible between ESP32 and ESP8266

olikraus, 2017-10-08:
### Basic Infos
This is a feature extension request to make the ESP32 and ESP8266 library compatible with each other.
From my analysis i would suggest to modify ESP8266 library and align with ESP32 library. However other modifications are also thinkable. For more specific suggestions, see below.

#### Hardware
Hardware:			n.a.
Core Version:      	head

### Description

Problem description

Assigning movable pins for ESP32 and ESP8266 differs between the projects.

### Settings in IDE

n.a.

### Sketch
n.a.

### Debug Messages

n.a.

### Analysis of the Problem

SPI Lib ESP8266:
	Link: https://github.com/esp8266/Arduino/blob/master/libraries/SPI/SPI.h#L56
	Uses "pins" member function to assign movable pins:
	pins(int8_t sck, int8_t miso, int8_t mosi, int8_t ss);
	Note, that ss and mosi are required areguments

SPI Lib ESP32:
	Link: https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/src/SPI.h#L55
	The begin() member function accepts several (optional) arguments for
	the assignment of the pin numbers:
	"void begin(int8_t sck=SCK, int8_t miso=MISO, int8_t mosi=MOSI, int8_t ss=-1);"
	
Wire Lib ESP8266
	Link: https://github.com/esp8266/Arduino/blob/master/libraries/Wire/Wire.h#L53
	Has a "pins" member function which is declared as deprecated.
	"pins(int sda, int scl)"
	Instead use "begin(int sda, int scl);"
	
Wire Lib ESP32
	Link: https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.h#L55
	The ESP32 Wire lib extends the begin() member function with optional pin
	arguments and also a frequency paramenter:
	"void begin(int sda=-1, int scl=-1, uint32_t frequency=100000);"
	
### Conclusion

1. All four extensions are none-standard extentions on the Arduino Core libraries (not
sure whether they are documented)
2. SPI Lib extensions are completly incompatible between ESP32 and ESP8266
3. Wire extensions partly compatible between ESP32 and ESP8266, but ESP has 
an extra frequency parameter and scl line is optional, this means you can specify "begin(X)"
to move the sd line to pin X, which is not possible on ESP8266.
4. The extension of "begin()" member function is used in 3 of 4 cases. So the 
approach in ESP8266 does not seem to fit.

### Suggestions for modification

ESP8266 and ESP32 should agree on common code regarding the extension 
for movable pins for SPI and Wire functionality.

From seeing the above differences my suggestion is to modify the ESP8266 project
to make it compatible to the ESP32 project:
- SPI Lib: Introduce void begin(int8_t sck=SCK, int8_t miso=MISO, int8_t mosi=MOSI, int8_t ss=-1);
- Wire Lib: Modify "begin(int sda, int scl);"  to "void begin(int sda=-1, int scl=-1, uint32_t frequency=100000);"

me-no-dev, 2017-10-08:
I agree :) We could also add Wire::pins(sda, scl) to ESP32 ;)

devyte, 2017-10-10:
Since we're discussing compatibility and api, do we want to consider #2677  here?

olikraus, 2017-10-10:
As a library author I fully agree that the API fuctions for SPI and Wire should be the same to avoid ifdefs.

freeck, 2017-10-10:
Same for the file system api's for SD, FS, SDFAT libraries...

me-no-dev, 2017-10-10:
FS compatibility is not that simple to do ;) ESP32 follows Arduino's SD API and SPIFFS on ESP8266 uses a different API from the rest. Changing either will bring more issues than anything.

olikraus, 2017-10-10:
I personally would be happy if at least ESP32 and ESP8266 extensions for SPI and Wire libs are compatible.