esp8266/Arduino #3693: OTA "Web Browser" updates failing when device in softAP mode

mwelchuk, 2017-10-08:
### Basic Infos

I'm attempting to utilise the OTA "Web Browser" update mechanism in my project, where I have my ESP8266 running in softAP mode.

#### Hardware
Hardware:		NodeMCU v0.9 (ESP-12 Module)
Core Version:      	2.4.0-rc1-55-g526f4fb

### Description

Attempting to perform update results in ESP8266 sitting in a loop trying to reconnect (see debug messages). Upload does not seem to complete before failure occurs.

### Settings in IDE

Module:  			NodeMCU v0.9 (ESP-12 Module)
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		Unknown (Not shown in GUI)
Flash Frequency:	Unknown (Not shown in GUI)
Upload Using:		SERIAL
Reset Method:		Unknown (Not shown in GUI)

### Sketch

```cpp
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPUpdateServer.h>

const char* ssid = "BattleKrakenTech";
const char* password = "thebkdwarf";

ESP8266WebServer httpServer(80);
ESP8266HTTPUpdateServer httpUpdater(true);

void setup(void){

  Serial.begin(115200);
  Serial.println();
  Serial.println("Booting Sketch...");

  WiFi.softAP(ssid, password);

  httpUpdater.setup(&httpServer);
  httpServer.begin();

  IPAddress myIP = WiFi.softAPIP();
  Serial.print("HTTPUpdateServer ready! Open http://");
  Serial.print(myIP);
  Serial.println("/update in your browser");
}

void loop(void){
  httpServer.handleClient();
}
```

### Debug Messages

```
Booting Sketch...
HTTPUpdateServer ready! Open http://192.168.4.1/update in your browser
Update: smoke.ino.nodemcu.bin
sleep disable
.....................................scandone
no BattleKrakenTech found, reconnect after 1s
Update was aborted
reconnect
scandone
no BattleKrakenTech found, reconnect after 1s
reconnect
scandone
no BattleKrakenTech found, reconnect after 1s
reconnect
scandone
no BattleKrakenTech found, reconnect after 1s
reconnect
scandone
no BattleKrakenTech found, reconnect after 1s
reconnect
scandone
no BattleKrakenTech found, reconnect after 1s
reconnect
...
```

devyte, 2017-10-09:
@mwelchuk the sketch attached (let's call it sketch1) serves the ssid BattleKrakenTech, and the update webserver. Then, using the update page, you upload a new sketch (let's call it sketch2). At this point, sketch1 no longer exists on the ESP, and hence BattleKrakenTech ssid is not there.
Your messages show that sketch2 is trying to connect to the ssid served by sketch1, which doesn't make sense.
Please attach sketch2 here to confirm.

mwelchuk, 2017-10-09:
sketch2 == sketch1

My understanding was that the update process would upload the file, reprogram the device then reboot into the new sketch. The process appears to me to be failing during the upload process (i.e. when it's still running sketch1). I believe this to be the case because the browser seems to be suggesting that it's failing before the transfer has got to 100%, the upload fails after a random period of time (sometimes only 4-5 dots are drawn on the serial port) and I see the "Update was aborted" message from ESP8266HTTPUpdateServer.cpp rather than the "Update Success: %u\nRebooting...\n" message.

devyte, 2017-10-10:
@mwelchuk I am currently using OTA via webserver, and it works well for me. Are you 100% sure sketch1 == sketch2?
Are you using Arduino IDE? If you enable extra output in the settings, you can see a message at the end where the .bin is placed after build.
Ok, please try the following:
- erase the entire flash (e.g.: with esptool)
- flash your sketch over USB
- reset the ESP via rst button
- do your OTA

mwelchuk, 2017-10-10:
Huh, interesting. Thanks @devyte, it seems that erasing the entire flash did the trick.

closed 2017-10-10 by mwelchuk

fredlcore, 2017-10-17:
The same problem happens to me, too - it seems to want to connect to a WiFi access point stored somewhere in flash that I connected the ESP some time ago, even though my current script runs in SoftAP mode only. Erasing the flash with 
`esptool -cd nodemcu -cp /dev/cu.wchusbserial1410 -ce`
did not help either.

fredlcore, 2017-10-18:
Ok, for me the solution was to call
`WiFi.disconnect();`
after the unsuccessful try to connect to an AP and before setting up the softAP. Then web OTA  update works fine. Unfortunately, I haven't found a way to only issue the command upon a firmware update so that the credentials are only cleared in that situation and not having to include it generally in the setup section.