esp8266/Arduino #3694: undefined reference to `__wrap_register_chipv6_phy'

yannS2016, 2017-10-09:
scons: Building targets ...
scons: building associated VariantDir targets: out/esp8266/esp8266/release
Linking out/esp8266/esp8266/release/resource/csdk/stack/samples/esp8266/SimpleClientServer/ocserver/SimpleClientServer
/home/eitvcs/iotivity_mcu/esp8266_arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: out/esp8266/esp8266/release/resource/csdk/stack/samples/esp8266/SimpleClientServer/ocserver/SimpleClientServer section `.text' will not fit in region `iram1_0_seg'
/home/eitvcs/iotivity_mcu/esp8266_arduino/hardware/esp8266com/esp8266/tools/sdk/lib/libmain.a(app_main.o):(.irom0.text+0x14): undefined reference to `__wrap_register_chipv6_phy'
/home/eitvcs/iotivity_mcu/esp8266_arduino/hardware/esp8266com/esp8266/tools/sdk/lib/libmain.a(app_main.o):(.irom0.text+0x31): undefined reference to `__wrap_register_chipv6_phy'
/home/eitvcs/iotivity_mcu/esp8266_arduino/hardware/esp8266com/esp8266/tools/sdk/lib/libphy.a(phy_sleep.o): In function `pm_wakeup_init':
(.irom0.text+0xadb): undefined reference to `__wrap_register_chipv6_phy'
out/esp8266/esp8266/release/esp8266/libgeneric_core.a(generic_core_esp8266_phy.o):(.iram.text+0x8): undefined reference to `__real_spi_flash_read'
out/esp8266/esp8266/release/esp8266/libgeneric_core.a(generic_core_esp8266_phy.o): In function `__wrap_spi_flash_read':
core_esp8266_phy.c:(.iram.text+0x26): undefined reference to `__real_spi_flash_read'
collect2: error: ld returned 1 exit status

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.3.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

jonatanolofsson, 2017-10-28:
I have the same problem, did you ever find a solution?

jonatanolofsson, 2017-10-28:
For future reference: 
In .platformio/platforms/espressif8266/builder/main.py, change the linkflags as:
`            
    "-Wl,-wrap,spi_flash_read"
    # "-Wl,-wrap,register_chipv6_phy"
`

earlephilhower, 2019-01-22:
Closing as this is a PlatformIO code specific error.  If someone does find this again using PIO, please open up something in their tracker as we cannot change any of their wrapper code.

closed 2019-01-22 by earlephilhower