esp8266/Arduino #3697: feature request:

tofrnr, 2017-10-09:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

tofrnr, 2017-10-09:
referce link:
https://github.com/esp8266/Arduino/issues/3688

vdeconinck, 2017-10-09:
@tofrnr I guess you understand that not everything that would be of use to you can qualify as a "feature request". Moreover, I am not sure what the policy is regarding feature requests, but due to limited developement resources, there is an ongoing effort here to focus on actual issues (bugs, unexpected behaviours) in the core. 
You have raised a usability question in #3685 but, as already anwsered, the right place to discuss such questions is on a forum such as esp8266.com or stackoverflow, that are meant to exchange help and experience, particularly if you are just a beginner, as you said. 
Now I'm going to give you [a last pointer](http://adityatannu.com/blog/post/2016/01/31/ESP8266-Witty-Cloud-Board-Demo.html) that demonstrates how websockets work (source included). As you will see, the bidirectional exchange is not http and there are a few things to learn before you master the technology, but I honestly don't think you will find anything much simpler than that, no matter the language. And if you need help, once again, there are remarkable people willing to help on the forums.

devyte, 2017-10-10:
Closing as duplicate of #3685 .

closed 2017-10-10 by devyte

tofrnr, 2017-10-10:
thank you, but I don't need help, that was not my intention.
Instead, it was **_a serious improvement request to make the whole thing simpler,_** for the entire  target group of average Arduino users (which all are non-engeneers and non-computer-scientists): 
Finally Arduino and Wiring were designed _"to make it easy for artists and designers to work with electronics, by abstracting away the often complicated details of electronics so they can focus on their own objectives."_ (Hernando Barragán, https://arduinohistory.github.io/).
So a big wrap around the complicated websocket sources is required to make it as esay like the Wire class, the Serial class, digital and analog Read/Write,  the String class, like graphic libs for TFTs, and all that abstaining to makefiles.
So this is my improvement request: 
As ESP8266WiFi is made for the Arduino IDE and API system, please  enhance ESP8266WiFi.h by websockets in a way to be handled by easy and handsome libs, suitable for all Arduinos users, in the sense of Barragan's basic Arduino/Wiring idea.
If not now immediately, perhaps some day later.

Pablo2048, 2017-10-10:
OMG are you serious? Arduino is built around libraries to be easily expandable so if somewhere does exist websocket library you have to use it (and there DO exist at least two implementations - one already mentioned in comments, another one in webserverasync library). So there is no need to put another implementation into core (you mention graphic libs for TFT - they also are not part of Arduino core).

tofrnr, 2017-10-10:
read the idea of Arduino: 
https://arduinohistory.github.io/
http://people.interactionivrea.org/h.barragan/thesis/thesis_low_res.pdf
Arduino is designed to make it useable to beginners, by basic keywords, commands, and classes, and providing handsome and convenient libs to be included.
No easy and handsome libs, no customers, no Arduino, and no ESP8266.

Pablo2048, 2017-10-10:
Does they wrote that every possible library should be in core? Definitely not and there is library for websocket - exactly in the way/idea of Arduino. BTW websocket has definitely nothing to do with ESP8266WiFi.h - its extension for web server, so your request is er... wrong...

tofrnr, 2017-10-10:
I meant just:
One has to think at the Arduino target group, and design the libs so that standard average Arduino users can use it, and wrap down complicated libs to much easier ons to be used conveniently.

Pablo2048, 2017-10-10:
So what exactly is unclear in suggested websocket library to you? And please post this to the right forum/gitter/github library discussion - not this one.

tofrnr, 2017-10-10:
my point is not anything unclear - my point is a general feature request:
to provide a simple lib which provides the described functionalities either as an ESP8266WiFi  extension or as a simple extra lib. 
but by simple wrappers, suitable for the average Arduino users.

Pablo2048, 2017-10-10:
Yep - you have got some suggestions - simple extra libs given so go on...

me-no-dev, 2017-10-10:
@tofrnr there are plenty of WebSocket libs for Arduno and ESP chips out there. There are 0 reasons to include yet another one in the built-in libs.

tofrnr, 2017-10-10:
it must work with the current  ESP8266WiFi.h functions, or it must provide extra ESP8266WiFi functionalities, to be compliant.
please see my example at https://github.com/esp8266/Arduino/issues/3685

Pablo2048, 2017-10-10:
Why "it must"? Compliant with what? Your mentioned example shows only that you know almost nothing about http protocol and how this protocol work...

tofrnr, 2017-10-10:
it must, according to my improvement request.

Pablo2048, 2017-10-10:
No it does not - your request is nonsense and if you continue this way Me-no-dev probably locks this conversation as he did in 3685..

tofrnr, 2017-10-10:
this is MY improvement request, not yours, so stop arguing.
People said that it's possible with websockets, so that's why I suggested websockets as extension of either kind.

Pablo2048, 2017-10-10:
It seems like YOUR "improvement request" was rejected about 9 hours ago ;-) - so im switching to "do not feed trolls" mode. Have a nice day...

me-no-dev, 2017-10-10:
@tofrnr enough people have answered your "request". It seems that you do not understand that what you ask for is not a "simple" task and requires skills on your side to achieve, more technologies to use and so on. This is not a standard behavior either, so you should not expect someone to jump in and write an example especially for you (that example would require Arduino+WebSocket lib+HTML/Ajax code as well). You say that you have no idea how to write a WebServer or anything of the sort, other languages are "gibberish" and you just demand that this is done to be easy for you, so as someone who wrote enough web servers, I am telling you that this can not and will not be done in the way that you describe it. This framework is not made to cover 100% of your desires, neither is the original AVR Arduino, there are libraries and learning curve involved to get things done. That learning part is for you to do ;) This is not an improvement request, but rather feature request that is already covered by external libs, thus invalid.

p.s. This topic is going out of hand just like the one you linked me to. Please spend a bit of time to read how such task can be accomplished and stop with the thumbs down madness... you are acting a bit childish here and I will lock it. Have a beer, get some fresh air, sleep it over and I hope that you will get to agreement with our points.

tofrnr, 2017-10-10:
no, I simply supposed it to be  a documentation issue.
But it wasn't, it's simply not there at all.
So this is now  an improvement request.