esp8266/Arduino #3698: Linker fails with message "undefined reference to_exit'"

x16, 2017-10-10:
Hello. I cannot compile my class that uses pass-by-reference of TFT_ST7735 object ([https://github.com/sumotoy/TFT_ST7735](https://github.com/sumotoy/TFT_ST7735)).

### Basic Infos

#### Hardware
Hardware:		ESP-12E
Core Version:      	2.4.0-rc1
IDE: Arduino IDE 1.8.4

### Description

The linker fails with message ```undefined reference to `_exit'```. When I switch to Optiboot Arduino (Optiboot at 32-pin CPUs) everything works fine.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/3MB
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL (CH341G)
Reset Method:		ck

### Sketch

```#include <Arduino.h>
#include "SGMenu.h"
#include "TFT_ST7735/TFT_ST7735.h"

#define __CS  16
#define __DC  5

TFT_ST7735 tft(__CS, __DC);

void setup()
{
  SGMenu t(tft);
};

void loop(){};
```

```SGMenu.h``` and ```SGMenu.cpp``` are attached to this message.

### Debug Messages (only the error part, the rest looks absolutely fine)

```
Linking everything together...
"C:\Users\Alexander\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -g -w -Os -nostdlib -Wl,--no-check-sections -u call_user_start -u _printf_float -u _scanf_float -Wl,-static "-LC:\Users\Alexander\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/lib" "-LC:\Users\Alexander\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/ld" "-LC:\Users\Alexander\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/libc/xtensa-lx106-elf/lib" "-Teagle.flash.4m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read  -o "C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_248898/ArduinoMenuTests.ino.elf" -Wl,--start-group "C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_248898\sketch\ArduinoMenuTests.ino.cpp.o" "C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_248898\sketch\SGMenu.cpp.o" "C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_248898\sketch\test.cpp.o" "C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_248898\libraries\SPI\SPI.cpp.o" "C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_248898\libraries\TFT_ST7735-1.0p1\TFT_ST7735.cpp.o" "C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_248898/arduino.ar" -lhal -lphy -lpp -lnet80211 -llwip_gcc -lwpa -lcrypto -lmain -lwps -laxtls -lespnow -lsmartconfig -lmesh -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group  "-LC:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_248898"
C:\Users\Alexander\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/libc/xtensa-lx106-elf/lib\libc.a(lib_a-exit.o):(.literal+0x4): undefined reference to `_exit'

C:\Users\Alexander\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/libc/xtensa-lx106-elf/lib\libc.a(lib_a-exit.o): In function `exit':

/Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/exit.c:70: undefined reference to `_exit'

collect2.exe: error: ld returned 1 exit status
```

It could be my bad as I just switched to C/C++ programming after 10 years of PHP/JS. But why this works on Arduino then?

_________________________
Project files:
[ArduinoMenuTests.zip](https://github.com/esp8266/Arduino/files/1370191/ArduinoMenuTests.zip)

x16, 2017-10-12:
Is there any way to fix this in my particular case or it needs to be fixed globally in the whole core?

igrr, 2017-10-12:
Sure, you can define the following somewhere in your sketch:

```c++
#include <unistd.h>
extern "C" void _exit(int status)
{
  (void) status;
  abort();
}
```

igrr, 2017-10-12:
Alternatively, you can add
```c++
#define oom() abort()
```
before including "ut_array.h". Then ut_array will not attempt to use `exit` function and will use `abort` instead.

x16, 2017-10-12:
The second one works flawlessly.
Thank you!

d-a-v, 2017-12-13:
@x16 can this issue be closed ?

closed 2017-12-13 by x16