esp8266/Arduino #3701: analogWriteFreq(new_frequency)  function didn't work,  the color of the function did't change

TSblue, 2017-10-10:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.1.0-rc2

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <ESP8266WiFi.h>
   
void setup() {
  pinMode(15,OUTPUT);
}

void loop() {
    analogWrite(15,0.5);
    analogWriteFreq();
}

### Debug Messages

```
messages here
```

devyte, 2017-10-10:
@TSblue this issue makes no sense. If what you're trying to do is change the color of a light by using pwm, then changing the frequency doesn't do that. You need to change the pwm duty output, which is done with analogWrite().
Closing due to lack of info.

igrr, 2017-10-10:
I think 'the color' means the text color in IDE. I.e. the function is not added to the keywords file, so doesn't get highlighted by the IDE the way `analogWrite` does.
I think we should fix this.

vdeconinck, 2017-10-20:
Am I correct that syntax highlighting is done with a keywords.txt that should be dedicated to these cores ([supported by Arduino](https://github.com/arduino/Arduino/pull/3847) since 1.6.6) but the ESP8266 cores don't have that file until now ?

igrr, 2017-10-21:
You are correct, we don't have keywords.txt for the core. PRs welcome :)

vdeconinck, 2017-10-21:
OK I'm taking this

vdeconinck, 2017-10-21:
Just filed [PR 3743](https://github.com/esp8266/Arduino/pull/3743).
I admit I could not figure out how to make the third field (link to reference) work for documentation sitting outside of the main Arduino install dir . 
For those interested, the only "specification" I could find about the keywords.txt file is at https://spencer.bliven.us/index.php/2012/01/18/arduino-ide-keywords/

igrr, 2017-10-22:
Thank you, merged.

closed 2018-01-02 by igrr

vickuuuy, 2023-02-17:
> 

What is that 0.5 means in the code

vdeconinck, 2023-02-17:
> What is that 0.5 means in the code
It probably was a mistake by the OP. IIRC analogWrite() takes an integer value as second argument, in a range defined by analogWriteRange().
But that particular snippet was just to illustrate non-working syntax highlighting and that issue has been fixed.

vdeconinck, 2023-02-18:
Remember that this "analog" output actually is PWM. I guess you know what PWM is - if not, please read https://randomnerdtutorials.com/esp8266-pwm-arduino-ide/ - so depending on your application, the default frequency of 1000 (=1kHz) may or may not be OK. 
As your question is not related to this issue, I suggest you turn to a search engine, and you'll find tons of documentation about analog outputs. e.g the first result for "analogWriteFreq esp8266" is that page https://arduino-esp8266.readthedocs.io/en/latest/reference.html . The section "Analog Output" includes hints at what values to use.