esp8266/Arduino #3702: Issue with UDP Multicast – link to #74

jirijasa, 2017-10-10:
Hi,
I would like to reopen issue #74 . I'm facing the same issue described in this original issue. Thing is that it doesn't happen on all ESP8266 devices. I own [Wemos D1 Mini Pro](https://www.aliexpress.com/item/WEMOS-D1-mini-Pro-16M-bytes-external-antenna-connector-ESP8266-WIFI-Internet-of-Things-development-board/32724692514.html) where the attached code runs perfectly fine (packets are sent to unicast and multicast). But if the same is loaded to the [NodeMCU v3](https://www.aliexpress.com/item/1pcs-NodeMCU-V3-Lua-WIFI-module-integration-of-ESP8266-extra-memory-32M-Flash-USB-serial-CH340G/32812846719.html) or [Wemos D1 Mini](https://www.aliexpress.com/store/product/WEMOS-D1-mini-Lite-V1-0-0-WIFI-Internet-of-Things-development-board-based-ESP8285-1MB/1331105_32795857574.html) no packet is sent to multicast addresses. Both devices send successfully unicast packets to 192.168.1.x and both of them are able to receive multicast packets. Only sending doesn't work on some devices. I used the same network and same access point to run the tests. I tried to use also beginPacketMulticast instead of beginPacket, but it has no effect whatsoever. I have also to point out that beginPacket always end up with successful result (1) also in the case when no packet is sent.

It would be great if you could load attached file to some other devices and check with packet sniffer if packets are sent or not.

If needed I can do some deeper debug if required – just let me know how to do that.

Thanks for checking this,
Jiri

### Project

[UDPMulticast.zip](https://github.com/esp8266/Arduino/files/1371791/UDPMulticast.zip)

### Basic Infos

#### Hardware
Hardware:			[Wemos D1 Mini](https://www.aliexpress.com/store/product/WEMOS-D1-mini-Lite-V1-0-0-WIFI-Internet-of-Things-development-board-based-ESP8285-1MB/1331105_32795857574.html) / [NodeMCU V3](https://www.aliexpress.com/item/1pcs-NodeMCU-V3-Lua-WIFI-module-integration-of-ESP8266-extra-memory-32M-Flash-USB-serial-CH340G/32812846719.html)
Core Version:      	2.3.0

### Description

Unicast UDP packets cannot are not sent.

### Settings in IDE

Module:  			[Wemos D1 mini](https://www.aliexpress.com/store/product/WEMOS-D1-mini-Lite-V1-0-0-WIFI-Internet-of-Things-development-board-based-ESP8285-1MB/1331105_32795857574.html) / [NodeMCU V3](https://www.aliexpress.com/item/1pcs-NodeMCU-V3-Lua-WIFI-module-integration-of-ESP8266-extra-memory-32M-Flash-USB-serial-CH340G/32812846719.html)

### Source code
```
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <WiFiUDP.h>

char ssid[] = "XXXX";
char password[] = "12345678";

void setup() {
  //Init Serial
  Serial.begin(115200);
  Serial.println("");

  // MAC address info
  byte mac[6];
  WiFi.macAddress(mac);
  Serial.print("WiFi adapter MAC address: ");
  Serial.print(mac[5], HEX); Serial.print(":"); Serial.print(mac[4], HEX); Serial.print(":");
  Serial.print(mac[3], HEX); Serial.print(":"); Serial.print(mac[2], HEX); Serial.print(":");
  Serial.print(mac[1], HEX); Serial.print(":"); Serial.print(mac[0], HEX); Serial.println(); 
  
  // Connect to Wifi
  Serial.print("Connecting to "); Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");

  // Print wifi info
  Serial.print("SSID: "); Serial.println(WiFi.SSID());
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: "); Serial.println(ip);
  long rssi = WiFi.RSSI();
  Serial.print("Signal strength (RSSI): "); Serial.print(rssi); Serial.println(" dBm");
}

void loop() {
  WiFiUDP udp;
  udp.beginMulticast(WiFi.localIP(), {224, 0, 0, 1}, 10000);
  
  for (int i = 1; i < 255; i++) {  
    Serial.print("Sending packet: 'HELLO' to "); Serial.print(IPAddress({192, 168, 1, i})); Serial.print(':'); Serial.println(10000);
    int result = udp.beginPacket({192, 168, 1, i}, 10000);
    if (result == 1) {
      udp.write("HELLO");
      udp.endPacket();
      Serial.println("Sending packet: 'HELLO' done");
    }
    else
      Serial.println("Cannot send packet");

    delay(100);
  }
  for (int i = 1; i < 255; i++) {
    Serial.print("Sending packet: 'HELLO' to "); Serial.print(IPAddress({224, 0, 0, i})); Serial.print(':'); Serial.println(10000);
    int result = udp.beginPacketMulticast({224, 0, 0, i}, 10000, WiFi.localIP());
    if (result == 1) {
      udp.write("HELLO");
      udp.endPacket();
      Serial.println("Sending packet: 'HELLO' done");
    }
    else
      Serial.println("Cannot send packet");

    delay(100);
  }
}
```

devyte, 2017-10-10:
@jirijasa please try:
- latest git
- PR #3215  (SDK2.1.0)
- PR #3362  (lwip2)

jirijasa, 2017-10-10:
@devyte Thanks for suggestions. When using version 2.1.0 it works fine both using NodeMCU and Wemos D1 Mini Pro.

zllimit, 2017-10-11:
@JirinSV just as you said, the problem happned in the process of multicasting. The xiaomi gate is not recoginzed by wemos, so it's failed to send command to the gate(I modified the loxone IP to the real IP am using, so reading is ok), I am using Wemos D1.

devyte, 2017-10-27:
@igrr this is reportedly fixed by SDK 2.1.0. Should the issue be labeled. or just closed?

closed 2017-10-27 by igrr