esp8266/Arduino #3706: i have problem with WiFiClientSecure :(

mechanic98, 2017-10-11:
### Basic Infos
hi
in this code i want give data from google Sheet and send data in telegram bot,
i use this project to give data from google sheet:
https://github.com/lorol/Googlesheet_DHT_simple

and use this project for recive and send data from telegram robat:
https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot

i use esp8266
both code use client , i changed one time below:
```
WiFiClientSecure client;
UniversalTelegramBot bot(BOTtoken, client);
```
to :
```
WiFiClientSecure test;
UniversalTelegramBot bot(BOTtoken, test);
```
and one time else, change:
`HTTPSRedirect* client= nullptr;`
 to
` HTTPSRedirect* test= nullptr;`
 and change every client in the code to test,
 but in both time i give error in serial monitor and go to crash

#### Hardware
Hardware: NodeMcu Lua WIFI Board Based on ESP8266 CP2102 Module (ESP-12E)

### Description

Problem description

### Settings in IDE

Module: Adafruit HUZZAH esp8266
Flash Size: 4MB(3MB SPIFFS)
CPU Frequency:	80Mhz
Upload Using:	SERIAL 115200

### Sketch

```
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
#include "HTTPSRedirect.h"

//======================From Telegram===============================
#include <WiFiClientSecure.h>
#include <UniversalTelegramBot.h>

#define BOTtoken "339475786:AAH7IjXu51xPom_sMx8IRnvQbcTD5FfO2Wc"  // your Bot Token (Get from Botfather)
WiFiClientSecure test;
UniversalTelegramBot bot(BOTtoken, test);

int Bot_mtbs = 1000; //mean time between scan messages
long Bot_lasttime;   //last time messages' scan has been done
bool Start = false;

const int ledPin = 2;
int ledStatus = 0;

void handleNewMessages(int numNewMessages) {
  Serial.println("handleNewMessages");
  Serial.println(String(numNewMessages));

  for (int i=0; i<numNewMessages; i++) {
    String chat_id = String(bot.messages[i].chat_id);
    String text = bot.messages[i].text;
[... 350 lines cut ...]
3fff1800:  fe91ea00 ff08ac3f fe83a43f 00000f3f  
3fff1810:  ff07f000 2065313f 21f2de40 ff07f040  
3fff1820:  6b08ec00 684b12bc 68736f6f 646e616d  
3fff1830:  617a6173 fe00336e feefeffe 0001c200  
3fff1840:  0000001c 21000000 23774071 35723465  
3fff1850:  40105872 000578c4 3ffea241 4020753c  
3fff1860:  4010592e 3ffeffe8 000578c4 00000000  
3fff1870:  40105b00 000578c4 3fff072c 00000000  
3fff1880:  3fff0010 3fff072c 3fff08c0 3fff072c  
3fff1890:  3ffe9110 3fff08ac 4020710c 3fff08c0  
3fff18a0:  402010ae 0000000a 0000000a 402010a3  
3fff18b0:  3ffe83a4 3fff06d0 3fff07f0 3fff08ac  
3fff18c0:  3fffdad0 00000000 3fff08a4 40207158  
3fff18d0:  feefeffe feefeffe 3fff08c0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
```

mechanic98, 2017-10-12:
hi 
i updated my question
please help me

igrr, 2017-10-12:
It looks like you have two WiFiClientSecure instances at the same time: one for telegram bot, another one for google sheets. That is not going to work, as single WiFiClientSecure instance needs around 25kB of heap (that is in current git version, and even more in release 2.3.0), and the total amount of heap at the start of the program is ~40kB. You need to refactor your program so it only uses one WiFiClientSecure at a time.

closed 2017-10-12 by igrr

mechanic98, 2017-10-12:
this is answer of my friend:
It looks like you have two WiFiClientSecure instances at the same time: one
for telegram bot, another one for google sheets. That is not going to work,
as single WiFiClientSecure instance needs around 25kB of heap (that is in
current git version, and even more in release 2.3.0), and the total amount
of heap at the start of the program is ~40kB. You need to refactor your
program so it only uses one WiFiClientSecure at a time.

but how i do this???

in below is my compile information:

Sketch uses 303141 bytes (29%) of program storage space. Maximum is 1044464
bytes.
Global variables use 41360 bytes (50%) of dynamic memory, leaving 40560
bytes for local variables. Maximum is 81920 bytes