esp8266/Arduino #3707: incorporate Wemos D1 Mini boards in boards.txt

JohnGlassmyer, 2017-10-11:
The board Wemos D1 Mini Lite is not in boards.txt and should be added.

Apparently Wemos have been instructing users of their board to download the Wemos fork of this repo, which has a modified boards.txt:
https://wiki.wemos.cc/tutorials:get_started:get_started_in_arduino
https://github.com/wemos/Arduino_D1

I was able to upload and run a sketch on a D1 Mini Lite just now by adding the d1_mini_lite section of boards.txt from their fork to my installation (which is at 2.4.0_rc1) and using that.

This recent issue might be the same issue for another of their boards, the D1 Mini Pro:
https://github.com/esp8266/Arduino/issues/3640

badlandz, 2017-10-27:
WeMos D1 Mini Lite is an ESP-8266EX so it fits here in ESP8266 IMHO.I agree it might as well be added to this boards.txt because then it covers the WeMos products. WeMos markets the "light" as "ESP-8285," it could lead to a little bit of user confusion, so, maybe with some notes/docs/etc?

d-a-v, 2017-11-03:
d1 mini lite is now in master's board.txt (dout). It's advertised as beeing an [esp8285](https://wiki.wemos.cc/products:d1:d1_mini_lite).
Unless @badlandz did I miss something ?

closed 2017-12-12 by devyte

michaelosthege, 2017-12-26:
These two references were also useful (I am not using the git branch..)
1. https://forum.wemos.cc/topic/274/how-to-flash-the-new-d1-mini-lite-with-arduino
2. https://forum.wemos.cc/post/448 (the board.txt section)

Just posting this here because it was hard to find =/

PR #3940 (2017-12-12): update and fix current generated boards.txt
fix 16M spiffs
less confusing lwip versions in menus
fix #3850
fix #3707
fix #3640
fix #3098
fix #2740
fix #2683
fix #2550
commit bd261c05 (2017-12-12): update and fix current generated boards.txt (#3940)

less confusing lwip versions in menus
fix #3850
fix #3707
fix #3640
fix #3098
fix #2740
fix #2683
fix #2550