esp8266/Arduino #3713: calloc problem results in Exception 29

tony-ozrelic, 2017-10-12:
### Basic Infos

#### Hardware
Hardware:			Adafruit ESP8266 Huzzah board
Core Version:      	2.3.0

### Description

calloc problem results in Exception 29

### Settings in IDE

Module:  			Adafruit ESP8266 Huzzah 
Flash Size: 		4M (3M SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		Serial COM6 115200 baud, AVRISP mkII
Reset Method:		?ck / nodemcu?

Questions:

- how much memory does ESP8266WebServer need?

- how often should server.handleClient() be called? I am calling it every 100msecs, if I call it faster, it tends to crash faster

- If I comment out the server, my program runs with no crashes

### Debug Messages

Decoding 21 results
0x4020c5c6: pp_attach at ?? line ?
0x402148e5: ieee80211_alloc_proberesp at ?? line ?
0x40214822: ieee80211_alloc_proberesp at ?? line ?
0x40214980: ieee80211_send_proberesp at ?? line ?
0x40214aac: ieee80211_send_proberesp at ?? line ?
0x401004f4: calloc at C:\Users\tonyo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1684
0x402116cf: hostap_input at ?? line ?
0x402115a1: hostap_input at ?? line ?
0x40209b40: loop_task at C:\Users\tonyo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40106650: igmp_timer at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/timers.c line 217
0x401004d8: malloc at C:\Users\tonyo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1664
0x40218b45: cnx_node_search at ?? line ?
0x40210e39: hostap_input at ?? line ?
0x4020c288: DefFreqCalTimerCB at ?? line ?
0x4020beba: pp_tx_idle_timeout at ?? line ?
0x4020b807: ppPeocessRxPktHdr at ?? line ?
0x40209b63: loop_task at C:\Users\tonyo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56

devyte, 2017-10-20:
@tony-ozrelic the handleClient() method should be calles as fast as possible. If you call it slowly, the internal buffers of the wifi stack could blow up. If calling it fast causes a crash, then most likely you have either a mem leak, or bad mem fragmentation (e.g.: typical when doing too many String concatenations).
Also, 2.3.0 is rather old, please try latest git, or 2.4.0-rc2.
Difficult to help you without a MCVE sketch. If you can't provide it, this issue will be closed.

devyte, 2018-01-05:
Closing due to no feedback in over 2 months.

closed 2018-01-05 by devyte