esp8266/Arduino #3715: OTA Port not showing after BasicOTA Upload

perigalacticon, 2017-10-13:
### Basic Infos

#### Hardware
Hardware:		Wemos D1 mini
Core Version:      	2.3.0

### Description

Problem description

The OTA Port does not show in the ports list after uploading the BasicOTA sketch.  I am able to use the WIFI through the webserver sketch however.  

I have uninstalled and reinstalled the Core, the IDE (1.8.5), and Python 2.7.14, and restarted the IDE and the computer many times.  I am using Windows 10 64bit.  I have a standard internet connection and router.  I checked Bonjour Services it is running.  

### Settings in IDE

Module:  			?Wemos D1 mini?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```
BasicOTA.ino

```

### Debug Messages

```
messages here
```

Pablo2048, 2017-10-13:
Hi,
this is probably problem with Your firewall settings. You have to enable all networks when asked after installing IDE and Python. Try to disable firewall and re-check if the problem persist...

MarkyAD, 2017-10-13:
Also, the first time an OTA sketch is loaded, reset the board with the reset switch or power cycle it.

perigalacticon, 2017-10-25:
This was working for a while, now the port is not showing again for a few days.  I have the antivirus firewall turned off.  I stop and start and restart Bonjour Services.  I restart the IDE, restart the computer, reset and restart the Wemos, and still no OTA port.  Can you please help me, thanks.

perigalacticon, 2017-10-25:
Sorry for the repeat posting, but I had the wrong sketch uploaded to the device.  It works again, thanks again.

closed 2017-10-25 by perigalacticon