esp8266/Arduino #3718: Esp8266 Freezes up when internet connection is broken while connected to Wi-Fi

jonathanc56, 2017-10-14:
### Basic Infos

#### Hardware
Hardware:			?ESP-07?
Core Version:      	?2.1.0-rc2?

### Description

This problem arises when the ESP8266 is already successfully connected to a network and function perfectly. However, in the case when the internet connection to the router is broken,but the wifi network is still available (removing the sim from a 4G wireless router, removing the telephone cable from a regular router, or disabling mobile data on a phone while a hotspot is generated) this causes the ESP8266 to become unresponsive and not do anything until resetted or the internet connection comes back.

I've tried constantly pinging a remote server to check internet connectivity, but this takes too much resources and time from my main loop.

Any help with this would be appreciated. 

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?dio?
Flash Frequency:	?40Mhz?
Upload Using:		?SERIAL?
Reset Method:		?ck?

RudyFiero, 2017-10-15:
> in the case when the internet connection to the router is broken, but the wifi network is still available ... causes the ESP8266 to become unresponsive and not do anything

Unresponsive in what way? Not executing loop? Not connecting to devices on the local network? Not control IO pins?

jonathanc56, 2017-10-16:
@RudyFiero 
Unresponsive in that the loop (or anything at all) does not execute. The ESP basically freezes up, doing nothing. The problem seems to rectify itself when internet is available again. I am using the ESP with Imroy's pubsub client.

devyte, 2017-10-20:
@d-a-v is this something where lwip2 may help?

d-a-v, 2017-10-20:
I think lwip2 won't help here. What could happen if internet if unreachable
- dns not working (and sitting/waiting for a reply or a nack)
- external addresses not responding (not even a nack) inside a loop waiting for data.
So things looping until an answer from the outside comes back must be identified in the sketch.
That's what @RudyFiero is asking about ("what could not work without internet").
@jonathanc56 you can try and bissect/isolate the locked part of your sketch with some smart println().

RudyFiero, 2017-10-21:
I finally had time to try something to test this. I had a sketch running. A version of https://github.com/squix78/esp8266-weather-station

https://cdn.instructables.com/F79/JCDV/II6EAJ8X/F79JCDVII6EAJ8X.LARGE.jpg

It gets NTP time, 
WundergroundClient gets weather information,
gets data from THINGSPEAK,
and other stuff from my home network.
It cycles through display frames showing the acquired information. 

After three minutes (my setting) it goes out to the Internet to acquire current values from the above sources. But before that happened I disconnected the Internet feed to my router. And then .....

```

Exception (9):
epc1=0x40104904 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000003 depc=0x00000000

ctx: sys 
sp: 3ffffd70 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff10:  0000003c 00000162 00000162 60000600  
3fffff20:  40104b30 0160f3cd 3fff0bd8 00000000  
3fffff30:  3ffef6b0 3fff0bd8 3fff0b55 40208c90  
3fffff40:  00000000 0160db63 402017d0 00000014  
3fffff50:  3fff0bd8 0bf33c6a 60000600 40202b5c  
3fffff60:  402356b4 00000000 3ffe8511 40202b9c  
3fffff70:  60000600 3ffef6b0 3fff3494 402356c1  
3fffff80:  40235706 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffef6c8 3fffdad0 3fff0e3c 4020f32f  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

```
```

Decoding 12 results
0x40104b30: ets_timer_arm_new at ?? line ?
0x40208c90: TwoWire::endTransmission(unsigned char) at C:\Users\Rudy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\Wire/Wire.cpp line 252
0x402017d0: delay at C:\Users\Rudy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_wiring.c line 46
0x40202b5c: displayOff() at C:\Users\Rudy\AppData\Local\Temp\arduino_build_230170\sketch/DisplayPages.h line 166
0x402356b4: ets_timer_handler_isr at ?? line ?
0x40202b9c: oncePerSecond() at C:\Users\Rudy\AppData\Local\Temp\arduino_build_230170\sketch/DisplayPages.h line 166
0x402356c1: ets_timer_handler_isr at ?? line ?
0x40235706: ets_timer_handler_isr at ?? line ?
0x4020f32f: loop_task at C:\Users\Rudy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_main.cpp line 57

```

Wow. This is bad.

RudyFiero, 2017-10-21:
I tried it again but with the  Example sketch, WiFiClient.ino. 
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiClient/WiFiClient.ino

This sketch tries to connect to Sparkfun and get some data (I guess). Since I didn't have an account setup I would get refused, as seen below. As to be expected.  But for the test I am accessing the internet and getting a response. Because it failed the sketch tries again and again. Again, good for my test. 

I disconnect my internet feed and then ......

```
............
WiFi connected
IP address: 
192.168.100.101
connecting to data.sparkfun.com
Requesting URL: /input/....................?private_key=....................&value=1
HTTP/1.1 302 Found
Cache-Control: no-cache
Content-length: 0
Location: https://data.sparkfun.com/input/....................?private_key=....................&value=1
Connection: close

closing connection
connecting to data.sparkfun.com
Requesting URL: /input/....................?private_key=....................&value=2
HTTP/1.1 302 Found
Cache-Control: no-cache
Content-length: 0
Location: https://data.sparkfun.com/input/....................?private_key=....................&value=2
Connection: close

closing connection
connecting to data.sparkfun.com
Requesting URL: /input/....................?private_key=....................&value=3
HTTP/1.1 302 Found
Cache-Control: no-cache
Content-length: 0
Location: https://data.sparkfun.com/input/....................?private_key=....................&value=3
Connection: close

closing connection
connecting to data.sparkfun.com
connection failed
connecting to data.sparkfun.com
connection failed
connecting to data.sparkfun.com
connection failed
connecting to data.sparkfun.com
connection failed
connecting to data.sparkfun.com
connection failed
connecting to data.sparkfun.com
connection failed
connecting to data.sparkfun.com
Requesting URL: /input/....................?private_key=....................&value=10
HTTP/1.1 302 Found
Cache-Control: no-cache
Content-length: 0
Location: https://data.sparkfun.com/input/....................?private_key=....................&value=10
Connection: close

closing connection
connecting to data.sparkfun.com
Requesting URL: /input/....................?private_key=....................&value=11
HTTP/1.1 302 Found
Cache-Control: no-cache
Content-length: 0
Location: https://data.sparkfun.com/input/....................?private_key=....................&value=11
Connection: close

closing connection
```

I reconnect, and it is all back to normal. No exception. No WDR. No real failure. 

So my guess is that the example program, WiFiClient.ino, looks at the return from the internet access function (whatever that is) and then fails gracefully. 

And this makes me think of all the programs that ignore the returned information from called functions. Because that is a lot of work and I'm to fuckin' lazy to take care of exceptions. (my opinion). And because I feel that I am entitled to be a lazy ass I want someone else to handle this shit and fail gracefully. Because they can do it a lot better than I can and I'm too lazy.  

Anyway, that's my best guess as to what is going on. Hey, I'm not a programmer. I'm just a hobbyist who spends way too much time on this stuff without really learning C++ in a proper manor. 

Anyway, I hope this helps.

RudyFiero, 2017-10-21:
I tried it all again with a ping test sketch.  https://github.com/dancol90/ESP8266Ping

```
..........................................................
WiFi connected with ip 192.168.100.101
Pinging ip 208.80.153.224
Success!!
Pinging ip 208.80.153.224
Success!!
Pinging ip 208.80.153.224
Success!!
Pinging ip 208.80.153.224
Success!!
Pinging ip 208.80.153.224
Error :(
Pinging ip 208.80.153.224
Error :(
Pinging ip 208.80.153.224
Error :(
Pinging ip 208.80.153.224
Error :(
Pinging ip 208.80.153.224
Error :(
Pinging ip 208.80.153.224
Success!!
Pinging ip 208.80.153.224
Success!!
Pinging ip 208.80.153.224
Success!!
```

After a ping test I had the code toggle a led in loop() and then rerun the test. 
```

/*
   With this library an ESP8266 can ping a remote machine and know if it's reachable.
   It provides some basic measurements on ping messages (avg response time).
*/

#include <ESP8266WiFi.h>
#include <ESP8266Ping.h>

const char* ssid = "********";  //  your network SSID (name)
const char* password = "*******";   // your network password

const IPAddress remote_ip(208, 80, 153, 224);   // www.wikipedia.org
const int Led15 = 15;

void setup() {
  pinMode(Led15, OUTPUT);
  digitalWrite(Led15, 1);
  
  Serial.begin(115200);
  Serial.println();
  Serial.println("Connecting to WiFi");

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    Serial.print(".");
  }

  Serial.println();
  Serial.print("WiFi connected with ip ");
  Serial.println(WiFi.localIP());
}

void loop() {
  Serial.print("Pinging ip ");
  Serial.println(remote_ip);

  if (Ping.ping(remote_ip)) {
    Serial.println("Success!!");
  } else {
    Serial.println("Error :(");
  }

  for (int i = 0; i <= 20; i++) {
    digitalWrite(Led15, 1);
    delay(250);
    digitalWrite(Led15, 0);
    delay(250);
  }
}
```
Ping, get result, toggle led a bunch of times,
ping, get result, toggle led a bunch of times,
ping, get result, toggle led a bunch of times,
KILL INTERNET!
ping, get result (FAIL), toggle led a bunch of times,
ping, get result (FAIL), toggle led a bunch of times,
ping, get result (FAIL), toggle led a bunch of times,
Restore Internet
Ping, get result, toggle led a bunch of times,
ping, get result, toggle led a bunch of times,
ping, get result, toggle led a bunch of times,

So I think what I said in my previous post was true. Killing the internet happens and if we are not looking at the returned results our programs can blow up. But if we do look at the results, and handle failure gracefully, then the rest of the program will continue to operate.  At least that is how I see it based on my tests.

RudyFiero, 2017-10-21:
Last test. 

I loaded an MQTT sketch.  https://github.com/philbowles/esparto/tree/master/examples
I used test.mosquitto.org as the broker. 
I used mqtt-spy as the publishing app.

I connected with the ESP8266, and my PC running mqtt-spy. I published and saw the results on the ESP8266. I then killed the internet connection. The ESP8266 continued to work flashing a led. I had lost connection Serial.print statements telling me that my connection was lost. When I reconnected the Internet feed the esparto mqtt application reconnected and continued to work as hoped. 

So I have not found any inherent problem with the ESP8266/Arduino code. I have not tested all possibilities, functions, but at this point I would need to be shown an example of the problem and where it is occurring.

devyte, 2017-10-21:
Alright, given that the issue seems to be in the app and not the core, I'm closing this.

closed 2017-10-21 by devyte

RudyFiero, 2018-09-01:
@aalvarez  I had used different applications to test this and none of them hanged the ESP8266. All of these applications continued. When the internet was restored the reestablished the connection and continued on. No hangs.

If you are having this problem then maybe you need to look closer at your code. Do you look at the return values after a function call to the core? If not then maybe you have to. Information is passed back for a reason. 

If you have this problem, and you can show that it is fault with an ESP8266-Arduino component, then please show where the problem occurs.

aalvarez, 2018-09-01:
All ok now , using 2.4.1 world like a charm ,older versions we get hangs ...

Thanks for aswer so fast !!!