esp8266/Arduino #3719: wrong type in declaration of ESP.deepSleep()

5chufti, 2017-10-15:
### Basic Infos

repeated issue as mentioned in "sdk update 2.1.0" branch

#### Hardware
Hardware:		not relevant
Core Version:      	2.1.0-rc2

### Description

deepsleep argument for µs is uint64 in sdk but only uint32 in esp8266 core

### Settings in IDE

not relevant

### Sketch

not relevant

### Debug Messages

not relevant

messages here

see attached diff
[Esp_core.diff.txt](https://github.com/esp8266/Arduino/files/1385269/Esp_core.diff.txt)

igrr, 2017-10-15:
Thanks, will add an overload with uint64_t.
What's the cali_proc for in ESP class, by the way?

5chufti, 2017-10-15:
according to documentation this value is needed to calculate max. deep_sleep time.
see #3302
but somehow this value even changes during operation, so it is more "orientation" than exact value.

igrr, 2017-10-15:
Right, but would it then be more useful to add a function to ESP class to
get the max sleep time?

5chufti, 2017-10-15:
whatever you like, just wanted to keep it simple.

5chufti, 2018-01-05:
@igrr 
again not in current release ....

5chufti, 2018-01-12:
@igrr @devyte 
could you please at least tag a milestone so that this topic isn't lost in next release?

igrr, 2018-01-12:
@5chufti Done, tagged as a bug

PR #4234 (2018-01-28): Change argument to Esp.deepSleep from uint32 to uint64 to match SDK, …
…add deepSleepMax based on the cali_proc function per SDK.

Fixes #3719 for 2.4.1.
closed 2018-01-28 by igrr