esp8266/Arduino #3720: minor prob: 2 warnings during compile

5chufti, 2017-10-15:
### Basic Infos

two easy solveable warning during compile

#### Hardware
Hardware:		not relevant
Core Version:      	2.1.0-rc2

### Description

superfluous variable declaration in spiffs and core Esp code

### Settings in IDE

not relevant

### Sketch

not relevant

### Debug Messages

esp8266\cores\esp8266\Esp.cpp: In member function "bool EspClass::eraseConfig()":
esp8266\cores\esp8266\Esp.cpp:399:10: warning: unused variable "ret" [-Wunused-variable]
     bool ret = true;
             ^

esp8266\cores\esp8266\spiffs\spiffs_hydrogen.c: In function "SPIFFS_buffer_bytes_for_filedescs":
esp8266\cores\esp8266\spiffs\spiffs_hydrogen.c:16:49: warning: unused parameter "fs" [-Wunused-parameter]
 u32_t SPIFFS_buffer_bytes_for_filedescs(spiffs *fs, u32_t num_descs) {
                                                 ^

devyte, 2017-10-20:
@5chufti I somehow doubt that the core version specified is correct...

5chufti, 2017-10-20:
yep, corrected

devyte, 2017-11-01:
Closing as a subset of #3778 . Will track there.

closed 2017-11-01 by devyte