esp8266/Arduino #3723: KeyPad for ESP8266

mkeyno, 2017-10-16:
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12F
Core Version:      	?2.4.0-rc2?

### Description
 I'm trying use 3x3 keypad with 6 ESP8266 pins and Arduino Keypad Library , according to its library, row pins set input pullup and column Pins set output and LOW and in pool all output pins versus input pins will be checked , I made same approach pin 2,0,5 and input pull up and attached to interrupt  with rest pin as output LOW  but after running the following sketch, notice only couple of key combination works (2&16 ->e 0&16->h) any clue really appreciated

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp
const byte ROWS = 3; //four rows
const byte COLS = 3; //four columns
//define the cymbols on the buttons of the keypads
char myKeys[ROWS][COLS] = {
 //16   12  4
  {'a','b','c' }, //5
  {'d','e','f' }, //2
  {'g','h','i' } //0
};
byte rowPins[ROWS] = {5, 2, 0}; //connect to the row pinouts of the keypad
byte colPins[COLS] = {16, 12, 4}; //connect to the column pinouts of the keypad

bool NEW_KEY=false;
byte ROW,COLUMN;
void FindKey0 (void){ROW=0;  NEW_KEY=true;}
void FindKey1 (void){ROW=1;  NEW_KEY=true;}
void FindKey2 (void){ROW=2;  NEW_KEY=true;}
void setup(){
  Serial.begin(115200);
  Serial.println("\n\n\nStarting.........");
  
pinMode(rowPins[0], INPUT_PULLUP);pinMode(rowPins[1], INPUT);pinMode(rowPins[2], INPUT);
attachInterrupt(rowPins[0], FindKey0, CHANGE);
attachInterrupt(rowPins[1], FindKey1, CHANGE);
attachInterrupt(rowPins[2], FindKey2, CHANGE);
pinMode(colPins[0], OUTPUT);pinMode(colPins[1], OUTPUT);pinMode(colPins[2], OUTPUT);
digitalWrite(colPins[0],LOW);
digitalWrite(colPins[1],LOW);
digitalWrite(colPins[2],LOW); 
}
  
void loop(){
 if(NEW_KEY){

         if(digitalRead(colPins[0])) {COLUMN=0;NEW_KEY=false; printf("New Key=%c  (%u,%u)\n",myKeys[ROW][COLUMN],rowPins[ROW],colPins[COLUMN]);}
   else  if(digitalRead(colPins[1])) {COLUMN=1;NEW_KEY=false; printf("New Key=%c  (%u,%u)\n",myKeys[ROW][COLUMN],rowPins[ROW],colPins[COLUMN]);}
   else  if(digitalRead(colPins[2])) {COLUMN=2;NEW_KEY=false; printf("New Key=%c  (%u,%u)\n",myKeys[ROW][COLUMN],rowPins[ROW],colPins[COLUMN]);}
 }
}
 
```

devyte, 2017-10-22:
@mkeyno I don't see anything obvious in your code.
Be aware that the ESP has inverted logic: a 0V input means HIGH, a 3.3V input means LOW.
I don't see an issue in the core, so closing per #3655 .

closed 2017-10-22 by devyte

mkeyno, 2017-10-24:
@devyte still most of keys not working , would you please give some hardware or software advise ?

devyte, 2017-10-24:
Check that the pin interrupts all work individually (no keypad at all). There are a lot of examples out there. Pay attention to logic level.
Make sure the keypad is outputting the levels and pins that you think it is (no esp connected at all). Pay attention to voltage, reverse/direct logic, and which pins activate.
Once you have both of the above working, put them together.