esp8266/Arduino #3726: rc2.4.0-rc2 undefined reference to `strnlen_P'

freeck, 2017-10-16:
After installing rc2 I encounter the following link-error:
> M:\workspace\Smartmeterv2_3\esp8266\arduino.ar(WString.cpp.o): In function `String::invalidate()':
> C:\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/WString.cpp:496: undefined reference to `strnlen_P'
After including #include <pgmspace.h> in file Wstring.cpp there were no link-errors.
(this issue looks very much like https://github.com/esp8266/Arduino/issues/401)
Version 2.3 and 2.4.0-rc1 linked successfully. This error only occurs within the context of my application.
Trying to reproduce this error within a small application did not succeed.

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.4.0-rc2

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:	        SERIAL?
Reset Method:		ck 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

devyte, 2017-10-20:
@freeck Do you have multiple versions of the core installed? e.g.: git, 2.3.0, 2.4.0-rc2, etc.
I suggest deleting the entire Arduino IDE, and installing from scratch, and doing a clean install of the core.

freeck, 2017-10-20:
Thanks for the sugestion.
Yes I have installed 2.3.0 and 2.4-r 2, I do not use Arduino ID but the Sloeber-Eclipse IDE
..but I will try this out anyway.

insertjokehere, 2017-10-26:
I'm seeing the same error. I am using PlatformIO, and installing the Arduino core using the instructions found here: http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

* I did not see the error using the 2.3 core
* The sketch posted above doesn't trigger the bug for me. I'm trying to pare down my code to work out what triggers the issue
* Adding ` #include <pgmspace.h>` to `Wstring.cpp` fixes the problem for me
* Compile output as follows:
```
Linking .pioenvs/bedroom/firmware.elf
.pioenvs/bedroom/libFrameworkArduino.a(WString.o):(.text._ZN6String6concatEPK19__FlashStringHelper+0x4): undefined reference to `strnlen_P'
.pioenvs/bedroom/libFrameworkArduino.a(WString.o):(.text._ZN6String6concatEPK19__FlashStringHelper+0x8): undefined reference to `strncpy_P'
.pioenvs/bedroom/libFrameworkArduino.a(WString.o): In function `String::concat(__FlashStringHelper const*)':
WString.cpp:(.text._ZN6String6concatEPK19__FlashStringHelper+0x2c): undefined reference to `strnlen_P'
WString.cpp:(.text._ZN6String6concatEPK19__FlashStringHelper+0x4e): undefined reference to `strncpy_P'
```

bofh69, 2017-11-27:
I'm using PlatformIO with the Atom plugin.

I removed ~/.platformio and the .pio* dirs in the project dir, I removed the plugin, then reinstalled it and let it reinstall PlatformIO. After that I built the project.

From the build output (platformio run -v):
espressif8266 @ 08e41d9 has been successfully installed!
The platform 'https://github.com/platformio/platform-espressif8266.git#feature/stage' has been successfully installed!
PackageManager: Installing toolchain-xtensa @ ~1.40802.0
[...]
PackageManager: Installing tool-esptool @ ~1.409.0
[...]
PackageManager: Installing framework-arduinoespressif8266
[...]
xtensa-lx106-elf-g++ -o .pioenvs/dev/firmware.elf -Os -nostdlib -Wl,--no-check-sections 
-u call_user_start -u _printf_float -u _scanf_float -Wl,-static -Wl,--gc-sections -Wl,-T
eagle.flash.1m128.ld -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read .pioenvs/dev/src/main.ino.o -L/home/bofh/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lib -L/home/bofh/.platformio/packages/framework-arduinoespressif8266/tools/sdk/ld -L/home/bofh/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/lib -L.pioenvs/dev -Wl,--start-group .pioenvs/dev/libFrameworkArduinoVariant.a .pioenvs/dev/libFrameworkArduino.a -lmesh -lwpa2 -lsmartconfig -lespnow -lpp -lmain -lwpa -llwip_gcc -lnet80211 -lwps -lcrypto -lphy -lhal -laxtls -lgcc -lm -lc -lstdc++ .pioenvs/dev/lib/libArduinoJson_ID64.a .pioenvs/dev/lib/libESP8266WiFi.a .pioenvs/dev/lib/libESPAsyncTCP_ID305.a .pioenvs/dev/lib/libAsyncMqttClient_ID346.a .pioenvs/dev/lib/libmy9291.a .pioenvs/dev/lib/libESPAsyncTCP_ID305@1.1.1.a .pioenvs/dev/lib/libHash.a ".pioenvs/dev/lib/libESP Async WebServer_ID306.a" .pioenvs/dev/lib/libESP8266WebServer.a .pioenvs/dev/lib/libESP8266HTTPUpdateServer.a .pioenvs/dev/lib/libESP8266mDNS.a .pioenvs/dev/lib/libArduinoOTA.a .pioenvs/dev/lib/libESP8266LLMNR.a .pioenvs/dev/lib/libESP8266HTTPClient.a .pioenvs/dev/lib/libESP8266httpUpdate.a .pioenvs/dev/lib/libEEPROM.a .pioenvs/dev/lib/libAiLight.a .pioenvs/dev/lib/libTicker.a -Wl,--end-group
.pioenvs/dev/src/main.ino.o: In function `StreamPrint_progmem(Print&, char const*, ...)':
main.ino.cpp:(.text._Z19StreamPrint_progmemR5PrintPKcz+0x25): undefined reference to `strncpy_P'
.pioenvs/dev/src/main.ino.o:(.text._Z11getDeviceIDv+0xc): undefined reference to `strncat_P'

Anything else I should try?

bofh69, 2017-12-12:
I changed  framework-arduinoespressif8266/cores/esp8266/pgmspace.cpp and pgmspace.h to have extern "C" {} around most of the files and then it works.

After digging around in the files I found that tools/sdk/libc/xtensa-lx106-elf/include/assert.h includes pgmspace.h inside an extern "C" block (and a lot of other files are including it within extern "C" blocks), but pgmspace.cpp does not export the symbols with extern "C".

I'll make a PR.

earlephilhower, 2018-11-09:
Looks like this was a -rc issue that's long since cleared up.  Closing, but if it's still there please do open a new issue or reopen this one with enough info to replicate on the 2.5.0-dev git head.

closed 2018-11-09 by earlephilhower