esp8266/Arduino #3728: Calling Wire.begin() a second time confuses other pins

mzealey, 2017-10-17:
### Basic Infos

#### Hardware
Hardware:			ESP-12F
Core Version:      	2.3.0

### Description

As per https://github.com/marcoschwartz/LiquidCrystal_I2C/issues/22 when I use this library with ESP8266 there is an issue due to multiple invocations of `Wire.begin()`. The first is called with custom SDA/SCL pins, the second (from inside the library) with no arguments. When this happens for some reason reading a DHT22 (nothing to do with I2C, set up on GPIO13) continually fails. Commenting out the second `Wire.begin()` in the library fixes the issue. From my reading of the code, calling the function a second time with no arguments should preserve the initial pins that were specified, and I can't see anything strange happening in `twi_init()` function so I'm a bit stumped as to how to go further debugging the problem.

devyte, 2017-10-22:
@mzealey The issue is in the lib. The second call to Wire.begin() without arguments means that the default pins are used, which renders your first call useless. This explains the "confuses other pins".
I suggest that you pursue the issue with the lib author to allow passing the pins as arguments from outside, and forwarding them to the internal Wire.begin() call. With that, your 1st call to Wire.begin() is no longer needed.
Alternatively, the lib should allow a flag that avoids calling Wire.begin() inside, thereby allowing an external call. 
Closing, as issue is not in this repo.

closed 2017-10-22 by devyte

mzealey, 2017-10-22:
@devyte But it looks from the library code https://github.com/esp8266/Arduino/blob/master/libraries/Wire/Wire.cpp#L57 (L57 & L69) that a second call with no arguments should reuse the arguments that were passed to the call with arguments rather than resetting them to their defaults?