esp8266/Arduino #3733: change in rc2 causing my telnet server to fail

dragondaud, 2017-10-17:
### Basic Infos

Change between 2.4.0-rc1 and rc2 has caused my use of WiFiServer for telnet to stop working.

#### Hardware
Hardware:		ESP-12E
Core Version:      	2.4.0-rc2

### Description

My sketch is using WiFiServer to handle a telnet port used for remote debugging. It was working fine in RC1, but stopped working when I tried updating to RC2. Rolling back to RC1 fixes the problem. Connection just hangs after a couple minutes.

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Upload Using:		OTA

### Sketch

https://github.com/dragondaud/SolarGuardn

I tried reproducing the problem in a smaller sketch, but have not yet recreated the problem in isolation. My project's sketch is using a WWW server on port 80, telnet server on port 23, and OTA server on 8266 as well as uploading data to AdafruitIO using MQTT.

Wanted to post about this now, but will continue to expand my test sketch in an attempt to recreate the issue outside of my project.

### Debug Messages

just hangs with no message

dragondaud, 2017-10-19:
New data point on this issue: When also connected to Serial on usb, the telnet session does not hang. Not running out of heap, as that is checked and displayed every 5 secs. Did not know that Serial not being connected would affect anything...

dragondaud, 2017-10-19:
When I connected Serial, I also added Serial.setDebugOutput(true); hoping it would reveal something. While doing further troubleshooting I disabled that debug, while still monitoring Serial, and the telnet server hung, again. Re-enabled debug, and it works again.

dragondaud, 2017-10-19:
Have now confirmed that simply adding Serial.setDebugOutput(true); to sketch, and not actually connecting Serial, causes my simple telnet server to function in 2.4.0-rc2, as it did in rc1 and prior. This is undesirable. :)

devyte, 2017-10-22:
@dragondaud Off the top of my head, I have no idea why the debug messages could make any difference, unless there is some mem corruption or race condition somewhere in your code.
I suggest you uninstall rc2 and install git, and investigate where the crash happens. What I do: I insert a bunch of Serial.println() calls to figure out where the code is going through.
Keep posting your findings here.

dragondaud, 2017-10-22:
I think the debug thing was a red herring, as once returned to the greenhouse the telnet server started hanging again. Rolling back to rc1 fixes the problem. Just tried with git version, same issue. The telnet session just stops outputting anything without disconnecting. The sketch continues to run, and I can force disconnect and then reconnect and its fine for a little bit again. Under rc1 I can leave the telnet connected for days without it stopping or hanging.

devyte, 2017-10-24:
Shouldn't your button ISR have the ICACHE_RAM_ATTR keyword? Also there's a huge delay(5000) in the loop. At this point I suspect your code was working before only due to luck.
I'm sorry, I can't help further with this within your code. It's big, has 3rd party lib usage, and several levels of complexity. This is an issue tracker, not a support forum to debug your repo. If you can't reduce your code to a MCVE sketch that shows an issue, then please close this.

dragondaud, 2017-10-24:
Nothing in any docs I'v seen about ICACHE_RAM_ATTR or in the provided
examples. The button handler isn't related to this issue anyhow, so I don't
understand why you would bring it up. My code working is not 'luck', do you
insult everybody you meet?

dragondaud, 2017-10-24:
Nothing in the docs or provided examples about ICACHE_RAM_ATTR, AFAIHS. The button handler isn't related to this issue anyhow, so I don't understand why you would bring it up. My code working is not 'luck', do you insult everybody you meet? I'm continuing to work on reducing the issue, and am working on a new version using pubsub directly, without Adafruit's IO library. But, nothing in the notes about changes between rc1 and rc2 would suggest a change that affects this, so it seems like a bug to me.

devyte, 2017-10-24:
It's possible that it's not documented, but it _is_ a known issue, per current opened issues.
All hardware ISRs must be held in RAM, otherwise a cache miss could result when trying to execute the function, which in turn means the function must be retrieved out of flash while execution is in the ISR, which will cause an exception.
I am not insulting you, my reference to luck meant that I suspect that with RC1 the binary layout was such that the issue just didn't manifest, i.e.: it was hidden, and with RC2 the layout is different, and hence it pops up.
Please don't take things personally, I will remind you that I am under no obligation to help you with your code, that this is a community effort of volunteers, and that currently there are very few active volunteers. As such, incoming issues must be filtered and prioritized to reduce the total workload.
Notice that I haven't closed the issue so far, precisely because I am waiting for you to reach a conclusion.

dragondaud, 2017-10-24:
I added the ICACHE_RAM_ATTR to my handler, as I certainly want my code to adhere to best practices, but that is unrelated to the output from telnet hanging. To further diagnose the issue, I broke out the handy, dandy, Wireshark. It appears that using 2.4.0-rc2 the ESP is outputting duplicate packets occasionally, and refusing to accept a duplicate ACK for the duplicate packet. The exact same code works perfectly fine using 2.4.0-rc1 and does not generate any duplicate packets. I can upload the packet capture if you'd like to analyse it.

devyte, 2017-10-24:
That's a interesting piece of info. I'm not good with wireshark dumps,  but post it here anyways, maybe someone else will see something.
Also, please try PR #3362 , it has a newer lwip vs. rc2.

dragondaud, 2017-10-24:
Telnet connection under rc1:
`1	0.000000000	10.10.10.153	10.10.10.163	TCP	66	57436→23 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM=1`
`2	0.283017000	10.10.10.163	10.10.10.153	TCP	58	23→57436 [SYN, ACK] Seq=0 Ack=1 Win=5840 Len=0 MSS=1460`
`3	0.283117000	10.10.10.153	10.10.10.163	TCP	54	57436→23 [ACK] Seq=1 Ack=1 Win=64240 Len=0`
`4	1.746465000	10.10.10.163	10.10.10.153	TELNET	55	Telnet Data ...`
`5	1.787260000	10.10.10.153	10.10.10.163	TCP	54	57436→23 [ACK] Seq=1 Ack=2 Win=64239 Len=0`
and repeats Telnet Data packets/ACKs without error until I disconnect.

The same code, compiled using rc2, starts out the same:
`1	0.000000000	10.10.10.153	10.10.10.163	TCP	66	57545→23 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM=1`
`2	0.271246000	10.10.10.163	10.10.10.153	TCP	58	23→57545 [SYN, ACK] Seq=0 Ack=1 Win=5840 Len=0 MSS=1460`
`3	0.271350000	10.10.10.153	10.10.10.163	TCP	54	57545→23 [ACK] Seq=1 Ack=1 Win=64240 Len=0`
`4	1.286804000	10.10.10.163	10.10.10.153	TELNET	55	Telnet Data ...`
`5	1.327056000	10.10.10.153	10.10.10.163	TCP	54	57545→23 [ACK] Seq=1 Ack=2 Win=64239 Len=0`

The next two packets however, are different. Under rc1 it sens a 512 byte packets, followed by a 95 byte packet containing the remaining transmission. With rc2 it sends a 511 byte packet followed by a 96 byte packet with the remaining transmission. Neither drops anything, but indicates something changed in the handling of large packets.

Then, after a seemingly random amount of time from a few seconds to a few minutes, the connection under rc2 has an issue:
`37	59.188898000	10.10.10.163	10.10.10.153	TELNET	96	Telnet Data ...`
`38	59.228580000	10.10.10.153	10.10.10.163	TCP	54	57545→23 [ACK] Seq=1 Ack=1055 Win=63186 Len=0`
`39	59.839026000	10.10.10.163	10.10.10.153	TELNET	96	[TCP Retransmission] Telnet Data ...`
`40	59.839071000	10.10.10.153	10.10.10.163	TCP	54	[TCP Dup ACK 38#1] 57545→23 [ACK] Seq=1 Ack=1055 Win=63186 Len=0`
`41	61.332081000	10.10.10.163	10.10.10.153	TELNET	96	[TCP Retransmission] Telnet Data ...`
`42	61.332151000	10.10.10.153	10.10.10.163	TCP	54	[TCP Dup ACK 38#2] 57545→23 [ACK] Seq=1 Ack=1055 Win=63186 Len=0`

There are also new, frequent, and undesired, IGMP packets which were not present when using rc1:
`28	43.583596000	10.10.10.163	224.0.0.251	IGMPv2	46	Membership Report group 224.0.0.251`
`45	66.932263000	10.10.10.163	224.0.0.251	IGMPv2	46	Membership Report group 224.0.0.251`

dragondaud, 2017-10-24:
Here are the pcap files from my diagnostic.
[temp.zip](https://github.com/esp8266/Arduino/files/1411998/temp.zip)

d-a-v, 2017-10-24:
I had a look to the -rc2 capture:
```
20:07:40.131496 IP ESP.telnet > XXX.57545: Flags [P.], seq 971:1013, ack 1, win 5840, length 42
20:07:40.171181 IP XXX.57545 > ESP.telnet: Flags [.], ack 1013, win 63228, length 0
20:07:45.244658 IP ESP.telnet > XXX.57545: Flags [P.], seq 1013:1055, ack 1, win 5840, length 42
20:07:45.284340 IP XXX.57545 > ESP.telnet: Flags [.], ack 1055, win 63186, length 0
```
...until here all is good.
From now on, lwip1.4 is telling garbage, it always resends the same data even though XXX says it is OK for them (ack 1055):
```
20:07:45.894786 IP ESP.telnet > XXX.57545: Flags [P.], seq 1013:1055, ack 1, win 5840, length 42
20:07:45.894831 IP XXX.57545 > ESP.telnet: Flags [.], ack 1055, win 63186, length 0
20:07:47.387841 IP ESP.telnet > XXX.57545: Flags [P.], seq 1013:1055, ack 1, win 5840, length 42
20:07:47.387911 IP XXX.57545 > ESP.telnet: Flags [.], ack 1055, win 63186, length 0
20:08:02.584288 IP XXX.57545 > ESP.telnet: Flags [F.], seq 1, ack 1055, win 63186, length 0
```
then XXX says something throught the FIN flags [F] (something like "are you kidding?") which causes lwip1.4 to further continue with correct seq but in a broken way with 0 data length packets.
```
20:08:02.828260 IP ESP.telnet > XXX.57545: Flags [R.], seq 1056, ack 2, win 5840, length 0
20:08:02.828306 IP XXX.57545 > ESP.telnet: Flags [.], ack 1055, win 63186, length 0
20:08:02.866628 IP ESP.telnet > XXX.57545: Flags [R.], seq 1056, ack 2, win 5840, length 0
20:08:03.421617 IP XXX.57545 > ESP.telnet: Flags [F.], seq 1, ack 1055, win 63186, length 0
20:08:03.440368 IP ESP.telnet > XXX.57545: Flags [R.], seq 1056, ack 2, win 5840, length 0
20:08:05.012114 IP XXX.57545 > ESP.telnet: Flags [F.], seq 1, ack 1055, win 63186, length 0
20:08:05.278046 IP ESP.telnet > XXX.57545: Flags [R.], seq 1056, ack 2, win 5840, length 0
```
It's a pretty example of broken lwip1.4 we have here. Typical stuff that leaded to #3362.
Between 2.4.0-rc1 and -rc2 there were updates in lwip1.4 from espressif, with not much details.
You should definitely try #3362 with -rc2.
Please note that lwip1.4 was also broken in 2.4.0-rc1 and before, and a very simple sketch that I should post can demonstrate it.

dragondaud, 2017-10-25:
Using rc2 with lwip2 seems to fix my telnet server issue, however, it uses far more RAM. I was running with over 12k free heap in loop() under rc2, but after installing lwip2 my sketch only has 4500 bytes free heap in loop(). Also seem to be having problems with OTA now, too, and will post updates after I collect more info.

dragondaud, 2017-10-25:
Didn't take long before it crashed:
0x401004d8: malloc at C:\Users\daud\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1668
0x40224f44: mem_malloc at /local/users/gauchard/arduino/arduino_esp8266/esp8266-lwip2v2/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c line 125
0x4021fbd1: pbuf_alloc_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/esp8266-lwip2v2/tools/sdk/lwip2/builder/lwip2-src/src/core/pbuf.c line 1306

d-a-v, 2017-10-25:
That's a good news, thanks for the feedback.
So there's work to do for improving lwip2's heap usage.
Progress about that will be visible in #3362 and also in this fresh dedicated lwip2 [project](https://github.com/esp8266/Arduino/projects/7).

dragondaud, 2017-10-25:
OTA problem seems to be related to just needing a hardware reset after update. Also, for some reason, it got a new IP address after the reset. It has accepted additional OTA updates, without intervention. No longer seeing spurious IGMP messages in wireshark, either.

d-a-v, 2017-10-25:
@dragondaud You should have plenty more flash + heap with the latest commit.

dragondaud, 2017-10-25:
Back up to around 12k free heap in loop() again now. I'll let you know if I see any issues, and thanks for your work. I will note that each time it crashed, since I started using lwip2, my dhcp assigned IP address changed, when it normally stays the same, 4ever. Sounds like a malloc fail leading to an overwritten buffer somewhere, due to the low heap. With my heap back up into the comfortable range now, I'm not likely to see that sort of issue, personally, but the underlying issue still exists.

d-a-v, 2017-10-26:
This is interesting. The overwritten buffer is just an assumption, but it could be useful to check global behaviour on low heap.
@igrr what do you think of having  a global debug option that throws a message every time new/*malloc() returns NULL ?

igrr, 2017-10-26:
I think the original SDK heap implementation has that, and adding one to
umm_malloc (or our malloc/pvPortMalloc wrappers) would be straightforward
and useful in some situations.

devyte, 2017-11-01:
@dragondaud your assumption is likely correct: after oom, you can't trust anything anymore, because most likely mem got corrupted somewhere.
How is it running with the latest lwip2? Is there any further issue?

dragondaud, 2017-11-01:
So far, it's working great. And, in my opinion, 2.4.0 should not be released with the new SDK and the broken lwip it includes. Not sure why the SDK was changed between rc1 and rc2, when one would expect a feature freeze, but having a WiFi module without a working networking stack is sorta pointless. Assuming the newer SDK is here to stay, then it seems 2.4.0's release needs to wait for either the SDK  lwip to be fixed, or for the new lwip2 to be ready for release.

I will be doing even more testing this week, as I received two additional ESP modules that I can test on, and monitor serial debugging, without affecting my live monitoring system.

igrr, 2017-11-01:
The only reason why SDK was updated was the KRACK fix.
Perhaps we can have a look at the diffs for LwIP changes and see if
anything looks suspicious.

d-a-v, 2017-11-01:
Here's the lwip [diff](https://gist.github.com/d-a-v/39f4d48f31a6c36c22ae954efa923f4f) between rc1 and rc2.
The relevant changes are the following:
- cc.h: we can't/must not change ```LWIP_ERR_T```. I'm not sure about the "re"definition of ```ICACHE_FLASH_ATTR``` to nothing.
- igmp.h: change ```LWIP_RAND``` from ```r_rand``` to ```os_random```. I don't think this can be related to the telnet sketch

That's it.
I think this is again a lucky situation where lwip1.4 was working. I personally have always had very bad luck with telnet server 3 years ago until now with lwip1.4. Back then I had already ported lwip-pre2 to esp8266, just to have a telnet properly working :) (but it was [not the good way](https://github.com/d-a-v/esp82xx-nonos-linklayer#history) of doing it)

dragondaud, 2017-11-08:
So much other stuff changed when the SDK version was changed, I doubt what caused my issue was an actual change in the lwip code, but more likely a side effect of low ram and bugs. I made a new version of my code without AdafruitIO/MQTT, using pubsubclient instead, and that doubled my available heap. And I have not seen any crashes, on 2.3.0, rc1 or rc2 or current git with my new code version. Since ultimately I don't need the AIO functionality, I'm not going to follow up with adafruit; i'm sure they know their library is a memory hog. :)

closed 2017-11-08 by dragondaud