esp8266/Arduino #3735: A small problem with ESP8266WiFi

szczukot, 2017-10-18:
#### Hardware
Hardware:		Witty mini ESP-12F
Core Version:      	2.4.0-rc2

### Description

I have a problem with WiFi server.
Below the simple test code.

### Sketch

`#include <ESP8266WiFi.h>

const char* ssid = "***";
const char* password = "***";

WiFiServer server(80);

void setup() {
  Serial.begin(115200);

  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println(WiFi.localIP());

  server.begin();
  Serial.println("Server started");
}

void loop() {
  WiFiClient client = server.available();
  if (!client) {
    return;
  }

  Serial.println("new client");
  
  unsigned long timeout = millis();
  while (client.available() == 0) {
    if (millis() - timeout > 2000) {
      Serial.println("\r\n>>> Client Timeout !");

      client.stop();
      return;
    }
  }

  String req = client.readStringUntil('\r');
  Serial.println(req);

  String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html>response</html>\n";

  client.print(s);

  Serial.println("Client disonnected");
}`

I connect to ESP8266 through the browser (chrome).
And sometimes i see a proper response (for one request) (via serial):

new client
GET /12 HTTP/1.1
Client disonnected
new client
GET /favicon.ico HTTP/1.1
Client disonnected

But sometimes i get (only one request !):

new client

Client Timeout !
new client
GET /12 HTTP/1.1
Client disonnected
new client
GET /favicon.ico HTTP/1.1
Client disonnected

Why ??
Why sometimes server get timeout, and in next loop handles previous requests (from previous loop)?

chschu, 2017-10-18:
That's actually a feature of Chrome. Your server needs to be able to deal with those "silent" connections. In ```ESP8266WebServer```, they are simply closed after one second (defined by ```HTTP_MAX_DATA_WAIT```).

Chrome predictively opens TCP connections. I found this WONTFIX bug report from 2012: https://bugs.chromium.org/p/chromium/issues/detail?id=116982

If you disable the "Use a prediction service to load pages more quickly" option in the Chrome settings, Chrome will only open connections on demand.  This is probably not a good idea (performance-wise) on the internet.

I managed to capture a case where Chrome opens six connections - not that I'm understanding WHY it does that:

[image]

As you can see, Chrome opens all six connections at the same time, but it uses only two of them (ports 57597 and 57598) for HTTP requests. Here's the same capture, filtered by one of the four unused TCP connections:

[image]

szczukot, 2017-10-19:
Ok, I understand.
I checked under Edge and curl and it works fine.
How is secure this code for proper working with chrome?

chschu, 2017-10-19:
The best thing you can do is to close the connection after a certain timeout. You already do that by calling ```client.stop()``` in your sketch. Just keep in mind that other clients will have to wait for that timeout, so you should keep it as short as possible.

Just change your sketch to print the time that you waited for the request. That should give you a hint for a reasonable timeout value.

szczukot, 2017-10-19:
In my sketch exists "client.stop();"
But this is not resolved problem

chschu, 2017-10-20:
Your sketch cannot control how Chrome opens connections, so you can't really fix this in your sketch.

Your current implementation (closing the client after a timeout) is the best you can do. You could reduce the timeout (to avoid becoming unresponsive for two seconds) and remove the Serial output. It's really just how Chrome works.

szczukot, 2017-10-20:
Ok.
This serial output is only for test.
I will reduce the timeout.
This Chrome ......

closed 2017-10-20 by szczukot

dachshund-digital, 2019-03-21:
Yeah, having the same issue, Chrome often times out trying to connect client session to ESP8266 Web Server.  But IE11, and Firefox work just fine.  Chrome on my android phone instantly times out when trying to connect to the ESP8266 Web Server.