esp8266/Arduino #3737: Serial port quickly receives data problems

haohanlin, 2017-10-19:
#### Hardware
Hardware:			ESP-12s

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		160Mhz
Flash Frequency:	40Mhz
Reset Method:		 nodemcu

The esp8266 serial port needs to receive data at 100ms per interval, and then use the HTTP.Get() request.But HTTP.GET() performs too long to result in the loss or overlap of data received by the serial port.Is there any solution？

d-a-v, 2017-10-19:
Perhaps you should use big enough circular buffer between serial and HTTP server.

cheng3100, 2017-10-19:
Maybe you can use a timer call back function to call the HTTP.GET every period which longer than 100ms

haohanlin, 2017-10-19:
@cheng3100 However, the timer calls  HTTP. get    time. The serial port to receive data, this will still lead to receive data errors. How to deal with the timely reception of serial data? MCU usually serial port interrupt, but this is not.

vdeconinck, 2017-10-19:
I guess you need to increase the serial receive buffer using Serial.setRxBufferSize() which was implemented in release 2.4.0 rc1 to prevent loss of bytes during http processing

rglissmann, 2017-11-01:
I've added Serial.setRxBufferSize(256) to my sketch and am compiling with release release 2.4.0 rc1 and am getting the error:
Arduino: 1.8.5 (Windows Store 1.8.10.0) (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

C:\Users\Randy\Documents\Arduino\CNC_server\CNC_server.ino: In function 'void setup()':

CNC_server:62: error: 'class HardwareSerial' has no member named 'setRxBufferSize'

   Serial.setRxBufferSize(256);

          ^

exit status 1
'class HardwareSerial' has no member named 'setRxBufferSize'

I've used https://github.com/esp8266/Arduino/releases/download/2.4.0-rc1/package_esp8266com_index.json to the Arduino IDE. Is this the correct way to compile?

haohanlin, 2017-11-01:
Serial.begin (115200 );
Serial.setRxBufferSize(1024); @rglissmann

vdeconinck, 2017-11-01:
OK, now that's another issue. It seems your board type cannot be found anymore. Maybe the tag of your board has changed between versions...
Can you choose another board type in the boards selection menu (say Wemos D1 mini) and check that it compiles, then select your NodeMCU board and try again ?

rglissmann, 2017-11-01:
That helped. It compiles and also I don't get an error with the setRxBufferSize. I'm able to receive a larger payload now! I need to retry uploads several times to get OTA to work. I'm getting an error 11.

vdeconinck, 2017-11-01:
If the serial problem is fixed, could you please close this issue.
If you have another problem, please discuss it first on esp8266.com or other forums, and if there is a consensus that you found an issue in the esp8266 cores, feel free to reopen a new issue here, with a minimal sketch exhibiting the problem.

rglissmann, 2017-11-02:
Closes #3737

closed 2017-12-27 by igrr

jeroenst, 2018-09-26:
Works like a charm, thanx