esp8266/Arduino #3744: ESP Crashes when writing IP to serial

tavrez, 2017-10-22:
### Basic Infos
####dump
Decoding 14 results
0x4010020c: _umm_free at C:\Users\reza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4010068c: free at C:\Users\reza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x402027b0: operator delete(void*) at C:\Users\reza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/abi.cpp line 84
0x40202063: HardwareSerial::~HardwareSerial() at C:\Users\reza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/HardwareSerial.cpp line 193
0x4020256f: Stream::timedRead() at C:\Users\reza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Stream.cpp line 34
0x402025b3: Stream::readBytes(char*, unsigned int) at C:\Users\reza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Stream.cpp line 211
0x40202b86: Print::print(Printable const&) at C:\Users\reza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 136
0x402023b1: Print::println(Printable const&) at C:\Users\reza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 76
0x40201d43: readConfig() at e:\Users\reza\Desktop\sketch_sep03a/sketch_sep03a.ino line 87
0x40202864: esp_yield at C:\Users\reza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x402010ae: delay at C:\Users\reza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring.c line 53
0x40201e5b: setup at e:\Users\reza\Desktop\sketch_sep03a/sketch_sep03a.ino line 105
0x402028a8: loop_wrapper at C:\Users\reza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\reza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109

#### Hardware
Hardware:			ESP-12,ESP-01,nodemcu
Core Version:      	2.3.0(but I'm not sure, everything is up to date from the repo)

### Description

I have a code that write a struct into a EEPROM with EEPROM.put(), when I read it back and use serial.print() to print the IP inside of that struct(for debug) it cause exception 3 and resets the module, interesting part is, if I write the IP via it's toString() method, everything works fine

I should also add that I tested a lot to reach to this point, at first it was always hardware watchdog reset, now it's like this, problem is in Print.cpp in Print::print(const Printable& x) method, when it pass *this to 
printTo(), these errors raise(sometime hardware watchdog, sometimes exception(0), and sometimes exception(3), look like a pointer error to me, but it's so complicated so I decided to share it with you guys, if you need any more info, I'll provide it here

### Settings in IDE

Module:  			Generic ESP8266 Module,  nodemcu 1.0(esp 12-e module)
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPClient.h>
#include <WiFiUDP.h>
#include <EEPROM.h>
#include <ESP8266HTTPUpdateServer.h>
#define DELAY 5
#define LOG Serial
struct ConfigData {
  bool useDHCP;
  IPAddress ip;
  IPAddress netmask;
  IPAddress gateway;
  IPAddress dns;
  IPAddress udp;
  int port;
  int udpPort;
  char ssid[20+1];
  char pass[20+1];
//DI specific:
  char di_on[8][50];
  char di_off[8][50];
  unsigned long time[8];
  
  ConfigData():
  useDHCP(true),
  ip(192, 168, 1, 200),
  netmask(255, 255, 255, 0),
  gateway(192, 168, 1, 1),
  dns(192, 168, 1, 1),
  udp(224, 0, 0, 30),
  port(80),
  udpPort(8000)
  {
    strcpy(ssid, "SmartLAB");
    strcpy(pass, "9123132184");

    for(int i=0;i<8;i++)
    {
      strcpy(di_on[i], "");
      strcpy(di_off[i], "");
      time[i] = 5000;
    }
  }
} data;

bool readConfig()
{
  if(EEPROM.read(0) == 33)
  {
    EEPROM.get(1, data);
    LOG.println("Reading config from EEPROM");
    LOG.println(data.useDHCP);
    LOG.println(data.ip);//this line will crash
    LOG.println(data.dns.toString());
    LOG.println(data.gateway.toString());
    LOG.println(data.netmask.toString());
    LOG.println(data.udp.toString());
    LOG.println(data.port);
    LOG.println(data.udpPort);
    LOG.println(data.ssid);
    LOG.println(data.pass);
    for(int i=0;i<8;i++)
    {
      LOG.println(data.di_on[i]);
      LOG.println(data.di_off[i]);
      LOG.println(data.time[i]);
    }
    return true;
  }
  LOG.println("EEPROM empty, using default values this added to make sure things get buggy");
  return false;
}

void writeConfig()
{
  EEPROM.write(0, 33);
  EEPROM.put(1, data);
  EEPROM.commit();
}

void eraseConfig()
{
  for(int i=0;i<1000;i++)
    EEPROM.write(i, 0);
  EEPROM.commit();
}

void setup()
{
#ifndef NDEBUG
  Serial.begin(115200);
  delay(DELAY);
#endif
  LOG.println();
  LOG.println();
  
  EEPROM.begin(1000);
  LOG.print("Need EEPROM: ");
  LOG.println(sizeof(ConfigData)+1);
  delay(DELAY);
  //eraseConfig();
  readConfig();
  //writeConfig();
}

void loop()
{
  
}

igrr, 2017-10-22:
Does it also crash if you do println(data.ip.toString())?

tavrez, 2017-10-22:
Should I include something for println? I'm getting 'println' not defined

igrr, 2017-10-22:
Well, your code already uses LOG.println, which is Serial.println.

tavrez, 2017-10-22:
Sorry I thought you mean bare println,
Serial.println(data.ip) will crash
but Serial.println(data.ip.toString()) won't
first time when we upload the code to the module, non of the above will crash, when you write something to EEPROM both will also work, but when you change a string inside code( like that "eeprom  empty" message I modified), after that you will always get crash when you do Serial.println(data.ip), even if you clear the contents of PrintTo method in IPAddress.cpp completely, but if you change Print::print(const Printable& x) in print.cpp to not call x.printTo(*this), it won't crash.

tavrez, 2017-10-23:
I think I found the problem.
Storing IPAddress class with EEPROM.put will save it's 4 byte pointers(it has virtual methods), so when we load it back with get, it cause some problems. I should avoid using put & get or it should be optimised so it use some methods from each class to just store it's data, not just cast it to byte and saves it.

devyte, 2017-10-23:
@tavrez the template put and get methods are meant for simple objects. What you are trying to do is called serialization, and you can't do it like this with complex objects. So your last comment is correct.
There are libs out there that implement serialization in general terms (most common is in stdlib), but you have to implement certain methods.
Closing as user error.

closed 2017-10-23 by devyte