esp8266/Arduino #3746: OTA Update with check md5 From File on a server

mcmega, 2017-10-22:
Hi All!
On the server is the file fw.txt.
It contains the firmware version and md5 bin files:
1 F52187696F59EE5A1D85285D0F0B62A8 29131F8A6A853402AAB5433D521B1EC0
I'm reading this file and I'm looking at the firmware version (the first number).
If the version is more current, I get the bin files and run the update.
From the file I take md5 and compare. But there is always a mistake, and monologue does not work.
Test avalible Ubdate:
```
long latest_version = VERSION;
char *latest_esp_md5 = "";
char *latest_fs_md5 = "";

bool otaAvailable() {
    if(WiFi.status() == WL_CONNECTED) {
        HTTPClient httpClient;
        httpClient.useHTTP10(true);
        httpClient.setTimeout(5000);
        char url[65];
        sprintf( url, "https://%s/%s.txt", OTA_HOST, DEVICE_MODEL );
        httpClient.begin(url, OTA_SSL_FP);
        int code = httpClient.GET();
        if (code != HTTP_CODE_OK) {
            httpClient.end();
            return false;
        }
        String str = httpClient.getString();
        httpClient.end();

        char buffer[110];
        str.toCharArray(buffer, 110);
        latest_version = atol(strtok(buffer, " "));
        latest_esp_md5 = strtok(NULL, " ");
        latest_fs_md5 = strtok(NULL, " ");

        if (latest_version > VERSION)
            return true;
        return false;
    } else {
        return false;
    }
}
```
Update:
```
void otaUpdate() {
    _otaUpdate(true) && _otaUpdate(false);
    uartSendCommand("SYS", "restart");
}

bool _otaUpdate(bool spiffs) {
    Serial.println(spiffs ? "Updating SPIFFS" : "Updating FLASH");
    HTTPClient httpClient;
    httpClient.useHTTP10(true);
    httpClient.setTimeout(5000);
    char url[128];
    sprintf(
        url, "http://%s/%s-%d-%s.bin",
        OTA_HOST,
        DEVICE_MODEL, latest_version,
        spiffs ? "fs" : "esp"
    );
    httpClient.begin(url);
    int code = httpClient.GET();
    if (code != HTTP_CODE_OK) {
        Serial.printf("Unable to fetch update %s", url); Serial.println();
        return false;
    }
    if(!Update.setMD5(spiffs ? latest_fs_md5 : latest_esp_md5)) {
        Serial.printf("Error: Update.setMD5(%s): ", spiffs ? latest_fs_md5 : latest_esp_md5);
        Update.printError(Serial);
        return false;
    }
    if (!Update.begin(httpClient.getSize(), spiffs ? U_SPIFFS : U_FLASH)) {
        Serial.printf("Error: Update.begin(%d, U_FLASH): ", httpClient.getSize());
        Update.printError(Serial);
        return false;
    }
    if (!Update.writeStream(*httpClient.getStreamPtr())) {
        Serial.print("Error: Update.writeStream(*httpClient.getStreamPtr()): ");
        Update.printError(Serial);
        return false;
    }
    if (!Update.end()) {
        Serial.print("Error: Update.end(): ");
        Update.printError(Serial);
        return false;
    }
    Serial.println(spiffs ? "SPIFFS update OK" : "FLASH update OK");
    httpClient.end();
    return true;
}
```
I find myself in if(!Update.setMD5(spiffs ? latest_fs_md5 : latest_esp_md5)). If you disable the condition check, the update works.
How to solve a problem?

devyte, 2017-10-23:
@mcmega no basic info, no IDE settings, sketch is incomplete. The issue template exists for a reason. Please update your post with the required info.

tranpic, 2017-10-23:
HI
i copy your OTA library and paste in C:\Users\Tran\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries

but have no ArduinoOTA.getCommand(). -> i delete it.

then can not update via OTA

15:01:52 [ERROR]: No response from device

Error[1]: Begin Failed (debug not arduino ide).

NodeMCU 1.0 (ESP-12E Module)
CPU Frequency: 80MHz
Flash Size: "4M (3MSPIFFS)"
Upload Speed: "115200"
Port: "esp8266-df04c4 at 192.168.25.130"

tranpic, 2017-10-23:
after install core 2.4.0 

16:01:13 [ERROR]: Bad Answer: ERR: ERROR[8]: Flash config wrong real: 2097152 IDE: 4194304

the selected serial port 
 does not exist or your board is not connected

how to fit it?

mcmega, 2017-10-23:
The project is large and divided into parts.
IDE Settings:
Board: D1 mini (ESP-12)
CPU Frequency: 80MHz
Flash Size: "1M (256KSPIFFS)"
Upload Speed: "115200"

I call the function otaAvailable(), Function reads a file txt (It contains the firmware version and md5 for bin files:
1 F52187696F59EE5A1D85285D0F0B62A8 29131F8A6A853402AAB5433D521B1EC0
I'm parsing this file (1 - version number, 2 - md5 FS, 3 - md5 Firmware, Separator - space).
Server with SSL

The function otaAvailable() returns true or false.
If true - update started otaUpdate()
Varibles  is global
long latest_version = VERSION;
char *latest_esp_md5 = "";
char *latest_fs_md5 = "";

But I get into the condition check and the update does not happen.
if(!Update.setMD5(spiffs ? latest_fs_md5 : latest_esp_md5)) {
        Serial.printf("Error: Update.setMD5(%s): ", spiffs ? latest_fs_md5 : latest_esp_md5);
        Update.printError(Serial);
        return false;
    }
If you disable this check, everything works correctly.

mcmega, 2017-11-16:
Updated topic!

closed 2017-12-11 by mcmega