esp8266/Arduino #3747: analogRead(A0) Issue after ESP.deepSleep(1*1000000, WAKE_RF_DISABLED)

entropia1ac, 2017-10-22:
Hardware
Hardware: ESP8266 E12F
Core Version: 2.4.0 SDK2.1.0

Sketch
void setup()
{
Serial.begin(9600);
delay(100);
}
//OK===================
void loop() {
  Serial.println();
 rawV=analogRead(A0);
Serial.print("rawV=");Serial.println(rawV);//rawV= = 500 ;VinA0=0,500V
 ESP.deepSleep(1*1000000);

}
//OK===================

//Issue=============
void loop() {
  Serial.println();
 rawV=analogRead(A0);
Serial.print("rawV=");Serial.println(rawV); // rawV== 1024 ; VinA0=0,500V
 ESP.deepSleep(1*1000000, WAKE_RF_DISABLED);
}
//Issue=============

devyte, 2017-10-22:
@entropia1ac and what is the issue?

entropia1ac, 2017-10-23:
In Sketch //Issue==========:
input voltage VinA0=500mV  and voltage measured analogRead(A0) =1024

devyte, 2017-10-23:
Oh, it took me a while o understand what you mean. 
The only difference is the rfcal on wakeup. What is your power source?'This looks like a power source issue.

entropia1ac, 2017-10-23:
Power supply? - impossible - I have  power supply 0-30V / 5A.
How long does a rfcal take? 1-3 seconds.?
This sketch also does not work:
void loop() {
delay(20000) ; //for  rfcal 
Serial.println();
rawV=analogRead(A0);
Serial.print("rawV=");Serial.println(rawV); // input voltage VinA0=500mV and voltage measured 
                                                                                          // analogRead(A0) =1024
ESP.deepSleep(1*1000000, WAKE_RF_DISABLED);
}

devyte, 2017-10-24:
Alright, I just tested this on a WittyCloud, and I can't reproduce. With or without WAKE_RF_DISABLED makes no difference, I get the same analogRead value. Of course, on a WittyCloud I'm measuring the light sensor and not a direct voltage, but the point is to test the core, not the electronics.
- Is the sketch you posted above complete?
- Did you do a full flash erase before flashing the sketch?
- Did you try a different board?
- Are you using a generic ESP12F with custom electronics, or a preassembled board like a Wemos D1 or WittyCloud?

entropia1ac, 2017-10-24:
//Is the sketch you posted above complete?//
-yes
//Did you do a full flash erase before flashing the sketch?//
-yes (esptool.py --port COM8  erase_flash)
//Did you try a different board?//
-yes
with the NONOS_SDK_V2.0.0_  works OK !
with the NONOS_SDK_V2.1.0_ does not work.

entropia1ac, 2017-10-24:
Today I updated the Arduino with git (I got the version from 18.07.2017) -analogRead(A0)  after ESP.deepSleep(1*1000000, WAKE_RF_DISABLED) works OK  :)
Thank you.

closed 2017-10-24 by entropia1ac