esp8266/Arduino #3752: Ticker once functions without callback argument missing

cbossut, 2017-10-24:
Hello

It's more of a feature request than an issue, and could have been a pull request if I had the time:

In the Ticker library, there are attach functions overloaded to pass an argument to the callback or not, but the once functions only comme in the version with an argument for the callback, not without.

devyte, 2017-10-24:
@cbossut no basic info, no example sketch. You pretty much ignored the entire issue template.
In Ticker.h I see:
	void attach(float seconds, callback_t callback)     //no arg
	void attach_ms(uint32_t milliseconds, callback_t callback)   //no arg
	void attach(float seconds, void (*callback)(TArg), TArg arg)   //arg
	void attach_ms(uint32_t milliseconds, void (*callback)(TArg), TArg arg) //arg

	void once(float seconds, callback_t callback)  //no arg
	void once_ms(uint32_t milliseconds, callback_t callback)  //no arg
	void once(float seconds, void (*callback)(TArg), TArg arg)  //arg
	void once_ms(uint32_t milliseconds, void (*callback)(TArg), TArg arg)  //arg

Where are you looking? Which version?

cbossut, 2017-10-24:
Ok, I'm really sorry, maybe I was on a branch, I looked 3 times in the
header file and did not found it.

Sorry again for the mixup. I should've wait instead of hurrying to post
that.

Thanks anyway for your work.

closed 2017-10-24 by cbossut