esp8266/Arduino #3754: Crashes in __ascii_wctomb

suculent, 2017-10-24:
### Basic Infos

#### Hardware
Hardware:			ESP-12S
Core Version:      	2.4.0-rc1

### Description

Crashes while connecting to WiFi in some background threads that are not in my application code. Seems like saving something (username/password?) to SPIFFS on that.

### Settings in IDE

Module:  			Wemos D1
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			default
Flash Frequency:	80Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

[Too complex. See here.](https://github.com/suculent/thinx-firmware-esp8266/blob/master/examples/thinx-lib-esp/thinx-lib-esp.ino)

### Debug Messages

```

Decoding 25 results
0x4021624c: __ascii_wctomb at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/wctomb_r.c line 35
0x4023a5dc: chip_v6_rf_init at ?? line ?
0x4023b7b4: chip_v6_set_chanfreq at ?? line ?
0x4023a4b0: chip_v6_set_chan at ?? line ?
0x402440ec: chip_v6_unset_chanfreq at ?? line ?
0x4023d68c: chip_v6_initialize_bb at ?? line ?
0x402440e4: chip_v6_set_sense at ?? line ?
0x4021ee50: umm_free_heap_size at /Users/sychram/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1743
0x40206bd8: xPortGetFreeHeapSize at /Users/sychram/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/heap.c line 73
0x402365cd: system_get_free_heap_size at ?? line ?
0x40104c36: os_printf_plus at ?? line ?
0x40216504: strchr at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/string/../../../.././newlib/libc/string/strchr.c line 118
0x40211c06: ULtod at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/strtod.c line 187
:  (inlined by) _strtod_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/strtod.c line 322
0x40211c14: ULtod at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/strtod.c line 198
:  (inlined by) _strtod_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/strtod.c line 322
0x40211c2c: ULtod at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/strtod.c line 202
:  (inlined by) _strtod_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/strtod.c line 322
0x40234550: user_uart_wait_tx_fifo_empty at ?? line ?
0x40243320: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x401044e3: wdt_feed at ?? line ?
0x401044c4: wdt_feed at ?? line ?
0x4021ec22: umm_info at /Users/sychram/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1010
0x4021ec22: umm_info at /Users/sychram/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1010
0x4021eb9e: umm_info at /Users/sychram/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 993
```

devyte, 2017-10-25:
@suculent this firmware doesn't really have "background threads", or even threads at all.
At first glance, the error looks like out of memory and/or mem corruption to me.

After a once over at the repo code, I several bad practices:
- You have MQTT_BUFFER_SIZE 512, a class member uint8_t buf[MQTT_BUFFER_SIZE];, two other data members of 128 each, for a total of 768 bytes between the three, and then an assignment from temporary thx = THiNX(apikey);. The total stack size in the ESP is 4K. Assuming this doesn't go over, you probably should check stack size here as a sanity check.
-  Pretty much all of your const char strings are in RAM instead of flash.
- You are using a mixture of WiFi methods and SDK api calls (e.g.: WiFi.mode() and wifi_station_disconnect() )

In addition, you have the following mistakes:
- you are using a string called api_key of length 64, and one of your storage buffers (thx_api_key) for that is also length 64. Then you use strXXX() calls, which rely on the string null terminator, which is not taken into account in the buffer. This will cause mem corruption and/or reading of garbage. The buffer should be size 65, not 64, to have space for the null terminator. 
- you have StaticJsonBuffer<1024> and StaticJsonBuffer<1280> as data members in your class, and these are being reused in class methods. JsonBuffers are NOT meant to be reused. They can't really be cleared, and they can't free memory. They are meant to be used and destroyed. Specifically, in the case of the StaticJsonBuffer, if you go over its size, it will cause mem corruption. Also, I _think_ they may count against the stack size in addition to the 768 bytes mentioned above.

The code has usage errors which are known to result in mem corruption, which is consistent with the error reported. I suggest you debug your project thoroughly. 

Closing as user error.

closed 2017-10-25 by devyte

suculent, 2017-10-26:
Awesome code review. Thanks a million!!!

Odesláno z iPhonu

suculent, 2017-10-27:
So the resulting practices that completely helped me out, thanks to @devyte 's help:

- always check for unused statically allocated arrays (or rather don't do it at all in header file)
- always count +1 for the /0 (NUL) c-string terminator
- move string constants to flash by using F("My debug log mesage.")
- use DynamicJsonBuffers, they can be safely allocated/deallocated on the go
- check the heap often... in my case it did even return minus values in cases the app didn't crashed even before that (and it did).