esp8266/Arduino #3755: ESP8266 crash trying to communicate with Azure

dilthings, 2017-10-25:
Hi All,

This one is related my previous issue #3753, which was solved.

Brief description: I am trying to re-purpose Sonoff TH10/16 for a project of mine. The units I got sometime back works fine and are very stable with my Azure IoT Hub software developed on the Arduino IDE. Recently got a new set of devises from ITEAD and there was the problem of crashing as described and solved in issue #3753. However after the above issue is solved the simple wifi sketches such as a sketch witch constantly prints the MAC address to the serial console works fine. But when I upload the AzureHub sketch, after the initial boot and the housekeeping work, following errors occur. Form there onwards it constantly prints the last line :wr !ERR_OK. This is the first time I saw this error. Any ideas?

-------Serial monitor output------------

Attempting to connect to SSID: mywifi
sta config unchangedscandone
f 0, ....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with mywifi, channel 11
dhcp client start...
wifi evt: 0
..ip:192.168.8.105,mask:255.255.255.0,gw:192.168.8.1
wifi evt: 3
.
Connected to wifi
please start sntp first !
Fetching NTP epoch time failed! Waiting 2 seconds to retry.
Wed Oct 25 04:52:34 2017

Fetched NTP epoch time is: Wed Oct 25 04:52:35 2017

IoT Hub SDK for C, version 1.1.21
deviceId=XXXXXXXXX
IoTHubClient accepted the message for delivery

[hostByName] request IP for: xxx-iot-hub-xxx.azure-devices.net
[hostByName] Host: xxx-iot-hub-xxx.azure-devices.net IP: xxx.xxx.xxx.xxx
:ref 1
ssl/tls1.c:545 malloc 6864, left 12088
State:	sending Client Hello (1)
:wr
:sent 52
:rn 1400
:ww
:rd 5, 1400, 0
:rdi 1400, 5
:rd 1395, 1400, 5
:rdi 1395, 1395
:c0 1395, 1400
:rn 1400
:rd 1400, 1400, 0
:rdi 1400, 1400
:c0 1400, 1400
:rn 441
:rd 441, 441, 0
:rdi 441, 441
:c0 441, 441
State:	receiving Server Hello (2)
State:	receiving Certificate (11)
crypto/bigint.c:1072 realloc 1032, left 7792
crypto/bigint.c:1072 realloc 1028, left 3768
crypto/bigint.c:1072 realloc 2056, left 1608
State:	receiving Certificate Request (13)
State:	receiving Server Hello Done (14)
State:	sending Certificate (11)
:wr !ERR_OK
:wr !ERR_OK
:wr !ERR_OK
:wr !ERR_OK
:wr !ERR_OK
:wr !ERR_OK
:wr !ERR_OK
:wr !ERR_OK
:wr !ERR_OK

devyte, 2017-10-25:
```
State:	receiving Server Hello (2)
State:	receiving Certificate (11)
crypto/bigint.c:1072 realloc 1032, left 7792
crypto/bigint.c:1072 realloc 1028, left 3768
crypto/bigint.c:1072 realloc 2056, left 1608
State:	receiving Certificate Request (13)
State:	receiving Server Hello Done (14)
State:	sending Certificate (11)
```
I suspect this is a heap issue, but I know pretty much nullptr about using TLS and Azure.
Is this with rc1? rc2? latest git?

dilthings, 2017-10-26:
@devyte, which rc1/rc2 are you referring to? Appreciate if you could elaborate.

Tnx

devyte, 2017-10-26:
Release candidate: 2.4.0-rc2. I am asking which version of the core did you test with?

dilthings, 2017-10-26:
@devyte sorry for the confusion, I have installed the esp core through the boards manager and it keep updating as and when new releases are available. I checked the _package_esp8266com_index.json_ file and it says I am running **version 2.3.0** which seems to be the latest release on git.

devyte, 2017-10-26:
No, 2.3.0 is the latest release, and it's very old. Please try latest git. Instructions are on readthedocs.

dilthings, 2017-10-26:
I have been trying various things last few hours and decided to upload the sketch on to a Lolin nodemcu board and see its response. I used the same Sonoff board definition I used to upload the sketch as shown below:
[image]

And when uploaded the Lolin board started through the same cycle and went on to give the same error over and over again. This proved that the error is not unique to Sonoff. Then I started to change the settings and when I changed the Flash size, Debug port, Debug method settings the sketch started to work with intermittent crashes. The working config is given below for anyone who is going through the issue:

[image]

Although the sonoff has only 1M flash I still used the 4M(3M SPIFFS) option. Not sure if this could bring out issues later. 

The sonoff also responded well with some intermittent core dumps and crashes. These crashes seems to be due to Azure not accepting the device MAC address as it was not created. Then I created a new device for the MAC address and inserted the connection string in the config. For the Last 10 minutes the sonoff seems to be running without crashes. I am keeping an eye just in case. Will update on the progress.

dilthings, 2017-10-26:
@devyte, being brave after the recent "success" I will also try the latest git as you have mentioned and see how it goes.

dilthings, 2017-10-28:
Its been 36+ hours since I last uploaded the sketch. It is still running, without a single reboot.  Looks stable. Overall the problem has been the configuration of Arduino IDE for the compilation and uploading of the sketch rather than a firmware or sdk issue.

closed 2017-10-28 by dilthings