esp8266/Arduino #3756: TCPIP_DEBUG for lwip isn't working

pouriap, 2017-10-25:
#### Hardware
Hardware:			ESP-201
Core Version:      	2.4.0-rc2

### Description

I am trying to see if I can debug the ARP issue([#2330](https://github.com/esp8266/Arduino/issues/2330)) that some ESPs have. I managed to get the lwip debug output for ARP requests but the problem is not there. So I'm trying to enable debug output for `tcpip.c` but it doesn't work.
It seems that `tcpip.c` does nothing at all? Because when I enable `ETHARP_DEBUG`, I can see the debug output of `etharp.c`. And if I put `os_printf()` in there I can see it in the serial output as well.
But enabling `TCPIP_DEBUG` does nothing. I added `os_printf()` to `tcpif.c` and it also isn't being outputted. So I decided to delete `tcpip.c`, and to my surprise everything was compiled and the ESP was working. I thought maybe `liblwip.a` is the cause, so I deleted that as well, and again to my surprise everything is compiled with no problem and the ESP is working fine.
I searched the whole sources and the only place the function `ethernet_input()` is being called is in `tcpip.c`. And I know it is being called because my `os_printf()` inside `ethernet_input()` is outputting my custom message to the serial output.
How come I can delete `liblwip.a` and nothing happens?
How can I get the `TCPIP_DEBUG` info? (And possibly modify `tcpip.c`)

### How I compile
I go to `tools/sdk/lwip/src` and `make clean`, then `make install`
A `liblwip_src.a` is created. Then I rename it to `liblwip_gcc.a` and copy(overwrite) it to `tools/sdk/lib`

### Other info
The file `tools/sdk/lwip/src/api/tcpip.c` is deleted
The file `tools/sdk/lib/liblwip.a` is deleted
Debug messages of `etharp.c` are displayed and every change I make there is applied.
ESP is working as before even though those two files are deleted

d-a-v, 2017-10-25:
You might use TCP_DEBUG instead.
From what I understand TCPIP_DEBUG is for the socket API, while the other is for the RAW API which we use here.

pouriap, 2017-10-28:
Thanks.

For other noobs like me encountering this problem: 
lwip has three APIs: the **raw** API, the **netconn** API and the **socket** API. The raw API is callback-based. But the two others are sequential and are used in threaded applications. ESP8266 uses the **raw** API. And the file [tcpip.c](https://github.com/esp8266/Arduino/blob/e71ec77a001e917ddbc1e8e722eb7f80a4cf6f80/tools/sdk/lwip/src/api/tcpip.c#L3) is used by the sequential API.

@d-a-v before I close the issue can you answer my other question too? I know it's not totally relevant to the issue title but since I've already asked it...
What does [liblwip.a](https://github.com/esp8266/Arduino/blob/master/tools/sdk/lib/liblwip.a) do and why deleting it doesn't cause any problem?

d-a-v, 2017-10-28:
I can say that:
- There is ```lib/liblwip_gcc.a``` which is the open source version compiled by @igrr from ```lwip/src/```. It is the one that every one uses.
- There is also the historical ```lib/liblwip.a``` provided by the espressif SDK accessible only from the ```coredev``` board.
- And there is  ```liblwip_src.a``` too accessible only from the ```coredev``` board that would be your version of lwip if you compile it from ```lwip/src``` and copy it to ```lib/```.

If ```coredev``` is not used, then one usually overwrites ```lib/liblwip_gcc.a``` by his own ```lwip/src/liblwip_src.a``` if lwip gets recompiled (that's what I do, and what you did).

@igrr corrects me if I'm wrong.

devyte, 2018-01-07:
@d-a-v @pouriap is anything further needed here? If not, please close.

closed 2018-01-07 by d-a-v