esp8266/Arduino #3757: ESP rst cause 4, boot mode (3,6)

Ryojikn, 2017-10-26:
### Basic Infos

#### Hardware
Hardware: ESP-07
Vendor: AI Thinker

### Description

I'm trying to run a simple program, uploading through Arduino IDE v1.8.5 and through a UART adapter.
Also, i'm powering my ESP from 3,3v pin and 5v from Arduino Uno, my serial adapter only provides rx/tx to my PC.

I'm uploading the HelloServer example, and all I can get from Serial Monitor is rst cause 4, boot mode 3,6.

Sometimes it changes to boot mode 1,6, but I still cannot understand why. 

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512K (64 SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Speed: 115200
Reset Method:		?ck / nodemcu?
Debug Mode:  disabled
Debug level: "None"
Port: /dev/ttyUSB1

### Sketch

```#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

const char* ssid = "Kuwae";
const char* password = "57896321";

ESP8266WebServer server(80);

const int led = 13;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp8266!");
  digitalWrite(led, 0);
}

void handleNotFound(){
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void){
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void){
  server.handleClient();
}
```

### Debug Messages

```
tate: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with Kuwae, channel 1
dhcp client start...
ip:192.168.0.104,mask:255.255.255.0,gw:192.168.0.1
```

devyte, 2017-10-27:
@Ryojikn which core version? Please make sure to try with latest git.

Ryojikn, 2017-10-27:
I'm using esp8266 2.3.0, I think this is the latest.
My esp is esp-07 model.

Any idea why this occurs?

devyte, 2017-10-27:
2.3.0 is rather old. Please uninstall and try latest git,  instructions are on readthedocs.

Ryojikn, 2017-10-28:
Why 2.3.0 is old, when it's the only arduino esp core available on the board manager?
You mean that the installation of ESP8266 inside arduino ide is old?

devyte, 2017-10-28:
2.3.0 is the last stable release, but it was released over a year ago. That means over a year of code changes and fixed issues behind.
2.4.0-rc2 is the most recent, it is release candidate 2 for stable release version 2.4.0.
The most recent is latest git. Please try that.

Ryojikn, 2017-10-28:
Just did, removed the 2.3.0 version and installed the new version 2.4.0
```
ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

Also, so you can see, this is the output during upload.

```
esptool v0.4.12 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from  to /dev/ttyUSB1
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/ttyUSB1 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 270640 bytes from /tmp/arduino_build_8585/NewEureka.ino.bin to flash at 0x00000000
	erasing flash
	size: 042130 address: 000000
	first_sector_index: 0
	total_sector_count: 67
	head_sector_count: 16
	adjusted_sector_count: 51
	erase_size: 033000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 15000
	setting timeout 1000
	espcomm_send_command: receiving 2 bytes of data
	writing flash
................................................................................ [ 30% ]
................................................................................ [ 60% ]
................................................................................ [ 90% ]
.........................                                                        [ 100% ]
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader
```

The message says Done Uploading.
Still, have no idea why this occurs, any help? D:

devyte, 2017-10-28:
Try erasing the entire flash with esptool, then upload the sketch. After flashing over serial, physically reset your board. Post the full boot message, there is a checksum msg on boot that needs to be checked.
Try removing MDNS from your code. Execution doesn't seem to pass that point.
Are you _sure_ you have latest git? There was an mdns bug that caused a crash in 2.3.0, but it should be fixed in latest git. Make sure by printing out the return of ESP.blah() methods, like the sdkVersion or coreVersion (I don't remember the ESP class method names off the top of my head, check ESP.h for the specifics).

devyte, 2017-10-28:
Also, rst cause 4 is watchdog, it's possible that it's just taking the wifi too long to connect. Try adding a yield() just before MDNS.begin() call.

Ryojikn, 2017-10-31:
This is the checksum that I'm receiving:

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
ve71ec77a
~ld

Ryojikn, 2017-10-31:
After trying to print out both Sdk and Coreversion through Serial.println(ESP.getSdkVersion()); and the same for getCoreVersion, this is the response:

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
ve71ec77a
~ld

Exception (0):
epc1=0x40233540 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe8cbb depc=0x00000000

ctx: sys 
sp: 3ffff7b0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffff950:  40233540 00000138 00000188 40233529  
3ffff960:  00000000 00000000 00000000 00000032  
3ffff970:  402335fc 00000019 00000001 00000000  
3ffff980:  00000138 00000000 3ffee9f8 00000000  
3ffff990:  00003332 3ffff9d0 3ffeea2a 402300af  
3ffff9a0:  00000000 00000006 00000001 3ffeea2a  
3ffff9b0:  00000006 00000000 00000001 00000001  
3ffff9c0:  00000001 00000000 00000002 40006784  
3ffff9d0:  00000000 3ffee9e8 3ffe8480 4022e205  
3ffff9e0:  00000000 40230640 00000000 3ffee9e8  
3ffff9f0:  00000006 00000001 00000004 4022e205  
3ffffa00:  1b161513 19161310 00000000 00000000  
3ffffa10:  00004053 4022e17c 00000d26 40231d24  
3ffffa20:  00000bf0 40231d39 3ffee9e8 00000001  
3ffffa30:  3ffeeb3a 3ffee9f8 00000001 00000003  
3ffffa40:  3ffee9e8 60000600 60000200 00000000  
3ffffa50:  4022da23 00000000 00000001 3ffeeb3a  
3ffffa60:  40105a22 60000e00 077e07be 077e077e  
3ffffa70:  40232a3c 05010307 02050102 fe8d130c  
3ffffa80:  00000000 b200fa29 fe5ccf7f f7e81309  
3ffffa90:  00000000 00007093 3fff07cc 40227fc7  
3ffffaa0:  3ffee54e 60000e00 3ffee52c 3ffee54e  
3ffffab0:  00000001 00000001 00000000 00000000  
3ffffac0:  4022892b 3fff07cc 00000000 402288d3  
3ffffad0:  3fff07ac 00001000 4010489e 0000007d  
3ffffae0:  000000d7 00000020 3fffff10 00000000  
3ffffaf0:  3fff084c 00000000 00000003 34fcbfcc  
3ffffb00:  32524556 00000000 ffff0003 00000005  
3ffffb10:  6177754b efef0065 efeffefe efeffefe  
3ffffb20:  efeffefe 002580fe 00001c00 00000000  
3ffffb30:  00030503 35000001 36393837 00313233  
3ffffb40:  feefeffe feefeffe 3fff1694 40201d81  
3ffffb50:  00000000 00000001 feefeffe feefeffe  
3ffffb60:  feefeffe feefeffe feefeffe 00002580  
3ffffb70:  0000001c 00000000 147937a9 9dc081d6  
3ffffb80:  27a1c98d 8477f02a 091d2e01 d1ae639a  
3ffffb90:  29d8d1a8 e8e15630 00000000 00000000  
3ffffba0:  ffffffff ffffffff 4ac000ff 28903600  
3ffffbb0:  00000005 6177754b 3fff0065 3ffef7c8  
3ffffbc0:  00000000 3ffef8b0 00000010 3fff1ad0  
3ffffbd0:  00000000 3ffef800 00000010 3fff22fc  
3ffffbe0:  00000000 4010053d 00000100 40004aa0  
3ffffbf0:  00000100 00000000 00000010 00000001  
3ffffc00:  3ffe891d 40209028 00000010 00000004  
3ffffc10:  3ffe85d4 ffffff00 ffffffff ffffffff  
3ffffc20:  ffffffff ffffffff ffffffff ffffffff  
3ffffc30:  ffffffff 000001ff ffffff04 ffff0001  
3ffffc40:  00000005 6177754b efef0065 efeffefe  
3ffffc50:  efeffefe efeffefe 002580fe 00001c00  
3ffffc60:  00000000 39383735 31323336 efeffe00  
3ffffc70:  efeffefe ff1694fe 201d813f 00000040  
3ffffc80:  00000100 efeffe00 efeffefe efeffefe  
3ffffc90:  efeffefe efeffefe 002580fe 00001c00  
3ffffca0:  00000000 ffffffff ffffffff ffffffff  
3ffffcb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffcc0:  ffffffff ffffffff ffffffff ffffffff  
3ffffcd0:  ffffffff ffffffff ffffffff ffffffff  
3ffffce0:  ffffffff ffffffff ffffffff ffffffff  
3ffffcf0:  ffffffff ffffffff ffffffff ffffffff  
3ffffd00:  ffffffff ffffffff ffffffff ffffffff  
3ffffd10:  ffffffff ffffffff ffffffff ffffffff  
3ffffd20:  ffffffff ffffffff ffffffff ffffffff  
3ffffd30:  ffffffff ffffffff ffffffff ffffffff  
3ffffd40:  ffffffff ffffffff ffffffff ffffffff  
3ffffd50:  ffffffff ffffffff ffffffff ffffffff  
3ffffd60:  ffffffff ffffffff ffffffff ffffffff  
3ffffd70:  ffffffff ffffffff ffffffff ffffffff  
3ffffd80:  ffffffff ffffffff ffffffff ffffffff  
3ffffd90:  ffffffff ffffffff ffffffff ffffffff  
3ffffda0:  ffffffff ffffffff ffffffff ffffffff  
3ffffdb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffdc0:  ffffffff ffffffff ffffffff ffffffff  
3ffffdd0:  ffffffff ffffffff ffffffff ffffffff  
3ffffde0:  ffffffff ffffffff ffffffff ffffffff  
3ffffdf0:  ffffffff ffffffff ffffffff ffffffff  
3ffffe00:  ffffffff ffffffff ffffffff ffffffff  
3ffffe10:  ffffffff ffffffff ffffffff ffffffff  
3ffffe20:  ffffffff ffffffff ffffffff ffffffff  
3ffffe30:  ffffffff 04080000 001c110c 01010000  
3ffffe40:  ffffffff 00000003 ffffff00 ffffffff  
3ffffe50:  ffffffff ffffffff ffffffff ffffffff  
3ffffe60:  ffffffff ffffffff ffffffff ffff0064  
3ffffe70:  00000000 00000000 ffffffff ffffffff  
3ffffe80:  ffffffff ffffffff ffffffff ffffffff  
3ffffe90:  ffffffff ffffffff ffffffff ffffffff  
3ffffea0:  ffffffff ffffffff ffffffff ffffffff  
3ffffeb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffec0:  ffffffff ffffffff ffffffff ffffffff  
3ffffed0:  ffffffff ffffffff ffffffff ffffffff  
3ffffee0:  ffffffff 147937a9 9dc081d6 27a1c98d  
3ffffef0:  8477f02a 091d2e01 d1ae639a 29d8d1a8  
3fffff00:  e8e15630 ffffffff ffffffff ffffffff  
3fffff10:  ffffffff ffffffff ffffffff ffffffff  
3fffff20:  ffffffff ffffffff ffffffff ffffffff  
3fffff30:  ffffffff ffffffff ffffffff ffffffff  
3fffff40:  ffffffff ffffffff ffffffff ffffffff  
3fffff50:  ffffffff ffffffff ffffffff ffffffff  
3fffff60:  ffffffff ffffffff ffffffff ffffffff  
3fffff70:  ffffffff ffffffff ffffffff ffffffff  
3fffff80:  ffffffff ffffffff ffffffff ffffffff  
3fffff90:  cf5cffff fe29fa7f 006f5b8c 5d7482b2  
3fffffa0:  000201e9 e79d8826 ffffff01 55aa55aa  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0xef
csum 0xef
csum err
ets_main.c 
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
ets_main.c

igrr, 2017-10-31:
Given the randomness this looks very much like a power supply/decoupling
issue. Do you have an integrated board (such as nodeMCU) to try this on?

Ryojikn, 2017-10-31:
No, my current config is the following:

I have one Arduino Uno connected to my USB port serving a level shifter, which is regulating 5v to 3,3v, and this, is providing the 3,3v needed for the ESP.

On the other hand, I have the ESP connected to a USB Serial Adapter, and this to my USB port.

Would this be the problem? Also, all my tests are being done with jumpers, bcuz I'm still prototyping before I can move to production and develop my board.

Ryojikn, 2017-11-02:
No, my current config is the following:

I have one Arduino Uno connected to my USB port serving a level shifter, which is regulating 5v to 3,3v, and this, is providing the 3,3v needed for the ESP.

On the other hand, I have the ESP connected to a USB Serial Adapter, and this to my USB port.

Do you think this is the problem? Also, all my tests are being done with jumpers, bcuz I'm still prototyping before I can move to production and develop my board.

devyte, 2017-11-02:
@Ryojikn it's possible. I use a 5V 20A power supply for prototyping to make sure any odd issues that pop up like this are not due to power supply.
Or you could have some other wiring issue.
Closing, as this is not the right place to ask for help with your prototyping. If you need further help, please ask at a community forum. If you do narrow down to a problem in the core, please open a new issue.

closed 2017-11-02 by devyte

Prakashmj, 2018-05-10:
Hi There, As a professional programmer I too have faced this kind of issues with ESP8266 like,
ets Jan  8 2013,rst cause:4, boot mode:(1,6)
ets Jan  8 2013,rst cause:4, boot mode:(3,6) and much more. But my advice is very basic, read your program twice or more, concentrate on the payload what you are decided to give. 

The solution what I found that, If the payload is even a bit higher, you may have this kind of issues with ur program. To conclude, split the load, give some time to do the next set of executions in the program. Like follows.,
mqtt.publish("{\"position.latitude\":\"11.3410364\",\"position.longitude\":\"77.71716420000007\",\"status\":\"online\",\"ident\":\"001\",\"Temperature\": \"30\",\"conductivity\":\"0.01\",\"liquid flow\":\"74\",\"FDV\":\"ON\",\"pressure\":\"121\",\"sensor1\": \"1121\",\"sensor2\":\"0.01\",\"sensor3\":\"74\",\"sensor4\":\"ON\",\"sensor5\":\"121\",\"sensor6\": \"185\"}");

above mentioned will be split into two or three statements, 

mqtt.publish("{\"conductivity\":\"0.01\",\"liquid flow\":\"74\",\"FDV\":\"ON\",\"pressure\":\"121\",\"sensor1\": \"1121\"}"); delay(2000);
and followed by the next remaining things in a shorter way. 

Thank you for reading my post...Cheers!!!