esp8266/Arduino #3758: OTA Update failed

DevUrandomFoo, 2017-10-26:
### Basic Infos

#### Hardware
Hardware:		ESP8266MOD  (ESP-07)
Vendor:                   AI-THINKER
Core Version:      	?

#### Flash Info
Flash real id:   001440E0
Flash real size: 1048576
Flash ide  size: 1048576
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration ok.

### Description

OTA does not work. Some month ago, it worked fine. For testing purpose, I've uploaded the original OTA-Sample to my ESP but it doesn't work.
IDE say:

09:03:56 [ERROR]: No response from device
09:03:56 [ERROR]: No response from device

OTA + Update Debug see below. I testet it on 2 different ESP's but same problem.

### DEBUG

Booting: OTA
Version: 2
OTA server at: esp8266-11cee6.local:8266
                                        Ready
IP address: 10.1.0.86
pm open,type:2 0
sleep disable
[begin] roundedSize:       0x0003D000 (249856)
[begin] updateEndAddress:  0x000BB000 (765952)
[begin] currentSketchSize: 0x0003D000 (249856)
[begin] _startAddress:     0x0007E000 (516096)
[begin] _currentAddress:   0x0007E000 (516096)
[begin] _size:             0x0003CFC0 (249792)
Start
Connect Failed
Error[2]: Connect Failed
premature end: res:0, pos:0/249792
Error[4]: End Failed
ERROR[0]: No Error

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:		40Mhz
Flash Mode:			DIO
Flash Frequency:	        80Mhz
Upload Using:		?OTA / SERIAL?
Reset Method:		ck

### Sketch
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "xxx";
const char* password = "xxx";

String Sketch = "OTA";
String Version = "2";

void setup() {
  Serial.begin(9600);
  Serial.println("");
  Serial.print("Booting: ");
  Serial.println(Sketch);
  Serial.print("Version: ");
  Serial.println(Version); 
  
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 8266
  // ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
  // ArduinoOTA.setHostname("myesp8266");

  // No authentication by default
  // ArduinoOTA.setPassword((const char *)"123");

  ArduinoOTA.onStart([]() {
    Serial.println("Start");
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();

  delay(1);
}

scropion86, 2017-10-26:
i have the similar problem using 2.3.0 Stable also 2.4 RC2 , i tried both
Using Arduino IDE and PlatformIO

DevUrandomFoo, 2017-10-26:
I just tried it with another Laptop and fresh installed arduino IDE but same problem. Also tried 2.3.0 as well as 2.2.0.

DevUrandomFoo, 2017-10-26:
When using the ESP8266HTTPUpdateServer Class and upload the new Sketch via Webbrowser, OTA will work for me.
Updating with "python espota.py -i 10.1.0.86 -p 8266 --auth= -f BasicOTA.ino.bin" still doesn't work :(

scropion86, 2017-10-26:
@DevUrandomFoo so updating through HTTP Server will work , but though IDE(or esptool) still not working for you ?!
i hope one of the maintainers look into it as i am in the middle of building a POC and without OTA the entire thing will not work as it should.

igrr, 2017-10-26:
If it doesn't work for you with 2.3.0, then there's a good chance you need
to open a port in your firewall or otherwise allow ESP connecting to a
random port on your PC.

DevUrandomFoo, 2017-10-26:
@igrr You made my Day! It was a Firewall issue by my operation system. Maybe someone want to add a hint in the documentation. Or better, resign the transfer to be more firewall friendly :)

closed 2017-10-26 by DevUrandomFoo

dcfusor, 2017-12-02:
Having the same problems here - neither the OTA example or the httpupdate one work more than once or twice, then fail from every machine on my network.  There is no firewall (most are running linux mint/mate 18.2), the ESP's are doing other UDP things, are ping-able, and show up on arp-scan, the arduino IDE in the case of the OTA, or the web page in the case of http updater.  But OTA fails with "17:19:53 [ERROR]: No response from device" and http updater just says "Update Failed" when it's been burned in (all burns via serial port).  Adding debug to the httpupdater gives me a bunch of errors on flash config wrong - real (#for 1 meg), IDE (#for 4 megs) on a module with 4 megs.
Either work the first couple of times, then never again.  It's a big deal for my lan of things as some will be going into what is better habitat for actual bugs that bite (spiders).
It seems every report of this gets closed, but it shows up again and again.
Ditto the issue with "can't change hostname" = nothing actually works but someone says it did once and the issue gets closed.  Might be related.

dcfusor, 2017-12-03:
Partly my bad....
Turns out an ESP-07 (which some dox say is the same as an ESP12e) isn't the same, and only has 1M of flash.  This can be properly selected under generic ESP module.  We dung into the basic compare and flash config calls to discover our error there.  Also, GPIO15 does in fact need a pulldown or reset is unreliable.

Still can't get the hostname to really change, though some landns code I wrote can broadcast anything, which oddly also shows up in arp-scan as my other half of landns puts in in host, and arp-scan takes the easy way out...
But that's OT on this thread.  Sorry for the rant.

dcfusor, 2017-12-03:
What IS interesting and not OT is that OTA (from IDE or browser, depending on the sketch) works despite the wrong sizing at least once, sometimes a few times....leading one astray when debugging what turned out ultimately to be my error.