esp8266/Arduino #3759: Buzzer issues annoying noise after tone()

Pako2, 2017-10-26:
### Basic Infos

#### Hardware
Hardware:	ESP-8266
Core Version:   ???

### Description
I found an error in the file **Arduino/cores/esp8266/Tone.cpp**.
There are two lines:
```
133 disableTimer (T1INDEX);
134 digitalWrite (tone_pins [T1INDEX], LOW);
```
### I think these should be in the opposite order.
In the current state, the second line has no function, because during execution of the command **disableTimer(T1INDEX)**, the variable **tone_pins [T1INDEX]** is overwritten to 255 (at line 99).

devyte, 2017-10-27:
@Pako2 which core version? Did you test this code change?

Pako2, 2017-10-27:
I'm sorry, before I did not know how to find the core version. Now I've already found it. This is **2_3_0_RC1**.
Yes, I tested my solution and it works. However, I can not guarantee that it does not have any side effects.  I did not see any.

vdeconinck, 2017-10-27:
2.3.0 RC1 is [18 months old](https://github.com/esp8266/Arduino/releases). Can you upgrade to the latest git, test your changes and report here ? For instructions on installing Git version, see http://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version

Pako2, 2017-10-28:
I have to apologize again.
As I wrote earlier, **I use the GIT version**.
But unfortunately, I was a bit inattentive and I used the version string that is mentioned in the commentary.

So it is completely different:
ESP.getCoreVersion() returns the value 00000000
This means **ARDUINO_ESP8266_GIT_VER**

Pako2, 2017-11-28:
I'm confused.
Here is (for a very long time) label "waiting for feedback".
Did I do something wrong?
Should I add something from my side?

devyte, 2017-11-28:
Nothing wrong, it just means I hadn't come back to this. There are currently 530+ open issues, and dealing with all is a bit overwhelming.
Please make a PR with your suggestion, put a link to this issue in it. I can't really test it, but I can take a look at the code based on that.

Pako2, 2017-11-30:
I assume that "PR" means "Pull Request".
If so, there is a problem. I unfortunately do not understand how to do a PR (my English is miserable and I did not understand the instructions).
Maybe someone could do it instead of me.

devyte, 2018-07-13:
@Pako2 There have been extensive changes merged to tone/pwm/etc recently. Please retest with latest git.

devyte, 2018-07-13:
Also, please advise whether the bug you found is still valid now, and whether your fix is still needed.

earlephilhower, 2018-11-09:
Closing due to lack of reproduction or feedback.

closed 2018-11-09 by earlephilhower