esp8266/Arduino #3761: espconn_recv_hold reset the module when using TLS/SSL

vitorborgo, 2017-10-26:
Hi,
@igrr

- I'm working with ESP8266/ESP-07 with 2.1.0 SDK.
- At the moment, I am developing a project using esp07 as a [TCP <-> UART0] bridge.
- The ESP MCU works as client and connect to a MQTT server.
- I enabled TLS comunication with the server, and its working fine.
- The free heap size is OK too.
- I have implemented 2 FIFOs, one at TCP side (10480 bytes size), and another one at UART0 (2048 bytes).
- At UART0 I have implemented CTS/RTS algorythm, but at the TCP side I had to use espconn_recv_hold (that is the reason of the 10kb buffer)
- I have already tested the firmware as a bridge without TLS/SSL, and works fine. Recv hold api function is triggered when TCP->uart0 FIFO reaches 2kb.
- Tested using a file 44Mb size, transmitting both sides twice. [(55kb/s TCP -> uart0);(42kb/s uart0 ->TCP)].

**The Problem:**
*** When I enable SSL/TLS mode, everything works fine until the FIFO buffer reaches 2kb and my algorythm call function "espconn_recv_hold".
  Then the module crashes and resets with the following message.
*** I made the this test twice: The first one setting the recv hold trigger up than the buffer size (to never hit) and the problems desappear, 
    and the seccond setting the trigger to 256b (what triggs the problem).
    
**Solutions already tested:**
*** Suply voltage was already tested using 3.3V/2A power source.
*** The reset pin is setted to high level all the time.

So I thing it is not energy problem, althought the "rst cause:2"
Im really confused about this issue, and would like to know if it is a bug or I am doing something wrong.

 
_DEBUG SERIAL MESSAGES:_

     ets Jan  8 2013,rst cause:2, boot mode:(3,6)

    load 0x40100000, len 29964, room 16
    tail 12
    chksum 0x98
    ho 0 tail 12 room 4
    load 0x3ffe8000, len 2212, room 12
    tail 8
    chksum 0xbb
    load 0x3ffe88b0, len 8252, room 0
    tail 12
    chksum 0xdf
    csum 0xdf
    ŒãìƒìÛ{‚òn|ì
    lŒìl`c’
    ž|{ÛlÜnà
    ‚nã
l`ŒârÛlŒlŒ

devyte, 2017-10-27:
@vitorborgo The issue template exists for a reason, you provided very few of the requested info. Also, it's very difficult to help you if you don't post a MCVE sketch.

vitorborgo, 2017-10-27:
### Basic Infos

#### Hardware
Hardware:			?ESP-07?
Core Version:      	?2.1.0-rc2?

### Description

When I enable SSL/TLS mode, everything works fine until the FIFO buffer reaches 2kb and my algorythm call function "espconn_recv_hold".

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4Mbit?
CPU Frequency:	?160Mhz?
Flash Mode:		?qio?
Flash Frequency:	?26Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck?

### Sketch

Sorry for not provide all the info. The project is very long, so i will post the code that matters.
I'm using last SDK 2.1.0

Here is the task I call every ms (or when data arrives to the bridge) to redirect the FIFOs data.
Im using ConfigRAM.SSL.security = true to work with TLS/SSL (already correct initialized)

```
#define MAX_TCP_WRITE_SIZE  256
#define MAX_TCP_READ_PROCESS_AT_A_TIME 1024
static unsigned int lenghtSendTCP = 0;
void ICACHE_FLASH_ATTR tcp_task()
{
	unsigned int i;
	unsigned char data;
	static bool pendingSendTCP = false;
	static uint8 TCPOutBuffer[MAX_TCP_WRITE_SIZE];

	struct espconn* serverTCP = getServerConnTCP();

	// (TCP -> UART0) ---------------------------------------------------------
	i = 0;
	while(i < MAX_TCP_READ_PROCESS_AT_A_TIME && TCP_RX_GetUsedFIFO() && GPIO_INPUT_GET(GPIO_ID_PIN(CTS_PIN)) == false)
	{//quando tiver dado na Fifo e o pino de CTS estiver em nível lógico baixo
		if(TCP_RX_GetDataFIFO(&data) == TRUE)
		{
			uart0_write_char(data);
			i++;
		}
	}

	if(TCP_RX_GetUsedFIFO() < (RECV_HOLD_TRIGGER/2) && get_recv_hold() == true)
	{
		espconn_recv_unhold(serverTCP);
		set_recv_hold(false);
	}
	// ------------------------------------------------------------------------

	// (UART0 -> TCP) ---------------------------------------------------------
	if (serverTCP->state == ESPCONN_CONNECT ||
		serverTCP->state == ESPCONN_WRITE ||
		serverTCP->state == ESPCONN_READ)
	{
		if (RXFIFO_GetUsed() || lenghtSendTCP > 0)
		{
			if (lenghtSendTCP == 0)
			{
				while(RXFIFO_GetUsed() && lenghtSendTCP < MAX_TCP_WRITE_SIZE) //-> Dont get more data from FIFO if there is remaining data to send
				{
					if (RXFIFO_GetData(&data))
					{
						TCPOutBuffer[lenghtSendTCP++] = data;
					}
				}
			}

			if(ConfigRAM.SSL.security == false)
			{
			   if(espconn_send(serverTCP, (uint8_t*)TCPOutBuffer, lenghtSendTCP) == ESPCONN_OK)
				   lenghtSendTCP = 0;
			}
			else
			{
				if (readyToSend == true)
				{
					sint8 answ;
					static sint8 aux;

					answ = espconn_secure_send(serverTCP, (uint8_t*)TCPOutBuffer, lenghtSendTCP);
					if(aux != answ)
					{
						aux = answ;
						INFO("SECURE SEND %db, return code: %d\r\n",lenghtSendTCP, aux);
					}

					if(answ == ESPCONN_OK)
					{
						readyToSend = false;
						lenghtSendTCP = 0;
					}
				}
			}
		}
	}

	// ------------------------------------------------------------------------
}
```

Here is the TCP receive callback function where recv_hold is called 
(The RECV_HOLD_TRIGGER is 2kb size and the FIFOs buffer is 10kb size)

```
#define RECV_HOLD_TRIGGER 2048
static void ICACHE_FLASH_ATTR serverRecvCb(void *arg, char *data, unsigned short len)
{

	//struct espconn *conn = (struct espconn *) arg;

	blink_LED(); //sinaliza que recebeu

	if (command_mode == DISABLED_CMD_MODE && os_strncmp(data,CMD_MODE_KEY,sizeof(CMD_MODE_KEY)-1) == 0)
	{
		command_mode = TCP_CMD_MODE;
		start_cmd_mode(serverConnTCP);
	}

	//INFO("RecvCB: recvhold - %d, data len - %d\r\n", recv_holded, len);

	if (command_mode == TCP_CMD_MODE)
		config_parse(data, len);
	else
		TCP_RX_SendStringToFIFO(data,len); //TCP RX -> uart0 TX

	if(TCP_RX_GetUsedFIFO() > RECV_HOLD_TRIGGER && if(TCP_RX_GetUsedFIFO() > RECV_HOLD_TRIGGER && get_recv_hold() == false))
	{
		set_recv_hold(true);
		espconn_recv_hold(serverConnTCP);
	}

	INFO("Received Data Cb\r\n");
}
```

Hope there is information enough now.
Thanks.

igrr, 2017-10-27:
That doesn't seem to be related to Arduino core... You are using espconn,
which is part of non-OS SDK. Arduino equivalent would be
WiFiClient/WiFiClientSecure.

vitorborgo, 2017-10-27:
Yes, it's right. Sorry, I think I posted it the issue on the wrong topic. Im working with NONOS SDK. I will close this issue and respost to the right topic.

igrr, 2017-10-30:
non-OS SDK issue tracker indeed seems to be the best place for this question, as it is related to non-OS SDK.

closed 2017-10-30 by igrr

devyte, 2017-10-30:
@igrr do we want/have hold functionality in out sockets? I remember a discussion in the nodemcu firmware repo some time ago about them implementing this to avoid overrunning some buffer or something. But then they got rid of the espconn layer altogether, so I'm not even sure if it was necessary.

igrr, 2017-10-31:
Same here, we also don't use espconn layer. I don't think this is necessary.