esp8266/Arduino #3764: bcn_timout,ap_probe_send_start when doing WiFi.getMode()

mamama1, 2017-10-27:
Hardware:	NodeMCU by RobotDyn
Core Version: 2.3.0 and 2.4.0 rc2

### Description

When doing WiFi.getMode() more frequently (currently every second), the following message appears and Wifi connection is lost:

`bcn_timout,ap_probe_send_start`

Depending on what else the sketch does, Wifi gets reconnected and stays connected or it reconnects and immediately disconnects again (in an endless loop). If the latter occurs, only 1 to 2 pings come through and any IP related things in the sketch don't receive any packets.

My workaround is to check WiFi.getMode() once and save the result to a variable. Issue doesn't occur anymore but it took me a hell of a time to find that cause in my 4k+ lines sketch, doing a vast amount of things every loop pass.

Debugging was kind of unpredictable too, because (seemingly) completely unrelated things sometimes altered if the connection got lost for good or if it returned. I guess those things just put some delay somewhere which allowed everything to calm down and survive another "WiFi.getMode()".

I also have to say, that I had "WiFi.getMode()" in place for quite a while and the issue started to occur out of nowhere, when changing things completely unrelated to any WiFi code.

I have triple-checked my code for memory violations caused by too small buffers, memcpy's and so on.

Don't know if anyone would be able to reproduce the issue with only a simple loop checking "WiFi.getMode()" every second. This might be related to timing and coincidences in my code.

Don't think that anyone would like to inspect my whole code (and too, I'd prefer not to publish it, for now) but maybe this is expected/known or rings a bell in someone's head?

Anyone experienced something like this before?
Any idea?

Thanks!

### Settings in IDE

Module: NodeMCU 1.0 (ESP-12E Module)
Flash Size: 4M
CPU Frequency:	80Mhz
Flash Mode:	?
Flash Frequency:	?
Upload Using:	SERIAL/USB
Reset Method:	?

### Sketch
WiFi.getMode()

### Debug Messages
bcn_timout,ap_probe_send_start

vdeconinck, 2017-10-27:
> Don't know if anyone would be able to reproduce the issue with only a simple loop checking "WiFi.getMode()" every second
Did you test that on your hardware ? Does it also exhibit the issue ? If not, could you increasingly add code from your project until it breaks ? In other words, a MCVE would be welcome...

devyte, 2017-11-01:
@mamama1 This does not ring a bell, and I read through about 1200 opened issues. Also, I check the mode quite often in my app, and I've never seen it. Although I don't think I check it every second...
Without a (mcve) sketch to reproduce the problem it is unlikely this can be pursued. Can you provide it?

devyte, 2017-11-01:
@mamama1 a thought: try disabling interrupts right before checking the mode, and enabling them again right after. Careful, though, make sure the code can't follow a path where they remain disabled!
I seem to remember this in somebody's code. I wondered about the why briefly, which is why it stuck in my brain.

tablatronix, 2017-11-04:
Never saw this either. Are you calling this inside some interrupt routine or some callback?

mamama1, 2017-11-04:
no i'm not using interrupts.. @devyte but i guess the ESP8266 core aka RTOS(is it RTOS?) uses interrupts internally.

however i think this is in conjunction with visualmicro debugging functionality because when i compile using the arduino ide instead of visualmicro (even if I've disabled all debugging features completely), it works. also the files are not the same and according to the visualmicro developer who is giving me great support, it seems that the files are differently linked when compiling using visualmicro...

devyte, 2018-09-06:
I hadn't seen that last comment. If it works with IDE, and not with visualmicro, then the issue is not in the core, and most likely with something visualmicro is doing, or not doing. 
Closing as not actionable.

devyte, 2018-09-06:
Oh, and we're not using RTOS here, this core is based on the NONOS sdk.

closed 2018-09-06 by devyte