esp8266/Arduino #3766: OTA Update stop responding

silviob72, 2017-10-28:
### Basic Infos

#### Hardware
Hardware:			ESP8266 WeMos D1
Core Version:      	?

### Description

When I try to upload the sketch through OTA, the ESP stop responding at random progress and never recover until manual reset. 
The problem could be my network configuration but why does ESP stop responding?
Is there any way to force restart the ESP in this situation?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			DOUT
Flash Frequency:	40Mhz
Upload Using:		OTA
Reset Method:		nodemcu

### Sketch
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "ssid";
const char* password = "pass";

IPAddress ip(192,168,101,15);
IPAddress gateway(192,168,100,2);
IPAddress subnet(255,255,254,0);
IPAddress dns(8,8,8,8);

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.config(ip, gateway, subnet, dns);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 8266
  // ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
  // ArduinoOTA.setHostname("myesp8266");

  // No authentication by default
  // ArduinoOTA.setPassword((const char *)"123");

  ArduinoOTA.onStart([]() {
    Serial.println("Start");
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r\n", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();
}

### Debug Messages

Arduino IDE:

Sketch uses 245075 bytes (23%) of program storage space. Maximum is 1044464 bytes.
Global variables use 33576 bytes (40%) of dynamic memory, leaving 48344 bytes for local variables. Maximum is 81920 bytes.
python.exe C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/espota.py -i 192.168.101.15 -p 8266 --auth= -f C:\arduinoHex/BasicOTA.ino.bin 
Uploading..............................................
12:44:42 [ERROR]: Error Uploading
12:44:42 [ERROR]: Error Uploading

ESP8266 Serial:

OTA server at: esp8266-d534cb.local:8266
Ready
IP address: 192.168.101.15
sleep disable
Start
Progress: 0%
pm open,type:0 0
Connect Failed
Error[2]: Connect Failed
Error[4]: End Failed
ERROR[0]: No Error
sleep disable
Start
Progress: 0%
Progress: 0%
Progress: 1%
Progress: 1%
Progress: 2%
Progress: 2%
Progress: 3%
Progress: 4%
Progress: 4%
Progress: 5%
Progress: 5%
Progress: 6%
Progress: 7%
Progress: 7%
Progress: 8%
Progress: 8%
Progress: 9%
Progress: 9%
Progress: 10%
Progress: 11%
Progress: 11%
Progress: 12%
Progress: 12%
Progress: 13%
Progress: 14%
Progress: 14%
Progress: 15%
Progress: 15%
Progress: 16%
Progress: 16%
Progress: 17%
Progress: 18%
Progress: 18%
Progress: 19%
Progress: 19%
Progress: 20%
Progress: 21%
Progress: 21%
Progress: 22%
Progress: 22%
Progress: 23%
Progress: 24%
Progress: 24%
Progress: 25%
Progress: 25%
Progress: 26%

lrmoreno007, 2018-01-04:
Possible duplicate of #3881

devyte, 2018-01-04:
@silviob72 Please retest with latest git.

devyte, 2019-11-06:
No feedback is a long time. Also, this thread is old, a lot has been fixed since it was opened.
Closing.
If you still encounter the problem, please open a new issue and follow the template instructions.

closed 2019-11-06 by devyte