esp8266/Arduino #3773: DHCP Transaction Id not random

TylerDurden23, 2017-10-29:
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:	ESP-12E
Core Version:      	2.4.0-rc1

### Description

The DHCP Transaction Id generated by the ESP8266es does not seem to be random - as required by rfc2131. That generates strange connection issues when multiple ESP8266es are speaking with the same DHCP Server. The ESP seems always to start with the transaction ID 0xabcd0001 - and only later seems to be counting up the ID. The correct behaviour would be always to use random Ids - or at least as the initial value.

Depending on the implementation. the server may ignore the dhcp request (because for the server, the transaction may be in another state and the request may be invalid).

For the ESP users, this shows as stange connection/reonnection issues.

devyte, 2017-10-29:
@TylerDurden23 do you have a wireshark dump or something else that shows this?
Does #3362 help?

TylerDurden23, 2017-10-29:
[image]

devyte, 2017-11-03:
@TylerDurden23 lwip2 is now merged, it can be selected from the Arduino IDE menu. Does using it help?

TylerDurden23, 2017-11-03:
Is this version available via Boardmanager?

devyte, 2017-11-03:
You have to uninstall any version you have installed via boardmanager, and install git (instructions are on readthedocs). Then restart the Arduino IDE, and you should see the option for lwip in the menu.
Be aware that the ESPAsync libs don't yet build with lwip2, in case you're using them.

TylerDurden23, 2017-11-03:
@devyte  The DHCP request id Looks random now - my project uses the pubsub library - that does not seem to work (async issue most likely).

Regards.

devyte, 2017-11-15:
Given that the original issue seems fixed with lwip2, I'm closing.
I understand that pubsub is a 3rd party lib. If it shows a problem, please open a new issue, and fill in the fields. Please be aware that in that case the problem should be narrowed down to relevant code in this repo, so most likely you will need help from pubsub's author first.

closed 2017-11-15 by devyte