esp8266/Arduino #3774: configTime ignores timezone in rc2

dragondaud, 2017-10-30:
### Hardware

Hardware:	ESP-12
Core Version:	2.4.0-rc2

### Description

After upgrading from 2.4.0-rc1 to 2.4.0-rc2, I noticed the timezone parameter seems to be ignored by configTime, and ctime() returns only GMT. The included sketch demonstrates this problem, and when compiled with rc1 or earlier it shows the correct time output, based on TZ setting. In rc2, the same sketch shows only GMT. If there is an expected change in behavior, I could find no documentation.

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Upload Using:		SERIAL

### Sketch

```
#include <ESP8266WiFi.h>
#include <time.h>

#define WIFI_SSID "SSID"
#define WIFI_PASS "PASS"
#define TZ -6

void setup() {
  Serial.begin(115200);
  while (!Serial);
  Serial.println();
  Serial.print("Last reset: ");
  Serial.println(ESP.getResetReason());
  Serial.print("Connecting to ");
  Serial.print(WIFI_SSID);
  Serial.print("...");
  WiFi.begin(WIFI_SSID, WIFI_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("Synchronize with NTP...");
  configTime((TZ * 3600), 0, "pool.ntp.org", "time.nist.gov");
  while (!time(nullptr)) {
    delay(1000);
    Serial.print(".");
  }
  Serial.println();
}

void loop() {
  time_t now = time(nullptr);
  String t = ctime(&now);
  t.trim();                       // ctime() output ends with \n, bug?
  long heap = ESP.getFreeHeap();
  Serial.print(heap);
  Serial.print(" free heap @ ");
  Serial.print(t);
  Serial.println();
  delay(5000);
}
```

5chufti, 2017-10-31:
since there are more than one time lib and the only one distributed via library manager does not have configTime()
we don't know what dependencies could be the culprit, maybe it is just a time.h conflict.
As a/the time lib is not part of the arduino esp8266 core, there most probably are other places to get this fixed as a/the time lib should not be affected by change of esp8266 sdk.

dragondaud, 2017-10-31:
Looks like its part of the core to me:

./cores/esp8266/time.c:void configTime(int timezone, int daylightOffset_sec, const char* server1, const char* server2, const char* server3)

dragondaud, 2017-10-31:
@d-a-v just noticed this bug only occurs when I'm using lwip2v2 from #3362 but not if I switch back to lwip2 v1.4.

d-a-v, 2017-10-31:
@dragondaud  thanks for reporting. This is now fixed and verified with your sketch.

5chufti, 2017-10-31:
@dragondaud: sorry, never thought on looking there.
For a 3rd party lib to arduino standards I think it is in the wrong place and should be found somewhere like /esp8266/libraries/time

devyte, 2017-11-01:
Closing, as the issue is in an unmerged PR, and also seems to be already fixed. If there are further comments about the time issue in lwop2, please comment in the PR itself.

closed 2017-11-01 by devyte