esp8266/Arduino #3775: Board d1_mini (platform esp8266, package esp8266) is unknown after upgrade to 2.4.0-rc2

reibuehl, 2017-10-31:
### Basic Infos

#### Hardware
Hardware:			ESP8266
Core Version:      	2.4.0-rc2

### Description

I upgraded to 2.4.0-rc2 to get the KRACK fix by replacing the URL in Arduino IDE preferences and installing it via Boardmanager. If I try to recompile, I now get the error:

Board d1_mini (platform esp8266, package esp8266) is unknown

Error compiling for board WeMos D1 R2 & mini.

### Settings in IDE

Module:  			WeMos D1 R2 & mini
Flash Size: 		4MB
CPU Frequency:	80Mhz

reibuehl, 2017-11-03:
After deleting the folder "packages" under \Users\<login>\AppData\Local\Arduino15 and then reinstalling all previously installed board packages via the Board Manager, the error is gone.

closed 2017-11-03 by reibuehl

afortiori85, 2018-10-24:
Same issue here, and this worked for me. Thank you!

Nikospax, 2019-05-14:
Thanks works for me.

wlraider70, 2020-01-06:
You fixed it.