esp8266/Arduino #3776: WiFi.scanNetworks() returns before scan is finished

wahabkhan1892, 2017-10-31:
I am using a WROOM02 board. When it is first powered up, it is configured as softAP and the user is given a list of available WiFi networks to connect to. To obtain the list, I'm calling WiFi.scanNetworks() functions. The problem I'm facing is that it's returning before the scan is finished, i.e. before _scanDone is called. As a result, the device thinks that there are no WiFi networks available because the scanNetworks() function returns 0.

When the device is powered up, and the client connects to the softAP, the home page is automatically opened in the default browser (I'm using captive portal functionality here). There the user has to click a button to call the "/networks" page which is supposed to display a list of networks. But due to the problem mentioned above, I'm getting a empty list from the device. The interesting part is that when I call scanNetworks() in the setup() function, it works as expected and returns the correct number of detected WiFi networks. But when called in the "/networks" callback, it is encountering this error. The code for "/networks" is shown below.
```
    server.on("/networks", HTTP_GET, [](AsyncWebServerRequest *request)
    {
        Serial.println("Calling /networks");
        Serial.print("caught request:");
        String netList = "[";
        int n = WiFi.scanNetworks();
        Serial.print(n);
        for (int i = 0; i < n; ++i)
        {
            // Print SSID and RSSI for each network found
            netList += "{\"ssid\":\"";
            netList += WiFi.SSID(i);
            netList += "\", \"power\":";
            netList += (-WiFi.RSSI(i));
            netList += ", \"secure\":";
            netList += (WiFi.encryptionType(i) == ENC_TYPE_NONE) ? "false" : "true";
            if(i == (n - 1))
                netList += "}";
            else
                netList += "},";
        }
        netList += "]";
        Serial.println(netList);
        request->send(200, "application/json", netList);
    });
```

Pablo2048, 2017-10-31:
The problem is that You used AsyncWebServer so You have to follow rules for async scanning (RTFM @ https://github.com/me-no-dev/ESPAsyncWebServer#scanning-for-available-wifi-networks ). BTW we are off-topic here...

devyte, 2017-10-31:
Previous comment is correct.
Closing as user error and off-topic.

closed 2017-10-31 by devyte