esp8266/Arduino #3778: Warning messages while compiling using the git version of 2.3.0 core

brucemiranda, 2017-10-31:
### Basic Infos

#### Hardware
Hardware:			?ESP-8266 
Core Version:      	?2.3.0

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

```

### Debug Messages

```
                                  ^

In file included from C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:444:0,

                 from sketch\web_server.h:31,

                 from sketch\ota.cpp:28:

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:67:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator ++(int i) { _count++; }

          ^

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:68:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator --(int i) {  if (_count > 0) { _count--; } ;  }

          ^

In file included from C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:444:0,

                 from sketch\web_server.h:31,

                 from C:\Users\ThePC\Documents\Arduino\EmonESP-master\src\src.ino:29:

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:67:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator ++(int i) { _count++; }

          ^

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:68:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator --(int i) {  if (_count > 0) { _count--; } ;  }

          ^

In file included from C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:444:0,

                 from sketch\web_server.h:31,

                 from sketch\web_server.cpp:31:

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:67:10: warning: unused parameter 'i' [-Wunused-parameter]
[... 473 lines cut ...]
     if (qos < 0 || qos > 1) {

               ^

D:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\spiffs\spiffs_hydrogen.c: In function 'SPIFFS_buffer_bytes_for_filedescs':

D:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\spiffs\spiffs_hydrogen.c:16:49: warning: unused parameter 'fs' [-Wunused-parameter]

 u32_t SPIFFS_buffer_bytes_for_filedescs(spiffs *fs, u32_t num_descs) {

                                                 ^

D:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\Esp.cpp: In member function 'bool EspClass::eraseConfig()':

D:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\Esp.cpp:399:10: warning: unused variable 'ret' [-Wunused-variable]

     bool ret = true;

          ^

Archiving built core (caching) in: C:\Users\THEMIR~1\AppData\Local\Temp\arduino_cache_936876\core\core_esp8266com_esp8266_generic_CpuFrequency_80,CrystalFreq_26,FlashFreq_40,FlashMode_dio,UploadSpeed_115200,FlashSize_512K64,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____d8fd2f18ef2457db5d6b1c756af3e43c.a
Sketch uses 428107 bytes (98%) of program storage space. Maximum is 434160 bytes.
Global variables use 43164 bytes (52%) of dynamic memory, leaving 38756 bytes for local variables. Maximum is 81920 bytes.

```

devyte, 2017-10-31:
@brucemiranda Is this for core v2.3.0 or latest git? They are not the same thing.
The Async libs are not hosted in this repo, please report that to the appropriate author. Please edit your post and remove those warning to reduce noise.
For the rest, please add appropriate code markdown.

brucemiranda, 2017-10-31:
Done. Sorry I didn't realise that a bulk of the messages were for the Async libraries.

yoursunny, 2018-08-18:
#4043 has been merged. I guess this issue can close?

devyte, 2018-08-18:
Yes, thank you.

closed 2018-08-18 by devyte